kuba-moo/mt7601u #1: include/trace/define_trace.h:83:43: fatal error: ./trace.h: No such file or directory

miltador, 2015-02-18:
I have the following setup:
 - Ubuntu 14.04 LTS
 - 3.19.0-031900-generic #201502091451 kernel

Trying to compile the driver but get the following error:

  LD      /tmp/mt7601u/built-in.o
  CC [M]  /tmp/mt7601u/usb.o
  CC [M]  /tmp/mt7601u/init.o
  CC [M]  /tmp/mt7601u/main.o
  CC [M]  /tmp/mt7601u/mcu.o
  CC [M]  /tmp/mt7601u/trace.o
In file included from /tmp/mt7601u/trace.h:392:0,
                 from /tmp/mt7601u/trace.c:19:
include/trace/define_trace.h:83:43: fatal error: ./trace.h: No such file or directory
 #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
                                           ^
compilation terminated.

I guess it has smth to do with defines for tracing in trace.h but I don't know how to fix that.

miltador, 2015-02-18:
Easy fix for those who got the issue:
 - in trace.h find line ```#define TRACE_INCLUDE_PATH ./```
 - replace ```./``` with path to your source directory ending with ```/```

kuba-moo, 2015-02-18:
Strange, that's the standard way to define trace points. the exact line 
```c 
#define TRACE_INCLUDE_PATH .
```
is used in Linux kernel 51 times :/ I'm downloading Ubuntu, maybe I'll be able to reproduce this problem.

miltador, 2015-02-18:
Now it works fine without editing. Thanks!

closed 2015-02-18 by miltador