kuba-moo/mt7601u #2: insmod: ERROR: could not insert module ./mt7601u.ko: Unknown symbol in module

miltador, 2015-02-18:
I have the following setup:
 - Ubuntu 14.04 LTS x64
 - 3.19.0-031900-generic #201502091451 kernel

Driver compiled succesfully with following output:

```
miltador@miltahome:~/mt7601u$ make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Вход в каталог `/usr/src/linux-headers-3.19.0-031900-generic'
  LD      /home/miltador/mt7601u/built-in.o
  CC [M]  /home/miltador/mt7601u/usb.o
  CC [M]  /home/miltador/mt7601u/init.o
  CC [M]  /home/miltador/mt7601u/main.o
  CC [M]  /home/miltador/mt7601u/mcu.o
  CC [M]  /home/miltador/mt7601u/trace.o
  CC [M]  /home/miltador/mt7601u/dma.o
  CC [M]  /home/miltador/mt7601u/eeprom.o
  CC [M]  /home/miltador/mt7601u/phy.o
  CC [M]  /home/miltador/mt7601u/mac.o
  CC [M]  /home/miltador/mt7601u/debugfs.o
  CC [M]  /home/miltador/mt7601u/tx.o
  LD [M]  /home/miltador/mt7601u/mt7601u.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/miltador/mt7601u/mt7601u.mod.o
  LD [M]  /home/miltador/mt7601u/mt7601u.ko
make[1]: Выход из каталога `/usr/src/linux-headers-3.19.0-031900-generic
```

But when I try insmod I get the error:

```
root@miltahome:/home/miltador/mt7601u# insmod ./mt7601u.ko
insmod: ERROR: could not insert module ./mt7601u.ko: Unknown symbol in module
```

kuba-moo, 2015-02-18:
I think in the kernel logs there should be information which symbol he has a problem with. Kernel logs are in the file kern.log or you can display them by typing *dmesg*.

miltador, 2015-02-18:
dmesg output:
```
[ 4152.654075] mt7601u: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err 0)
[ 4152.654099] mt7601u: Unknown symbol ieee80211_tx_status_noskb (err 0)
[ 4152.654116] mt7601u: Unknown symbol ieee80211_free_hw (err 0)
[ 4152.654139] mt7601u: Unknown symbol ieee80211_register_hw (err 0)
[ 4152.654147] mt7601u: Unknown symbol ieee80211_get_hdrlen_from_skb (err 0)
[ 4152.654157] mt7601u: Unknown symbol ieee80211_wake_queue (err 0)
[ 4152.654171] mt7601u: Unknown symbol ieee80211_get_tx_rates (err 0)
[ 4152.654178] mt7601u: Unknown symbol ieee80211_send_bar (err 0)
[ 4152.654196] mt7601u: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err 0)
[ 4152.654206] mt7601u: Unknown symbol ieee80211_queue_delayed_work (err 0)
[ 4152.654220] mt7601u: Unknown symbol ieee80211_rx (err 0)
[ 4152.654231] mt7601u: Unknown symbol ieee80211_wake_queues (err 0)
[ 4152.654239] mt7601u: Unknown symbol ieee80211_free_txskb (err 0)
[ 4152.654246] mt7601u: Unknown symbol ieee80211_alloc_hw_nm (err 0)
[ 4152.654259] mt7601u: Unknown symbol ieee80211_tx_status (err 0)
[ 4152.654265] mt7601u: Unknown symbol ieee80211_stop_queue (err 0)
[ 4152.654271] mt7601u: Unknown symbol ieee80211_stop_queues (err 0)
[ 4152.654283] mt7601u: Unknown symbol ieee80211_unregister_hw (err 0)
```

kuba-moo, 2015-02-18:
Ouch, sure, my bad. Type ```modprobe mac80211``` and then ```insmod ./mt7601u.ko```.

miltador, 2015-02-18:
Now it works! Thanks. Please, consider to update this in README.

kuba-moo, 2015-02-18:
Yep, will do. Thanks for being the first user ;)

miltador, 2015-02-18:
My aim is to get my wifi dongle working with Raspberry Pi and OpenELEC as there is no driver in there. The previous driver https://github.com/porjo/mt7601 didn't work for me. I am going to test this one on raspberry today.

closed 2015-02-18 by miltador

kuba-moo, 2015-02-18:
Ok, I pushed two commits which should fix both of your issues. Let me know how it goes with the Pi.