kuba-moo/mt7601u #4: AP/Master mode?

archnous, 2015-02-22:
Thanks for this. I've tested on a i686/3.19 machine and it works as a charm in STA mode. However, while "iw phy0 info" shows AP in capabilities and hostapd configures it in master mode, the intended ESSID isn't visible (or invisible but existent, for that matter).

```
archbox:[root]:/temp/mt7601u-master# lsusb | grep 76
Bus 003 Device 002: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
archbox:[root]:/tmp/mt7601u-master# insmod  ./mt7601u.ko 
[dmesg]
***************************************************************************
[ 2314.963157] usb 3-1: reset high-speed USB device number 2 using ehci-pci
[ 2315.089319] ASIC revision: 76010001  MAC revision: 76010500
[ 2315.132444] mt7601u 3-1:1.0: EEPROM ver:0b fae:00
[ 2315.347619] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
[ 2315.348683] usbcore: registered new interface driver mt7601u
[ 2315.354302] mt7601u_add_interface idx:0
[ 2315.354312] mt76_configure_filter changed:180 total:80000180
[ 2315.354533] mt7601u_bss_info_changed 0000000e
[ 2315.354542] [prot transition] mode:0000 bgprot:0 non-gf:0 non-ht:0
[ 2315.357546] mt7601u_conf_tx 03 <- 0000
[ 2315.365426] mt7601u_conf_tx 02 <- 0001
[ 2315.368799] mt7601u_conf_tx 01 <- 0002
[ 2315.373178] mt7601u_conf_tx 00 <- 0003
[ 2315.376048] mt7601u_bss_info_changed 00002000
[ 2315.376059] mt7601u_config ffffffff ch:1
[ 2315.390442] mt76_configure_filter changed:100 total:80000180
[ 2315.390752] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[ 2315.390887] mt76_configure_filter changed:100 total:80000180
****************************************************************************

archbox:[root]:/temp/mt7601u-master# iwconfig wlan1
wlan1     IEEE 802.11bgn  Mode:Master  Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
archbox:[root]:/temp/mt7601u-master# iw dev wlan1 info
Interface wlan1
        ifindex 5
        wdev 0x100000001
        addr 00:e0:51:00:29:49
        ssid MT7601
        type AP
        wiphy 1
        channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
```
Launching hostapd sends these additional messages:
```
****************************************************************************
Feb 22 15:48:16 archbox kernel: [ 2034.748777] mt7601u_add_interface idx:0
Feb 22 15:48:16 archbox kernel: [ 2034.748785] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.749019] mt7601u_bss_info_changed 0000000e
Feb 22 15:48:16 archbox kernel: [ 2034.749023] [prot transition] mode:0000 bgprot:0 non-gf:0 non-ht:0
Feb 22 15:48:16 archbox kernel: [ 2034.752775] mt7601u_conf_tx 03 <- 0000
Feb 22 15:48:16 archbox kernel: [ 2034.755024] mt7601u_conf_tx 02 <- 0001
Feb 22 15:48:16 archbox kernel: [ 2034.757157] mt7601u_conf_tx 01 <- 0002
Feb 22 15:48:16 archbox kernel: [ 2034.758894] mt7601u_conf_tx 00 <- 0003
Feb 22 15:48:16 archbox kernel: [ 2034.760764] mt7601u_bss_info_changed 00002000
Feb 22 15:48:16 archbox kernel: [ 2034.760778] mt7601u_config ffffffff ch:7
Feb 22 15:48:16 archbox kernel: [ 2034.772677] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.772785] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Feb 22 15:48:16 archbox kernel: [ 2034.772903] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.776393] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.829968] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.849119] mt7601u_config 00000100 ch:7
Feb 22 15:48:16 archbox kernel: [ 2034.849126] mt7601u_config 00000040 ch:7
Feb 22 15:48:16 archbox kernel: [ 2034.859278] mt7601u_bss_info_changed 00004000
Feb 22 15:48:16 archbox kernel: [ 2034.859290] mt7601u_bss_info_changed 00088340
Feb 22 15:48:16 archbox kernel: [ 2034.859401] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Feb 22 15:48:16 archbox kernel: [ 2034.859426] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.859657] mt7601u_bss_info_changed 0000002e
Feb 22 15:48:16 archbox kernel: [ 2034.859690] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:16 archbox kernel: [ 2034.859691] basic rates: 0000000f
Feb 22 15:48:16 archbox kernel: [ 2034.862252] [prot transition] mode:0000 bgprot:0 non-gf:0 non-ht:0
Feb 22 15:48:16 archbox kernel: [ 2034.865730] mt7601u_set_key cmd:0 flg:0 kid:1 wid:7e
Feb 22 15:48:16 archbox kernel: [ 2034.865736] setting key for idx:7e
Feb 22 15:48:16 archbox kernel: [ 2034.866375] setting key for vif_idx:00 key_idx:01
Feb 22 15:48:16 archbox kernel: [ 2034.868185] mt7601u_conf_tx 03 <- 0000
Feb 22 15:48:16 archbox kernel: [ 2034.870336] mt7601u_bss_info_changed 00002000
Feb 22 15:48:16 archbox kernel: [ 2034.870672] mt7601u_conf_tx 02 <- 0001
Feb 22 15:48:16 archbox kernel: [ 2034.872500] mt7601u_bss_info_changed 00002000
Feb 22 15:48:16 archbox kernel: [ 2034.872563] mt7601u_conf_tx 01 <- 0002
Feb 22 15:48:16 archbox kernel: [ 2034.875002] mt7601u_bss_info_changed 00002000
Feb 22 15:48:16 archbox kernel: [ 2034.875103] mt7601u_conf_tx 00 <- 0003
Feb 22 15:48:16 archbox kernel: [ 2034.877581] mt7601u_bss_info_changed 00002000
Feb 22 15:48:17 archbox kernel: [ 2036.000155] mt76_configure_filter changed:100 total:80000180
Feb 22 15:48:29 archbox kernel: [ 2047.700524] Error: TX skb needs more tail - fail!!
Feb 22 15:49:18 archbox kernel: [ 2096.894645] Error: TX skb needs more tail - fail!!
****************************************************************************
```

I can provide any additional information you need.

kuba-moo, 2015-02-22:
Thanks for the report. Unfortunately AP/master is pretty low on my TODO list. In fact it should not be advertised as supported but the commit which disables it is buried in one of my WIP branches. I'm sure the support will be added one day together with mesh. I don't care that much for AP actually, the APs on Ralink/MediaTek USB devices are pretty terrible anyway. And I haven't decided whether to write ad-hoc or not. Seems like a waste of time these days.

So to make it short: AP is simply not there yet. BTW did you get the AP to work with the vendor driver?

archnous, 2015-02-22:
I've been using the code from https://github.com/housq/mt7601u (which is a patchlevel below the latest "official" driver). I've also created a PKGBUILD for Arch (https://aur.archlinux.org/packages/mt7601u-ap-dkms/) and used it for a month or so, out of necessity since I fried a RT5370 dongle. The aforementioned code is AP-only and has served me quite well, except for the TX rate which stays within 11g limits (i.e. <=54MBps). This is a 3-module case but I've also seen another AP project which integrates it in 1; haven't used it though as I had already created a satisfactory PKGBUILD.

I took the time to create a PKGBUILD for your code too: https://aur.archlinux.org/packages/mt7601u-dkms and, while I agree about the ad-hoc functionality, I've seen from my earlier search that a lot of people (and a great many rPi users among them) would put good use of an up-to-date driver with AP support. Thanks again for your work, I'll be watching for updates.

kuba-moo, 2015-02-22:
Thanks for information about AP stuff. I will definitely look into it at some point. My plan is to upstream the STA functionality first and then expand. My time resources are painfully limited though. I was even thinking about creating some sort of survey of which functions are most desired by users (AP, ad-hoc, power saving). There is also a MT7630E PCIe device which needs a proper Linux driver...

My feelings on packaging this driver are a bit mixed though. Because I make a lot of changes every single day and there are no version numbers I'm worried that people will start reporting bugs in old versions and I won't be able to reproduce them. Also will it be possible to push updated code to users when I fix something important on GH?

archnous, 2015-02-22:
The package is not binary; every time a user builds it they get the latest source from GH. However, since you mentioned it, if you version the code I will be able to "snapshot" the package with that version and users will see whether a newer version is available. For now, I've arbitrarily set the version number to 3.1 because it made some sense as your code is actually newer than the original 3.0.0.4. You don't need to implement any fancy stuff, any simple incrementing number will do - even a datestamp on every update.

kuba-moo, 2015-02-22:
OK, we will see what happens and introduce versioning if need be.

erwinkendo, 2015-06-03:
@archnous thank you for the tip about the AP driver and @kuba-moo , thank you for your great work and congratulations on making it to linux kernel :congratulations: 
It would be great if some progress is done to support AP mode. I am also using both drivers but I have come up with a problem if switching from using this STA driver to the AP driver (I shut down the network interfaces and remove the driver modules of course), the STA driver makes the AP driver not to work, I can't connect to it and I have to remove and plug the wifi dongle again for it to work. @archnous have you faced this problem?
I see two repos for AP mode more updated that the one @archnous mentioned  (this one) and [this one](https://github.com/kerpz/mt7601u) and [this one](https://github.com/eywalink/mt7601u), is the one mentioned more stable in some way?

archnous, 2015-06-03:
@erwinkendo I use [this repo](https://github.com/housq/mt7601u/) for my package and is *very* stable, but only works at 11g speeds (i.e. 54MBps TX and RX).

erwinkendo, 2015-06-03:
Thank you @archnous , I am using it as well, but if I load and use the module @kuba-moo created, and then try to use the one you mention again, I can't connect to the AP (even when it is broadcasting) and the only solution is to unplug and plug the usb again. Have you faced this problem?

archnous, 2015-06-04:
No, I only use it as an AP. Have you tried unloading all modules (both AP and kuba-moo's) and then reloading the ones desired, instead of un/pluging the dongle?

erwinkendo, 2015-06-04:
Yes, of course, I turn the interface off with ´´´ip link set ra0 down´´´
and remove all the modules for the AP driver, and for STA driver I set the
interface down and remove all the modules the mt7601u module started
(mac80211, ctr, ccm, arc4, cfg80211) and when starting the AP modules after
kuba-moo's have been removed, the AP is broadcasting but no connection can
be achieved.

cryptage21, 2015-12-01:
@kuba-moo thanks for your work. Driver is working well in Kernel 4.3 in STA.
I'd like to use my MT7601 dongles as AP but as indicated, that's not supported.

Do you think you'll work soon on AP mode ? 

Thanks :)

dimastt, 2016-04-30:
Hey Kuba in backtrack we can connect with airbase-ng any android devices with sucess. I, m using kali 2.0 and i think is a version of aircrack 1.2 r3 have a bug and hostapd dont work for most devices with dongle form.its complicated friend.hhave any good version of airbase ng for work with android devices?

johnoday, 2016-10-13:
+1 for ad-hoc mode

biberino, 2017-05-17:
Hey archnous,
The repo u mentioned looks good, im fine with the low speed as long as the AP mode runs stable.
which steps would i take to get the repo u [mentioned](https://github.com/housq/mt7601u/) to work under a raspberry? I cloned the repo, installed kernel headers and dependencie stuff and ran miwifi_build.sh successfully. How would i now get the AP mode to work? Thanks :)

adonespitogo, 2018-01-15:
I wish I knew how to make drivers :/ 
@kuba-moo thanks you for work, and to the people like you who make drivers for linux. Would really love to work on projects like this if I had the knowledge. Looking forward to AP capability being supported.

triswan, 2018-07-19:
Same here.. i need AP mode for my raspi