kuba-moo/mt7601u #5: Unable to compile: No such file or directory

Sam-Izdat, 2015-03-09:
Maybe I'm missing something obvious...

$ make
    make -C /lib/modules/`uname -r`/build M=$PWD
    make[1]: Entering directory `/usr/src/linux-headers-3.19.0-031900-generic'
    make[1]: Makefile: No such file or directory
    make[1]: *** No rule to make target `Makefile'.  Stop.
    make[1]: Leaving directory `/usr/src/linux-headers-3.19.0-031900-generic'
    make: *** [default] Error 2

$ ls /lib/firmware/ | grep mt7601u
    mt7601u.bin

$ ls /usr/src/ | grep linux-headers-3.19.0-031900-generic
    linux-headers-3.19.0-031900-generic

$ uname -r
    3.19.0-031900-generic

$ lsb_release -a
    Distributor ID: Ubuntu
    Description:    Ubuntu 14.04.2 LTS
    Release:        14.04
    Codename:       trusty

I wish I could say more, but I'm really not even sure what's happening.

kuba-moo, 2015-03-09:
Do you have linux-headers istalled? Is there anything inside /usr/src/linux-headers-3.19.0-031900-generic?

Sam-Izdat, 2015-03-09:
Upon inspecting further, I think you're right and the problem is the linux headers. After upgrading, `make` can't compile much of anything. I'll try to upgrade to a different kernel, check that the headers are there and report back.

closed 2015-03-09 by Sam-Izdat