kuba-moo/mt7601u #6: not working for TP-Link AC600/T2U 148f:761a

infinity0, 2015-03-25:
When I add this id to `usb.c` and load, here is what I get:

~~~~
[ 1631.317857] mt7601u 4-1.5:1.0: ASIC revision: 76100002 MAC revision: 76502000
[ 1631.319624] mt7601u 4-1.5:1.0: firmware: direct-loading firmware mt7601u.bin
[ 1631.319635] mt7601u 4-1.5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201308222153____
[ 1634.775834] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-110
[ 1637.915455] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-110
[..]
# physically unplug
[ 1654.101924] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-71
[ 1654.262111] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-71
[..]
[ 1660.677450] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-71
# rmmod
[ 1660.680976] usbcore: deregistering interface driver mt7601u
[ 1660.829589] mt7601u 4-1.5:1.0: Vendor request req:07 off:x0730 failed:-71
[..]
[ 1669.703792] mt7601u: probe of 4-1.5:1.0 failed with error -110
[ 1669.704504] usb 4-1.5: USB disconnect, device number 7
~~~~

The above is with `Archer T2U_V1_150108/Windows*/FW_7601.bin` (it's the same file) from Archer_T2U_V1_150108.zip from http://www.tp-link.com.de/support/download/?model=Archer+T2U&version=V1 installed as `/lib/firmware/mt7601u.bin`, but results are similar (i.e. error codes 110 and 71) with MT7601.bin from porjo/mt7601.

Removing `/lib/firmware/mt7601u.bin` makes the system ignore the device completely, so I guess they are doing *something*.

FWIW:

~~~~
$ sha256sum Archer_T2U_V1_150108.zip 
caec3a4c209dcb9d1c89f7ecec358134aae76b824aedae76f09af4022b4a3373  Archer_T2U_V1_150108.zip
~~~~

Let me know if you get a different checksum.

kuba-moo, 2015-03-25:
Your device is a MT7610, a newer and more powerful device which supports 802.11ac. This driver is for MT7601 - older and simpler 802.11n model. I'm not aware of anyone working on a driver for MT7610, so you probably should just use the vendor driver: http://www.mediatek.com/en/downloads/mt7610u-usb/

infinity0, 2015-03-26:
Thanks, I tried that and got it to "work". However, it kernel panics and it contains its own crypto primitives, so I don't think I really want to use it. Do you think it would be much work to adapt this driver to the newer models?

kuba-moo, 2015-03-26:
Yes, unfortunately I think it would not make much sense. In fact it may be a better idea to add support for the USB 802.11ac devices to the Felix's [mt76](https://github.com/openwrt/mt76). I did not dig deep into the code of the newer devices but they seem to use a different control commands and access rf/bbp in a different way. Plus obviously the whole PHY is redesigned to support 802.11ac...

For now I think it would make more sense to just fix up the vendor driver. I would be happy to do that, but I'm totally out of money until the end of March, so I can't afford any MT7610 device ;)

closed 2015-03-31 by kuba-moo

susmith97, 2015-10-28:
I have installed mt7601u driver. Im using TP-link  WN727n v4 . When i plug the adapter the log shows the error -110, vendor request failed. Whats the problem ? Help me please.

kuba-moo, 2015-11-04:
Does your card work with any other driver?

ysangkok, 2016-01-19:
@kuba-moo Did you get some money? I would like to sponsor a mt7610u for you if that would make you work on it.

kuba-moo, 2016-01-19:
@ysangkok Unfortunately I didn't and now I'm busy with other projects :(