kuba-moo/mt7601u #7: not working for Xiaodu mini wifi and Tencent mini wifi

whitehamster, 2015-03-31:
There are two devices which like Xiaomi mini wifi from China.
I'm sure they are mt7601u devices
Xiaodu mini wifi:
Bus 001 Device 011: ID 2955:1001
Tencent mini wifi:
Bus 001 Device 013: ID 2a5f:1000
Add support for them please;-)
Sorry for my poor english....

whitehamster, 2015-03-31:
More device ID
{USB_DEVICE(0x148f,0x760b)}, /* 360 Wifi */
{USB_DEVICE(0x2955,0x1001)}, /* baidu Wifi */
{USB_DEVICE(0x2a5f,0x1000)}, /* Tencent WIFI */
{USB_DEVICE(0×2955,0×0001)}, /* XiaoDu Wifi */
{USB_DEVICE(0x2717,0x4106)}, /* Xiaomi Wifi */

From [there](http://shumeipai.nxez.com/2014/12/07/raspberry-pi-use-360wifi.html)

kuba-moo, 2015-03-31:
Done, thanks for the report :)

closed 2015-03-31 by whitehamster