kuba-moo/mt7601u #8: make error

ArthurJam, 2015-04-02:
I try to compile your druver on my raspberry but i get an error.
Have you an idea to reach a success ?

root@raspberrypi:/home/pi/mt7601u# make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/mnt/storage/linux'
  LD      /home/pi/mt7601u/built-in.o
  CC [M]  /home/pi/mt7601u/usb.o
  CC [M]  /home/pi/mt7601u/init.o
  CC [M]  /home/pi/mt7601u/main.o
/home/pi/mt7601u/main.c:21:2: error: #error Your kernel is too old, please upgrade to at least 3.19
/home/pi/mt7601u/main.c:455:2: error: initialization from incompatible pointer type [-Werror]
/home/pi/mt7601u/main.c:455:2: error: (near initialization for ‘mt7601u_ops.sw_scan_start’) [-Werror]
/home/pi/mt7601u/main.c:456:2: error: initialization from incompatible pointer type [-Werror]
/home/pi/mt7601u/main.c:456:2: error: (near initialization for ‘mt7601u_ops.sw_scan_complete’) [-Werror]
/home/pi/mt7601u/main.c:461:2: error: unknown field ‘sta_rate_tbl_update’ specified in initializer
/home/pi/mt7601u/main.c:461:2: error: initialization from incompatible pointer type [-Werror]
/home/pi/mt7601u/main.c:461:2: error: (near initialization for ‘mt7601u_ops.get_survey’) [-Werror]
cc1: all warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/home/pi/mt7601u/main.o' failed
make[2]: *** [/home/pi/mt7601u/main.o] Error 1
Makefile:1381: recipe for target '_module_/home/pi/mt7601u' failed
make[1]: *** [_module_/home/pi/mt7601u] Error 2
make[1]: Leaving directory '/mnt/storage/linux'
Makefile:4: recipe for target 'default' failed
make: *** [default] Error 2

kuba-moo, 2015-04-02:
Please just read the error you got ;)
```
/home/pi/mt7601u/main.c:21:2: error: #error Your kernel is too old, please upgrade to at least 3.19
```
You need to upgrade the your kernel to 3.19. It seems that Raspbian is still using 3.18, you can try this method: http://raspberrypi.stackexchange.com/questions/17171/why-i-cant-update-kernel-version or build the kernel manually.

ArthurJam, 2015-04-02:
Im not proud of me ...
Sorry

closed 2015-04-02 by ArthurJam

ArthurJam, 2015-04-02:
Stuck on 3.18 on raspbian.
Is it possible to force the make on 3.18 ?

kuba-moo, 2015-04-02:
Unfortunately 3.18 does not have the *sta_rate_tbl_update* callback, so driver won't work properly. But Raspberry Pi has a really nice tutorial on compiling a kernel manually http://www.raspberrypi.org/documentation/linux/kernel/building.md you may want to give it a go.