kuba-moo/mt7601u #9: submit patch to kernel?

porjo, 2015-04-02:
I've just tried this on Fedora 20 (3.19 kernel landed in the last week) and it works well!

Just curious to know what the next steps are to get the driver into the official kernel source tree? (I'm not familiar with the process).

Thanks for efforts with this :smile:

kuba-moo, 2015-04-03:
Hi Ian :)
I'm almost there, I was hoping to have the driver in shape for first submission before Easter but I think it's not gonna happen. The code is mostly ready, I pushed a lot of changes to the *wip* branch and now I have to test it... Once that is done I will probably post the first version of the driver to the linux-wireless maling list and see what other developers have to say about it.

Linux 4.0 is coming out soon (next week, maybe two weeks from now) so I don't think I will be able to get it into 4.1 after all (once 4.0 is released, code of 4.1 will be frozen). 

I will keep you posted about the progress :)

kuba-moo, 2015-05-03:
Submitted for review:
http://marc.info/?l=linux-wireless&m=143057171032692&w=2

kuba-moo, 2015-06-01:
Accepted:
https://git.kernel.org/cgit/linux/kernel/git/next/linux-next.git/commit/?id=c869f77d6abb5d5f9f2f1a661d5c53862a9cad34

closed 2015-06-01 by kuba-moo

porjo, 2015-06-01:
Thanks, good work! Which kernel version will this appear in?

archnous, 2015-06-01:
Great news and thanks!

(Gnawing nails for AP/Master support)

kuba-moo, 2015-06-01:
@porjo It will appear in v4.2. First release candidate of v4.2 should be out in 3 weeks or so.
@archnous Actually I'm in contact with MediaTek and they suggest that support for other chips is more important than bringing AP/Master to client USB devices like mt7601u. I think I will be more productive cooperating with them on new devices than trying to figure out how old ones work on my own. Not to mention that they may give me some hardware which is a big plus transitioning into expensive 11ac world with a budget as slim as mine. Anyway AP is put aside for now :disappointed:

archnous, 2015-06-01:
Congrats, I'm glad they've acknowledged your contributions! :)