kuba-moo/mt7601u #12: automatically load after reboot

tfcarlin, 2015-04-04:
First off, thanks for all your work on this driver.

I'm a complete noob with linux drivers and as such I'm not really clear on how to make this automatically survive reboots. A "make install" might be helpful (and I would grab from another driver's makefile if I knew what I was looking for).

I'm not exactly sure what sequence of commands ultimately caused the driver to "stick" but it has been working since the following.

enabled wlan0 in /etc/network/interfaces

auto wlan0
iface wlan0 inet dhcp
    wpa-ssid "ssid"
    wpa-psk  "security_key"

added two lines to /etc/modules

mac80211
mt7601u

And finally (I'm guessing one of the cp commands isn't needed)

cp mt7601u.ko /lib/modules/3.19.3-bone4/kernel/drivers/net/wireless/
cp /lib/modules/3.19.3-bone4/kernel/drivers/net/wireless/mt7601u.ko /lib/modules/3.19.3-bone4/kernel/net/
depmod

rrthomas, 2015-06-05:
Most of the above is not necessary: all I needed was:

    cp mt7601u.ko /lib/modules/`uname -r`/kernel/drivers/net/wireless/
    depmod

kuba-moo, 2015-06-05:
Hi guys! I avoid `make install` because most users will not realize that it have to be rerun after each kernel update. I will add some info to README.md though, thanks.

closed 2015-06-05 by kuba-moo

rrthomas, 2015-06-05:
Thanks for the extra documentation. I appreciate not wanting to confuse users, but I think we can have the best of both worlds here: add a make install target (which can do the bits left out of your instructions, such as "sudo cp", and "chmod" the file (I have umask 0027, so the file is built 0640)), and then document that it needs to be done for each new kernel version. This avoids users who will now use your driver until they get 4.2 having to type several commands each time (in my case that will be at least six months, when the Ubuntu 15.10 kernel packages for 14.04 LTS are released).

In other words, move the commands now in README.md into the Makefile under the "install" target, and write "make install" in README.md.

kuba-moo, 2015-06-07:
@rrthomas I'm OK with the changes you are describing but I'm quite short on time. Please make a fork and send me a pull request with those changes. Can you do that? :)

rrthomas, 2015-06-07:
Sure, I'll do that when I get a moment, most likely when I next need it (next time my distro pushes a kernel update!).

prakashkannan123, 2015-12-16:
After reboot laptop don't recognize the driver of mt7601u...

prakashkannan123, 2015-12-16:
So. Every Time I need to install my driver.. To use the wifi dongle..