kuba-moo/mt7601u #13: problem with ubuntu 15.04

helium78, 2015-04-04:
It doesn't work for me
Should give me a wlan1 device.

````
$ uname -r
3.19.0-12-generic
````

Kernel log:
````
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.841983] usb 1-1: new high-speed USB device number 5 using xhci_hcd
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.981183] usb 1-1: New USB device found, idVendor=148f, idProduct=7601
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.981192] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.981197] usb 1-1: Product: 802.11 n WLAN
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.981200] usb 1-1: Manufacturer: MediaTek
Apr  4 21:29:05 timo-Aspire-E1-572G kernel: [  263.981203] usb 1-1: SerialNumber: 1.0
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.397217] mt7601u: module verification failed: signature and/or  required key missing - tainting kernel
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398049] 
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398049] **********************************************************
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398055] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398057] **                                                      **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398059] ** trace_printk() being used. Allocating extra memory.  **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398061] **                                                      **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398063] ** This means that this is a DEBUG kernel and it is     **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398064] ** unsafe for production use.                           **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398066] **                                                      **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398068] ** If you see this message and you are not debugging    **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398070] ** the kernel, report this immediately to your vendor!  **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398072] **                                                      **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398074] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.398076] **********************************************************
Apr  4 21:30:08 timo-Aspire-E1-572G kernel: [  327.564852] usb 1-1: reset high-speed USB device number 5 using xhci_hcd
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694805] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abcd8
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694809] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc00
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694810] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abd20
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694812] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc48
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694813] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abd68
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694814] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abdb0
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694816] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc90
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.694817] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abdf8
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.695294] mt7601u 1-1:1.0: ASIC revision: 76010001 MAC revision: 76010500
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  327.707372] mt7601u 1-1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  328.084392] mt7601u 1-1:1.0: Warning: unsupported EEPROM version 0d
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  328.084396] mt7601u 1-1:1.0: EEPROM ver:0d fae:00
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  328.149995] usb 1-1: USB disconnect, device number 5
Apr  4 21:30:09 timo-Aspire-E1-572G kernel: [  328.435923] mt7601u 1-1:1.0: Warning: mt7601u_mcu_wait_resp retrying
Apr  4 21:30:10 timo-Aspire-E1-572G kernel: [  328.735696] mt7601u 1-1:1.0: Warning: mt7601u_mcu_wait_resp retrying
Apr  4 21:30:10 timo-Aspire-E1-572G kernel: [  329.035441] mt7601u 1-1:1.0: Warning: mt7601u_mcu_wait_resp retrying
Apr  4 21:30:10 timo-Aspire-E1-572G kernel: [  329.335246] mt7601u 1-1:1.0: Warning: mt7601u_mcu_wait_resp retrying
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  329.635012] mt7601u 1-1:1.0: Warning: mt7601u_mcu_wait_resp retrying
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  329.635022] mt7601u 1-1:1.0: Error: mt7601u_mcu_wait_resp timed out
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  329.637869] mt7601u: probe of 1-1:1.0 failed with error -110
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  329.637974] usbcore: registered new interface driver mt7601u
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  329.878829] usb 1-1: new high-speed USB device number 6 using xhci_hcd
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.017998] usb 1-1: New USB device found, idVendor=148f, idProduct=7601
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.018006] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.018011] usb 1-1: Product: 802.11 n WLAN
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.018014] usb 1-1: Manufacturer: MediaTek
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.018017] usb 1-1: SerialNumber: 1.0
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.130897] usb 1-1: reset high-speed USB device number 6 using xhci_hcd
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260860] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abcd8
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260865] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc00
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260867] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abd20
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260869] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc48
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260871] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abd68
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260873] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abdb0
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260875] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abc90
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.260877] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800948abdf8
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.261453] mt7601u 1-1:1.0: ASIC revision: 76010001 MAC revision: 76010500
Apr  4 21:30:11 timo-Aspire-E1-572G kernel: [  330.262063] mt7601u 1-1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[... 55 lines cut ...]
Apr  4 21:32:26 timo-Aspire-E1-572G kernel: [  464.940460] Error: BBP reg changed!?
Apr  4 21:32:29 timo-Aspire-E1-572G kernel: [  468.059216] mt7601u 1-1:1.0: Vendor request req:07 off:x101c failed:-110
Apr  4 21:32:32 timo-Aspire-E1-572G kernel: [  471.177985] mt7601u 1-1:1.0: Vendor request req:02 off:x101c failed:-110
Apr  4 21:32:35 timo-Aspire-E1-572G kernel: [  474.296721] mt7601u 1-1:1.0: Vendor request req:07 off:x101c failed:-110
Apr  4 21:32:38 timo-Aspire-E1-572G kernel: [  477.415562] mt7601u 1-1:1.0: Vendor request req:02 off:x101c failed:-110
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254455] INFO: task systemd-udevd:2687 blocked for more than 120 seconds.
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254465]       Tainted: G           OE  3.19.0-12-generic #12-Ubuntu
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254468] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[... cut ...]
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254560]  [<ffffffff811f3d78>] __vfs_read+0x18/0x50
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254566]  [<ffffffff811f3e3a>] vfs_read+0x8a/0x140
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254573]  [<ffffffff811f3f36>] SyS_read+0x46/0xb0
Apr  4 21:32:41 timo-Aspire-E1-572G kernel: [  480.254581]  [<ffffffff817c928d>] system_call_fastpath+0x16/0x1b
Apr  4 21:32:42 timo-Aspire-E1-572G kernel: [  480.534324] mt7601u 1-1:1.0: Vendor request req:07 off:x101c failed:-110
````

kuba-moo, 2015-04-04:
I think you are connecting to a USB3.0, right? Can you try using a USB2.0 port?

helium78, 2015-04-04:
This is with USB2.0
````
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.182994] usb 1-3: new high-speed USB device number 5 using xhci_hcd
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.322671] usb 1-3: New USB device found, idVendor=148f, idProduct=7601
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.322679] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.322684] usb 1-3: Product: 802.11 n WLAN
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.322687] usb 1-3: Manufacturer: MediaTek
Apr  4 23:12:41 timo-Aspire-E1-572G kernel: [ 5873.322690] usb 1-3: SerialNumber: 1.0
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.716188] mt7601u: module verification failed: signature and/or  required key missing - tainting kernel
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717101] 
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717101] **********************************************************
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717105] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717107] **                                                      **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717108] ** trace_printk() being used. Allocating extra memory.  **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717110] **                                                      **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717112] ** This means that this is a DEBUG kernel and it is     **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717113] ** unsafe for production use.                           **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717115] **                                                      **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717116] ** If you see this message and you are not debugging    **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717118] ** the kernel, report this immediately to your vendor!  **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717119] **                                                      **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717121] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.717123] **********************************************************
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5937.883689] usb 1-3: reset high-speed USB device number 5 using xhci_hcd
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013646] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f8d8
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013651] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f800
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013652] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f920
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013654] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f848
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013655] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f968
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013657] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f9b0
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013658] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f890
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.013660] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880158c5f9f8
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.014137] mt7601u 1-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.014658] mt7601u 1-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.394306] mt7601u 1-3:1.0: Warning: unsupported EEPROM version 0d
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.394315] mt7601u 1-3:1.0: EEPROM ver:0d fae:00
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.397466] mt7601u 1-3:1.0: Error: RX urb failed:-71
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.397513] mt7601u 1-3:1.0: Error: RX urb failed:-71
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.397601] usb 1-3: USB disconnect, device number 5
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.425894] mt7601u_rxdc_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436781] mt7601u_rxdc_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436794] mt7601u_tssi_dc_gain_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436796] mt7601u_tssi_dc_gain_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436799] mt7601u_tssi_dc_gain_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436801] mt7601u_tssi_dc_gain_cal timed out
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436803] ------------[ cut here ]------------
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436816] WARNING: CPU: 3 PID: 2986 at /home/timo/wifi/mt7601u/phy.c:571 lin2dBd+0xb5/0xc0 [mt7601u]()
Apr  4 23:13:46 timo-Aspire-E1-572G kernel: [ 5938.436818] Modules linked in: mt7601u(OE+) ctr ccm rfcomm bnep nls_iso8859_1 snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_intel snd_soc_rt5640 acer_wmi  [... cut ...]
Apr  4 23:16:45 timo-Aspire-E1-572G kernel: [ 6117.278128]  [<ffffffff8126ce50>] kernfs_fop_read+0xa0/0x170
Apr  4 23:16:45 timo-Aspire-E1-572G kernel: [ 6117.278138]  [<ffffffff811f3d78>] __vfs_read+0x18/0x50
Apr  4 23:16:45 timo-Aspire-E1-572G kernel: [ 6117.278145]  [<ffffffff811f3e3a>] vfs_read+0x8a/0x140
Apr  4 23:16:45 timo-Aspire-E1-572G kernel: [ 6117.278152]  [<ffffffff811f3f36>] SyS_read+0x46/0xb0
Apr  4 23:16:45 timo-Aspire-E1-572G kernel: [ 6117.278161]  [<ffffffff817c928d>] system_call_fastpath+0x16/0x1b
````

kuba-moo, 2015-04-08:
I've looked into this and unfortunately it looks like a hardware failure to me. Can you try the vendor driver (https://github.com/porjo/mt7601) and report if it works?

helium78, 2015-04-08:
I don't have problems using USB hard drives at these ports. What kernels can I use with the vendor driver? Are there any Linux live systems that I can test it with an old kernel and the vendor driver?

kuba-moo, 2015-04-09:
I meant that the WiFi dongle can be broken, especially if it's one of the cheap "black and red" no-names.

I gave you a link to @projo's driver because it should work with any kernel, new or old. If it doesn't let me know I will fix it for you.

cusspvz, 2015-05-19:
Same here with `15.04`, running inside a VB vm on a Macbook Pro retina.

kuba-moo, 2015-05-19:
@cusspvz Logs or didn't happen. Also please test if @projo's driver works as I suggested above.

cusspvz, 2015-05-20:
@projo driver is working, with some flaws.
Didn't posted logs since they were equal to the ones provided by @helium78 .

`mt7601u 1-3:1.0: Vendor request req:02 off:x0506 failed:-110` keeps showing up on `syslog`.

kuba-moo, 2015-06-05:
After a lot of thinking I might have an idea - can you test the `debug` branch?

rrthomas, 2015-06-07:
I'm still getting these crash logs with the latest debug branch; is there any other information I can supply?

kuba-moo, 2015-06-07:
In @helium78's case the device got completely mad for reason which is not clear to me. Your device @rrthomas just has incomplete eeprom and everything is fine, I should ignore the values and use zeros instead. @cusspvz didn't feel it's appropriate to put logs in his error report so I have no idea what his problem was.

Please do not assume that if something looks like "the same crash" to your untrained eyes, it's actually the same problem ;)

cusspvz, 2015-06-07:
@kuba-moo didn't tested yet since i have to install ubuntu again on VirtualBox. :disappointed: 
But ASAP i would do that (on the next days i'm busy as hell).

I can advance to you that my VM crashed every time on about 3 minutes after I loaded the driver.

rrthomas, 2015-06-07:
Sorry, when you said in issue #18 "I think we may be dealing with two separate issues here - *the xHCI xhci_drop_endpoint called with disabled ep which I've already been asked about*" I thought that referred to this issue.

q4a, 2015-07-07:
Ubuntu 15.04 works fine:
$ uname -a
Linux amd 3.19.0-22-generic #22-Ubuntu SMP Tue Jun 16 17:15:15 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
$ lsmod|grep mt76
mt7601u                98304  0 
mac80211              724992  1 mt7601u
cfg80211              540672  2 mac80211,mt7601u

But i need to run for wifi:
sudo rfkill unblock wifi
sudo wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/example.conf
sudo dhcpcd wlan0

kuba-moo, 2015-07-08:
@q4a Sounds like a configuration issue with higher layer software. If wpa_supplicant works fine the driver is OK.

closed 2015-07-08 by kuba-moo