kuba-moo/mt7601u #14: bulk upload fails: Error: TX skb - not enough tailroom

chrisbainbridge, 2015-04-19:
With 4.0 kernel (64fb1d0) and latest mt7601u (ba391b3)

Go to http://www.speedtest.net
- Download test completes ok
- Upload test fails completely

dmesg shows:
```
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:32:55 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:01 debian kernel: mt7601u_skb_rooms: 23 callbacks suppressed
Apr 19 13:33:01 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:01 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:02 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:03 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:08 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:08 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
Apr 19 13:33:08 debian kernel: mt7601u 1-1:1.0: Error: TX skb - not enough tailroom
```

kuba-moo, 2015-04-23:
Can you please check the fix I pushed to master? :)

chrisbainbridge, 2015-04-23:
Bug fixed - ran speedtest with no errors.

closed 2015-04-23 by chrisbainbridge