kuba-moo/mt7601u #15: mt7601u.h:161:27: error: field ‘chandef’ has incomplete type

Centurion, 2015-04-30:
Hello,
when I try to make, the above error occurs (Ubuntu 12.04, gcc 4.8.1)
thanks

kuba-moo, 2015-04-30:
Please post the full build log and tell me which version of the kernel do you have?

Centurion, 2015-05-01:
$make 
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Betrete Verzeichnis '/usr/src/linux-headers-3.2.0-32-generic'
  LD      /home/robert/mt7601u/built-in.o
  CC [M]  /home/robert/mt7601u/usb.o
In file included from /home/robert/mt7601u/usb.c:18:0:
/home/robert/mt7601u/mt7601u.h:161:27: error: field ‘chandef’ has incomplete type
  struct cfg80211_chan_def chandef;
                           ^
/home/robert/mt7601u/mt7601u.h:263:14: error: ‘IEEE80211_NUM_TIDS’ undeclared here (not in a function)
  u16 agg_ssn[IEEE80211_NUM_TIDS];
              ^
/home/robert/mt7601u/mt7601u.h:363:10: error: ‘struct ieee80211_tx_control’ declared inside parameter list [-Werror]
   struct sk_buff *skb);
          ^
/home/robert/mt7601u/mt7601u.h:363:10: error: its scope is only this definition or declaration, which is probably not what you want [-Werror]
/home/robert/mt7601u/usb.c:342:2: error: unknown field ‘disable_hub_initiated_lpm’ specified in initializer
  .disable_hub_initiated_lpm = 1,
  ^
/home/robert/mt7601u/usb.c:342:2: error: excess elements in struct initializer [-Werror]
/home/robert/mt7601u/usb.c:342:2: error: (near initialization for ‘mt7601u_driver’) [-Werror]
/home/robert/mt7601u/usb.c:344:1: error: data definition has no type or storage class [-Werror]
 module_usb_driver(mt7601u_driver);
 ^
/home/robert/mt7601u/usb.c:344:1: error: type defaults to ‘int’ in declaration of ‘module_usb_driver’ [-Werror=implicit-int]
/home/robert/mt7601u/usb.c:344:1: error: parameter names (without types) in function declaration [-Werror]
/home/robert/mt7601u/usb.c:333:26: error: ‘mt7601u_driver’ defined but not used [-Werror=unused-variable]
 static struct usb_driver mt7601u_driver = {
                          ^
cc1: all warnings being treated as errors
make[2]: *** [/home/robert/mt7601u/usb.o] Fehler 1
make[1]: *** [_module_/home/robert/mt7601u] Fehler 2
make[1]: Verlasse Verzeichnis '/usr/src/linux-headers-3.2.0-32-generic'
make: *** [default] Fehler 2

well, maybe 3.2.0 is too old?
thanks again

kuba-moo, 2015-05-03:
Yes, 3.2.0 is way too old. This driver needs at least 3.19.

closed 2015-05-03 by kuba-moo