kuba-moo/mt7601u #16: Multiple devices failure

maggedotno, 2015-05-02:
On a Raspberry Pi running kernel 3.19.3 I am able to initiate 1 x 148f:760b Ralink Technology, Corp. device, but not 2.

I've tried changing around the devices, it does not matter, it will always be the first one that intialized with success.

Apologies if this is a non-mt7601u limitation or if it was not supposed to work.

Tailing the syslog when inserting device 1:

```
May  2 09:44:04 magge-berry kernel: [ 1489.082405] usb 1-1.3.2: new high-speed USB device number 17 using dwc_otg
May  2 09:44:04 magge-berry kernel: [ 1489.194068] usb 1-1.3.2: New USB device found, idVendor=148f, idProduct=760b
May  2 09:44:04 magge-berry kernel: [ 1489.194111] usb 1-1.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May  2 09:44:04 magge-berry kernel: [ 1489.194133] usb 1-1.3.2: Product: 802.11 n WLAN
May  2 09:44:04 magge-berry kernel: [ 1489.194151] usb 1-1.3.2: Manufacturer: MediaTek
May  2 09:44:04 magge-berry kernel: [ 1489.194208] usb 1-1.3.2: SerialNumber: 1.0
May  2 09:44:04 magge-berry kernel: [ 1489.282440] usb 1-1.3.2: reset high-speed USB device number 17 using dwc_otg
May  2 09:44:04 magge-berry kernel: [ 1489.385792] mt7601u 1-1.3.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
May  2 09:44:04 magge-berry kernel: [ 1489.388285] mt7601u 1-1.3.2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
May  2 09:44:05 magge-berry kernel: [ 1489.803998] mt7601u 1-1.3.2:1.0: Warning: unsupported EEPROM version 0d
May  2 09:44:05 magge-berry kernel: [ 1489.804039] mt7601u 1-1.3.2:1.0: EEPROM ver:0d fae:00
May  2 09:44:05 magge-berry kernel: [ 1490.083012] ieee80211 phy8: Selected rate control algorithm 'minstrel_ht'
May  2 09:44:05 magge-berry ifplugd(wlan0)[3771]: ifplugd 0.28 initializing.
May  2 09:44:05 magge-berry ifplugd(wlan0)[3771]: Using interface wlan0/00:36:76:07:F0:A4 with driver <mt7601u> (version: 3.19.3+)
May  2 09:44:05 magge-berry ifplugd(wlan0)[3771]: Using detection mode: SIOCETHTOOL
May  2 09:44:05 magge-berry ifplugd(wlan0)[3771]: Initialization complete, link beat not detected.
```
When inserting device 2:

```
May  2 09:45:52 magge-berry kernel: [ 1596.876598] usb 1-1.3.1: new high-speed USB device number 18 using dwc_otg
May  2 09:45:52 magge-berry kernel: [ 1596.988458] usb 1-1.3.1: New USB device found, idVendor=148f, idProduct=760b
May  2 09:45:52 magge-berry kernel: [ 1596.988498] usb 1-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May  2 09:45:52 magge-berry kernel: [ 1596.988517] usb 1-1.3.1: Product: 802.11 n WLAN
May  2 09:45:52 magge-berry kernel: [ 1596.988534] usb 1-1.3.1: Manufacturer: MediaTek
May  2 09:45:52 magge-berry kernel: [ 1596.988551] usb 1-1.3.1: SerialNumber: 1.0
May  2 09:45:52 magge-berry kernel: [ 1597.076664] usb 1-1.3.1: reset high-speed USB device number 18 using dwc_otg
May  2 09:45:52 magge-berry kernel: [ 1597.180214] mt7601u 1-1.3.1:1.0: ASIC revision: 76010001 MAC revision: 76010500
May  2 09:45:52 magge-berry kernel: [ 1597.182766] mt7601u 1-1.3.1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
May  2 09:45:53 magge-berry kernel: [ 1597.569937] mt7601u: probe of 1-1.3.1:1.0 failed with error -12
```

kuba-moo, 2015-05-03:
Your Pi doesn't have enough free memory to allocate receive buffers for two devices. You can just limit the number of buffers allocated by the driver from 64 to 32 or 16. I chose 64 arbitrarily, there should be no impact if you change it to a lower number. 

To change the number of buffers replace 64 at [line 72 of mt7601u.h](https://github.com/kuba-moo/mt7601u/blob/master/mt7601u.h#L72) (N_RX_ENTRIES) with 32.

maggedotno, 2015-05-03:
Thank you! :+1: And also thank you for taking the effort for making the driver, I should have written that initially.

I confirm this works and I'm now able to init 2 x 192.168.10.160 on the RPi.

Cheers!

closed 2015-05-03 by maggedotno

maggedotno, 2015-05-03:
Confirmed. master and N_RX_ENTRIES	 16 works.

kuba-moo, 2015-05-03:
Thanks for testing!