kuba-moo/mt7601u #18: Crash on init

rrthomas, 2015-06-05:
First, thanks VERY much for this driver. Hardware vendors have been rubbish twice: Broadcom, whose driver for the built-in chipset of my machine barely works, and Mediatek, as already documented. Your efforts to compensate for others' faults are much appreciated.

The driver crashes on init, giving the following backtrace. It works fine afterwards (as far as I can tell), but the kernel backtrace makes me nervous in case it's corrupted something! I'm using a MacBookPro11,1.

    [   12.379093] init: failsafe main process (875) killed by TERM signal
    [   12.469563] usb 1-2: reset high-speed USB device number 2 using xhci_hcd
    [   12.565591] usb 1-3.2: USB disconnect, device number 6
    [   12.598642] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781cd8
    [   12.598646] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781c00
    [   12.598648] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781d20
    [   12.598650] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781c48
    [   12.598652] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781d68
    [   12.598654] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781db0
    [   12.598656] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781c90
    [   12.598658] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff880457781df8
    [   12.599033] mt7601u 1-2:1.0: ASIC revision: 76010001 MAC revision: 76010500
    [   12.626764] mt7601u 1-2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
    [   13.017017] mt7601u 1-2:1.0: EEPROM ver:0b fae:00
    [   13.017240] ------------[ cut here ]------------
    [   13.017248] WARNING: CPU: 2 PID: 562 at /home/rrt/.local/var/repo/mt7601u/eeprom.h:126 mt7601u_set_power_rate+0x5e/0x90 [mt7601u]()
    [   13.017249] Modules linked in: joydev mt7601u(OE+) mac80211 btusb snd_hda_codec_cirrus snd_hda_codec_generic applesmc input_polldev bcm5974 i915_bpo intel_ips intel_rapl iosf_mbi x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm bnep snd_hda_intel snd_hda_controller snd_hda_codec rfcomm snd_hwdep wl(POE) bluetooth snd_pcm snd_seq_midi lpc_ich snd_seq_midi_event mei_me snd_rawmidi snd_seq thunderbolt snd_seq_device mei snd_timer cfg80211 bdc_pci shpchp snd soundcore binfmt_misc sbs sbshc apple_bl mac_hid nls_iso8859_1 parport_pc ppdev lp parport btrfs xor raid6_pq dm_crypt hid_generic hid_apple usbhid hid crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul i915 glue_helper ablk_helper uas cryptd ahci libahci i2c_algo_bit usb_storage drm_kms_helper drm video
    [   13.017295] CPU: 2 PID: 562 Comm: systemd-udevd Tainted: P           OE  3.19.0-18-generic #18~14.04.1-Ubuntu
    [   13.017297] Hardware name: Apple Inc. MacBookPro11,1/Mac-189A3D4F975D5FFC, BIOS MBP111.88Z.0138.B07.1402121134 02/12/2014
    [   13.017298]  ffffffffc09128e0 ffff880456847a48 ffffffff817adfa3 0000000000000000
    [   13.017302]  0000000000000000 ffff880456847a88 ffffffff81074c2a 000000000000100c
    [   13.017305]  ffff8804591935cc ffff8804590c1ac0 0000000000000000 0000000000000000
    [   13.017308] Call Trace:
    [   13.017315]  [<ffffffff817adfa3>] dump_stack+0x45/0x57
    [   13.017319]  [<ffffffff81074c2a>] warn_slowpath_common+0x8a/0xc0
    [   13.017322]  [<ffffffff81074d1a>] warn_slowpath_null+0x1a/0x20
    [   13.017328]  [<ffffffffc090ab9e>] mt7601u_set_power_rate+0x5e/0x90 [mt7601u]
    [   13.017333]  [<ffffffffc090afbb>] mt7601u_eeprom_init+0x3eb/0x730 [mt7601u]
    [   13.017339]  [<ffffffffc09052ae>] mt7601u_init_hardware+0x42e/0x4a0 [mt7601u]
    [   13.017344]  [<ffffffffc090478c>] mt7601u_probe+0x1dc/0x240 [mt7601u]
    [   13.017348]  [<ffffffff815b014b>] usb_probe_interface+0x1bb/0x300
    [   13.017352]  [<ffffffff814f4c6d>] driver_probe_device+0x12d/0x3e0
    [   13.017355]  [<ffffffff814f4ffb>] __driver_attach+0x9b/0xa0
    [   13.017359]  [<ffffffff814f4f60>] ? __device_attach+0x40/0x40
    [   13.017362]  [<ffffffff814f2b73>] bus_for_each_dev+0x63/0xa0
    [   13.017365]  [<ffffffff814f464e>] driver_attach+0x1e/0x20
    [   13.017367]  [<ffffffff814f4240>] bus_add_driver+0x180/0x240
    [   13.017369]  [<ffffffff814f57e4>] driver_register+0x64/0xf0
    [   13.017371]  [<ffffffff815ae912>] usb_register_driver+0x82/0x160
    [   13.017374]  [<ffffffffc0772000>] ? 0xffffffffc0772000
    [   13.017378]  [<ffffffffc077201e>] mt7601u_driver_init+0x1e/0x20 [mt7601u]
    [   13.017381]  [<ffffffff81002144>] do_one_initcall+0xd4/0x210
    [   13.017384]  [<ffffffff811cedf9>] ? kmem_cache_alloc_trace+0x199/0x220
    [   13.017387]  [<ffffffff810f77dd>] ? load_module+0x162d/0x1ca0
    [   13.017389]  [<ffffffff810f7816>] load_module+0x1666/0x1ca0
    [   13.017391]  [<ffffffff810f30e0>] ? store_uevent+0x40/0x40
    [   13.017393]  [<ffffffff810f7fc6>] SyS_finit_module+0x86/0xb0
    [   13.017396]  [<ffffffff817b598d>] system_call_fastpath+0x16/0x1b
    [   13.017398] ---[ end trace 2346238fc44cfb5f ]---
    [   13.017399] ------------[ cut here ]------------
    [   13.017402] WARNING: CPU: 2 PID: 562 at /home/rrt/.local/var/repo/mt7601u/eeprom.h:126 mt7601u_set_power_rate+0x7c/0x90 [mt7601u]()
    [   13.017403] Modules linked in: joydev mt7601u(OE+) mac80211 btusb snd_hda_codec_cirrus snd_hda_codec_generic applesmc input_polldev bcm5974 i915_bpo intel_ips intel_rapl iosf_mbi x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm bnep snd_hda_intel snd_hda_controller snd_hda_codec rfcomm snd_hwdep wl(POE) bluetooth snd_pcm snd_seq_midi lpc_ich snd_seq_midi_event mei_me snd_rawmidi snd_seq thunderbolt snd_seq_device mei snd_timer cfg80211 bdc_pci shpchp snd soundcore binfmt_misc sbs sbshc apple_bl mac_hid nls_iso8859_1 parport_pc ppdev lp parport btrfs xor raid6_pq dm_crypt hid_generic hid_apple usbhid hid crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul i915 glue_helper ablk_helper uas cryptd ahci libahci i2c_algo_bit usb_storage drm_kms_helper drm video
    [   13.017436] CPU: 2 PID: 562 Comm: systemd-udevd Tainted: P        W  OE  3.19.0-18-generic #18~14.04.1-Ubuntu
    [   13.017438] Hardware name: Apple Inc. MacBookPro11,1/Mac-189A3D4F975D5FFC, BIOS MBP111.88Z.0138.B07.1402121134 02/12/2014
    [   13.017439]  ffffffffc09128e0 ffff880456847a48 ffffffff817adfa3 0000000000000000
    [   13.017442]  0000000000000000 ffff880456847a88 ffffffff81074c2a 000000000000100c
    [   13.017445]  ffff8804591935cc 000000000000003f 0000000000000000 0000000000000000
    [   13.017447] Call Trace:
    [   13.017451]  [<ffffffff817adfa3>] dump_stack+0x45/0x57
    [   13.017454]  [<ffffffff81074c2a>] warn_slowpath_common+0x8a/0xc0
    [   13.017456]  [<ffffffff81074d1a>] warn_slowpath_null+0x1a/0x20
    [   13.017461]  [<ffffffffc090abbc>] mt7601u_set_power_rate+0x7c/0x90 [mt7601u]
    [   13.017464]  [<ffffffffc090afbb>] mt7601u_eeprom_init+0x3eb/0x730 [mt7601u]
    [   13.017467]  [<ffffffffc09052ae>] mt7601u_init_hardware+0x42e/0x4a0 [mt7601u]
    [   13.017470]  [<ffffffffc090478c>] mt7601u_probe+0x1dc/0x240 [mt7601u]
    [   13.017472]  [<ffffffff815b014b>] usb_probe_interface+0x1bb/0x300
[... 660 lines cut ...]
    [   14.956675] vboxdrv: Successfully loaded version 4.3.28 (interface 0x001a000a).
    [   15.171335] vboxpci: IOMMU not found (not registered)
    [   15.423768] wlan1: authenticate with 7c:4c:a5:a3:6c:f5
    [   15.440174] wlan1: send auth to 7c:4c:a5:a3:6c:f5 (try 1/3)
    [   15.441723] wlan1: authenticated
    [   15.442511] wlan1: associate with 7c:4c:a5:a3:6c:f5 (try 1/3)
    [   15.446239] wlan1: RX AssocResp from 7c:4c:a5:a3:6c:f5 (capab=0x411 status=0 aid=2)
    [   15.473889] wlan1: associated
    [   15.473908] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
    [   16.001288] wlan1: deauthenticating from 7c:4c:a5:a3:6c:f5 by local choice (Reason: 2=PREV_AUTH_NOT_VALID)
    [   16.039882] wlan1: authenticate with 7c:4c:a5:a3:6c:f5
    [   16.048572] wlan1: send auth to 7c:4c:a5:a3:6c:f5 (try 1/3)
    [   16.048714] cfg80211: Calling CRDA to update world regulatory domain
    [   16.050150] wlan1: authenticated
    [   16.050680] cfg80211: World regulatory domain updated:
    [   16.050683] cfg80211:  DFS Master region: unset
[... cut ...]
    [   16.050689] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
    [   16.050691] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm), (N/A)
    [   16.050693] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
    [   16.050694] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm), (N/A)
    [   16.054754] wlan1: associate with 7c:4c:a5:a3:6c:f5 (try 1/3)
    [   16.058072] wlan1: RX AssocResp from 7c:4c:a5:a3:6c:f5 (capab=0x411 status=0 aid=2)
    [   16.089977] wlan1: associated

kuba-moo, 2015-06-05:
Thanks for kind words!
I think we may be dealing with two separate issues here - the `xHCI xhci_drop_endpoint called with disabled ep` which I've already been asked about and weird EEPROM content. Could you please pull the new code from the repo and build the debug branch?
After doing so and reloading the driver (`rmmod mt7601u; insmod ./mt7601u`) could you post the content of following files:
`cat /sys/kernel/debug/ieee80211/phy*/mt7601u/eeprom_param`
`hexdump -C /sys/kernel/debug/ieee80211/phy*/mt7601u/eeprom`
BTW are you running on bare metal or inside some sort of hipervisor (VirtualBox, KVM, Parallels etc)?

rrthomas, 2015-06-05:
Thanks very much for your attention to this bug (and very glad to hear you got the driver upstream!). I am running "on the metal".

Building the debug branch fails for me:

      CC [M]  /home/rrt/.local/var/repo/mt7601u/debugfs.o
    /home/rrt/.local/var/repo/mt7601u/debugfs.c:125:2: error: initialization from incompatible pointer     type [-Werror]
      .read = mt7601u_eeprom_read,
      ^
    /home/rrt/.local/var/repo/mt7601u/debugfs.c:125:2: error: (near initialization for ‘fops_eeprom.read’) [-Werror]
    cc1: all warnings being treated as errors

kuba-moo, 2015-06-07:
Oh. So sorry, I only tested a 32bit build. Please try now.

rrthomas, 2015-06-07:
Thanks for the fix. `/sys/kernel/debug/ieee80211/phy*/mt7601u/eeprom_param`:

    RF freq offset: 57
    RSSI offset: 0 0
    Reference temp: ff
    LNA gain: 0
    Reg channels: 1-14
    Per rate power:
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    	 raw:3f bw20:ff bw40:ff
    Per channel power:
    	 tx_power  ch1:00 ch2:00
    	 tx_power  ch3:00 ch4:00
    	 tx_power  ch5:00 ch6:00
    	 tx_power  ch7:00 ch8:00
    	 tx_power  ch9:00 ch10:00
    	 tx_power  ch11:00 ch12:00
    	 tx_power  ch13:00 ch14:00

and hexdump -C /sys/kernel/debug/ieee80211/phy*/mt7601u/eeprom:

    00000000  01 76 00 0b 00 e1 b0 10  10 01 ff ff ff ff ff ff  |.v..............|
    00000010  01 02 8f 14 01 76 00 00  4a 00 01 00 80 50 08 00  |.....v..J....P..|
    00000020  03 00 02 00 00 00 ff 02  00 20 40 ff ff ff ff ff  |......... @.....|
    00000030  ff ff ff ff 11 ff 00 00  ff ff 57 01 ff ff 99 99  |..........W.....|
    00000040  8c 88 ff 0f 00 00 00 00  00 00 00 00 00 00 00 00  |................|
    00000050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
    *
    00000100

kuba-moo, 2015-06-07:
And did the `xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep` messages go away?

kuba-moo, 2015-06-07:
Thanks for the info. Please try `debug` branch now, warnings should be gone. Let me know if the `xhci` message persists.

rrthomas, 2015-06-07:
Great, now I get no warnings or traceback (and yes, the `xhci` message is gone).

Does that mean I can just go back to `master` and install the module?

Thanks very much for your work on this.

kuba-moo, 2015-06-08:
Please hold off with going back to master - I want to consult the `xhci` problem with the Linux USB maintainers. I'll let you know when I push the fixes to master.

kuba-moo, 2015-06-08:
Actually the `xhci` was supposedly fixed by torvalds/linux@a6134136d938ed9298f15 so I'm not pushing the other fix from `debug` unless guys at #13 indicate that it helps them.
To sum up:
  * you can use `master` now;
  * "crashes" should be gone;
  * `xhci` message will persist until you upgrade your kernel.
Thanks for participating :)

rrthomas, 2015-06-08:
Confirmed, with current master the kernel tracebacks are gone, and I still get the xhci messages but no problem.

closed 2015-06-08 by kuba-moo