kuba-moo/mt7601u #23: DMA memory mapping error

Rethil, 2015-06-18:
While insmoding driver on raspberry pi with 4.0.5 kernel kmsg shows that:
```
[    9.075183] ------------[ cut here ]------------
[    9.075218] WARNING: CPU: 0 PID: 162 at lib/dma-debug.c:1166 check_for_stack+0xc4/0x100()
[    9.075226] dwc_otg bcm2708_usb: DMA-API: device driver maps memory from stack [addr=aa217c54]
[    9.075240] Modules linked in: mt7601u(O+)
[    9.075255] CPU: 0 PID: 162 Comm: systemd-udevd Tainted: G           O    4.0.5-v7+ #6
[    9.075259] Hardware name: BCM2709
[    9.075316] [<80016618>] (unwind_backtrace) from [<80012c1c>] (show_stack+0x20/0x24)
[    9.075338] [<80012c1c>] (show_stack) from [<80827344>] (dump_stack+0x9c/0xdc)
[    9.075361] [<80827344>] (dump_stack) from [<8002b7d4>] (warn_slowpath_common+0x98/0xc8)
[    9.075383] [<8002b7d4>] (warn_slowpath_common) from [<8002b844>] (warn_slowpath_fmt+0x40/0x48)
[    9.075403] [<8002b844>] (warn_slowpath_fmt) from [<8033a67c>] (check_for_stack+0xc4/0x100)
[    9.075418] [<8033a67c>] (check_for_stack) from [<8033add8>] (debug_dma_map_page+0xd8/0xf8)
[    9.075438] [<8033add8>] (debug_dma_map_page) from [<80435478>] (usb_hcd_map_urb_for_dma+0x4fc/0x544)
[    9.075456] [<80435478>] (usb_hcd_map_urb_for_dma) from [<804357b4>] (usb_hcd_submit_urb+0x150/0x7fc)
[    9.075475] [<804357b4>] (usb_hcd_submit_urb) from [<80436d4c>] (usb_submit_urb+0x2fc/0x4b8)
[    9.075493] [<80436d4c>] (usb_submit_urb) from [<80437384>] (usb_start_wait_urb+0x5c/0xcc)
[    9.075512] [<80437384>] (usb_start_wait_urb) from [<804374ac>] (usb_control_msg+0xb8/0xe8)
[    9.075564] [<804374ac>] (usb_control_msg) from [<7f0003b0>] (mt7601u_vendor_request+0xc8/0x1dc [mt7601u])
[    9.075622] [<7f0003b0>] (mt7601u_vendor_request [mt7601u]) from [<7f000558>] (mt7601u_rr+0x54/0x158 [mt7601u])
[    9.075687] [<7f000558>] (mt7601u_rr [mt7601u]) from [<7f005f60>] (mt7601u_wait_asic_ready+0x2c/0x74 [mt7601u])
[    9.075749] [<7f005f60>] (mt7601u_wait_asic_ready [mt7601u]) from [<7f000748>] (mt7601u_probe+0xec/0x220 [mt7601u])
[    9.075790] [<7f000748>] (mt7601u_probe [mt7601u]) from [<8043ab04>] (usb_probe_interface+0xdc/0x1a8)
[    9.075813] [<8043ab04>] (usb_probe_interface) from [<8038c084>] (really_probe+0x8c/0x220)
[    9.075832] [<8038c084>] (really_probe) from [<8038c324>] (__driver_attach+0xa8/0xac)
[    9.075849] [<8038c324>] (__driver_attach) from [<8038a3b4>] (bus_for_each_dev+0x7c/0xb0)
[    9.075868] [<8038a3b4>] (bus_for_each_dev) from [<8038bc04>] (driver_attach+0x2c/0x30)
[    9.075888] [<8038bc04>] (driver_attach) from [<8038b88c>] (bus_add_driver+0x15c/0x204)
[    9.075906] [<8038b88c>] (bus_add_driver) from [<8038c7fc>] (driver_register+0x88/0x108)
[    9.075923] [<8038c7fc>] (driver_register) from [<8043a4a8>] (usb_register_driver+0x80/0x144)
[    9.075969] [<8043a4a8>] (usb_register_driver) from [<7f013028>] (mt7601u_driver_init+0x28/0x2c [mt7601u])
[    9.076014] [<7f013028>] (mt7601u_driver_init [mt7601u]) from [<800088a0>] (do_one_initcall+0x9c/0x1e8)
[    9.076033] [<800088a0>] (do_one_initcall) from [<808264a8>] (do_init_module+0x70/0x1bc)
[    9.076050] [<808264a8>] (do_init_module) from [<8009e1c4>] (load_module+0x17c8/0x1e44)
[    9.076067] [<8009e1c4>] (load_module) from [<8009e9d4>] (SyS_finit_module+0x78/0x88)
[    9.076084] [<8009e9d4>] (SyS_finit_module) from [<8000ed20>] (ret_fast_syscall+0x0/0x54)
[    9.076092] ---[ end trace 6db4547001e02d7c ]---```

It looks like mt7601u_rr is passing local reg variable to mt7601u_vendor_request.
```
	__le32 reg;

	ret = mt7601u_vendor_request(dev, MT_VEND_MULTI_READ, USB_DIR_IN,
					 0, offset, &reg, sizeof(reg));
```

kuba-moo, 2015-06-20:
Thanks a lot for the report, I thought I had DMA_API_DEBUG=y on my test system :confounded: You may expect a fix very soon.

Rethil, 2015-06-20:
Thanks a lot. Does your test system have any other debug options enabled? I was able to spot few more problems while having debugs turned on.

kuba-moo, 2015-06-20:
Can you please try the `debug` branch and confirm that this issue is solved there?

Rethil, 2015-06-21:
After short test it seems that issue is fixed. Thanks ;).

closed 2015-06-21 by Rethil