kuba-moo/mt7601u #24: Support for ASUS USB N53 B1 dual-band wireless adapter?

davidcsterratt, 2015-06-20:
I've bought an ASUS USB N53 B1 dual-band wireless adaptor, mistakenly thinking it was a ASUS USB N53, which is supported by the rt2800.ko module ( https://wikidevi.com/wiki/ASUS_USB-N53 )

I've installed the driver from the bundled CD in Windows, where the string in the "Inf section" of "Details of driver" is OS61_RTWLANR_7662_C5.ndi . I suppose this indicates that the chipset may be a MT7662. 

The Mediatek downloads site ( http://www.mediatek.com/en/downloads/ ) has a Windows USB driver for the MT7612U, but not MT7662 - perhaps this means the same driver works for both chipsets?  There are both MT7612U and MT7662 mentioned at https://wikidevi.com/wiki/MediaTek .

Mediatek also have a Linux driver for MT7612U on their downloads page. I've tried compiling this but it fails after a pile of warnings. 

I was wondering if the mt7601u might work for MT7662 chipset, and if so, how to go about testing this. I'm not experienced at hacking drivers.

Apologies if this is the wrong place to ask - if so I'd be grateful if you could point me to a better place.

kuba-moo, 2015-06-20:
This driver is not appropriate for mt7662u, unfortunately.
Felix (aka @nbd) wrote the [mt76](https://github.com/openwrt/mt76) driver which works for mt7662 chipset on PCIe bus. MediaTek is aware that they lack Linux driver for mt7662u (the usb version of mt7662) and they contacted me some time ago about writing it. I said I will be glad to help and since then I'm waiting for them to finish paperwork and send me some devices.
Open Source things are known to progress slowly at MediaTek, maybe @nbd has more information about their plans. Or you can email MediaTek yourself and see what they have to say :) 
Anyway I can't help you until I get my hands on hardware samples. As you seen yourself buying a device with a specific chipset from a shop can be hard so I don't feel particularly eager to shed  €20 on a device which may turn out to be sth else...

davidcsterratt, 2015-06-20:
Thanks for your speedy reply and the pointer to Felix's work. I'd be happy to send either of you the device I have, if it would help. I'm not sure who's work would be the best starting point for a USB driver.

davidcsterratt, 2015-06-23:
I've sent the following message to MediaTek:

> I recently bought an ASUS USB N53 B1 dual-band wireless adapter (vendorID 0b05, productID 180b). I thought it was an ASUS USB N53, which is supported by the rt2800.ko module ( https://wikidevi.com/wiki/ASUS_USB-N53 ).   I was disappointed to learn that it is not supported by Linux "out of the box".

> I understand that the chipset may be a MediaTek MT7662.  I've tried compiling the Linux driver for MT7612U on the MediaTek downloads page, but it failed to compile.

> I understand that there is a community effort to write a working driver ( https://github.com/kuba-moo/mt7601u/issues/24#issuecomment-113781329 ) and that it depends on you completing paperwork and sending a device to Jakub Kicinski. I would be very grateful if you could complete the paperwork and send him a device as soon as possible, so that I can enjoy your excellent hardware.

ccpaging, 2015-10-12:
+1 Waiting for MT7612U driver...

kuba-moo, 2015-11-04:
I'm closing this issue, guys. Sadly, I have no plans to work on MT7612U/MT7662U in a foreseeable future.

closed 2015-11-04 by kuba-moo

davidcsterratt, 2016-05-18:
In the School of Informatics at the University of Edinburgh students have the option of working on a project during a "summer of code", I was wondering if writing an MT7662 driver would be a feasible project, If so would it be best to start from this codebase or @nbd's [mt76](https://github.com/openwrt/mt76) code?

kuba-moo, 2016-05-19:
With MT7662u I would probably lean more towards Felix's driver.  This one is significantly simpler, though, which could be important.