kuba-moo/mt7601u #30: Can't get ra0

egnascimento, 2015-08-09:
I cross compiled it for Freescale VF50 running Angstrom Kernel 4.0.5. It went just fine. I copied the module and loaded with insmod after copying firmware .BIN to path as indicated. Messages I got were exactly as expected. There is only one warning about EEPROM 0x0d which is said not a big deal in step by step. After all module is loaded (checking with lsmod) but there is no ra0/wlan0/wifi0 or any new interface by checking with ifconfig.

Any tips?

egnascimento, 2015-08-10:
Being more precise, that is what I get from dmesg after loading module.

[  218.886019] usb 1-1: reset high-speed USB device number 2 using ci_hdrc
[  219.041130] mt7601u 1-1:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  219.117774] mt7601u 1-1:1.0: Warning: unsupported EEPROM version 0d
[  219.124148] mt7601u 1-1:1.0: EEPROM ver:0d fae:00
[  219.423482] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[  219.478250] usbcore: registered new interface driver mt7601u

Should be ok, right? Why can't I see ra0 then?

egnascimento, 2015-08-10:
I found out what was going on. rfkill was blocking it. I hope it is useful for someone in the future.

kuba-moo, 2015-08-10:
Cool. I'm closing the bug then.

closed 2015-08-10 by kuba-moo