kuba-moo/mt7601u #34: System hangs when using speedof.me (kernel 4.2.0-rc8, x86_64)

euspectre, 2015-08-31:
OS: ROSA R6 x86_64
Device: TP-Link WN727Nv4, based on MT7601U.
USB IDs: 148f-7601
Kernel: 4.2.0-rc8

With kernel 4.2.0-rc8 and mt7601u that comes with it, the system hangs when I test the network speed with http://speedof.me/ or open YouTube. 
No reaction even to the magic SysRq. 

Unfortunately, I could not obtain the backtrace of the crash itself yet. However, there was the following kernel warning some time before it, not sure if it is related:

<pre>
------------[ cut here ]------------
WARNING: CPU: 0 PID: 0 at /home/tester/work/kernel/kernel.org/linux-4.2-rc8/kernel/softirq.c:150
Modules linked in: nf_log_ipv4 nf_log_common xt_LOG xt_recent iptable_nat nf_nat_ipv4 xt_comment
 slhc af_packet msr hwmon_vid arc4 mt7601u mac80211 cfg80211 rfkill coretemp iTCO_wdt iTCO_vendo
CPU: 0 PID: 0 Comm: swapper/0 Tainted: G          I     4.2.0-rc8-test03 #3
Hardware name: System manufacturer System Product Name/Rampage III GENE, BIOS 0704    09/21/2010
 ffffffff81a88e20 ffff8801b9203ba8 ffffffff8170c08d 0000000000000007
 0000000000000000 ffff8801b9203be8 ffffffff81071c9a ffff8801b4d4a230
 0000000000000201 ffff8801b5caa42e ffff8801b52c26a0 ffff8800369bb0b0
Call Trace:
 <IRQ>  [<ffffffff8170c08d>] dump_stack+0x4c/0x6e
 [<ffffffff81071c9a>] warn_slowpath_common+0x8a/0xc0
 [<ffffffff81071d8a>] warn_slowpath_null+0x1a/0x20
 [<ffffffff810757d7>] __local_bh_enable_ip+0x77/0xb0
 [<ffffffff81712c7e>] _raw_spin_unlock_bh+0x1e/0x20
// include/linux/spinlock.h:362, inlined at
// net/mac80211/rate.h:50, inlined at
// net/mac80211/status.c:792
 [<ffffffffc0650f0b>] ieee80211_tx_status+0x72b/0xc80 [mac80211] 
 [<ffffffffc05cae54>] ? ieee80211_get_hdrlen_from_skb+0x24/0x40 [cfg80211]
// drivers/net/wireless/mediatek/mt7601u/tx.c:120:
 [<ffffffffc0725007>] mt7601u_tx_status+0x87/0xa0 [mt7601u]
// drivers/net/wireless/mediatek/mt7601u/dma.c:241:
 [<ffffffffc071f98e>] mt7601u_complete_tx+0x8e/0x1c0 [mt7601u]
 [<ffffffff81597e25>] __usb_hcd_giveback_urb+0x85/0x130
 [<ffffffff81597fd3>] usb_hcd_giveback_urb+0x43/0x140
 [<ffffffffc0092b23>] xhci_irq+0xdb3/0x2060 [xhci_hcd]
 [<ffffffffc0093de1>] xhci_msi_irq+0x11/0x20 [xhci_hcd]
 [<ffffffff810c5dd9>] handle_irq_event_percpu+0x39/0x1c0
 [<ffffffff810c5fa4>] handle_irq_event+0x44/0x70
 [<ffffffff810c8ed8>] handle_edge_irq+0x78/0x140
 [<ffffffff81017392>] handle_irq+0x22/0x40
 [<ffffffff81016bbf>] do_IRQ+0x4f/0xe0
 [<ffffffff81713d6b>] common_interrupt+0x6b/0x6b
 <EOI>  [<ffffffff8160d171>] ? cpuidle_enter_state+0xf1/0x260
 [<ffffffff8160d150>] ? cpuidle_enter_state+0xd0/0x260
 [<ffffffff8160d317>] cpuidle_enter+0x17/0x20
 [<ffffffff810afd2b>] call_cpuidle+0x3b/0x70
 [<ffffffff8160d2f3>] ? cpuidle_select+0x13/0x20
 [<ffffffff810affd3>] cpu_startup_entry+0x273/0x330
 [<ffffffff816ff57a>] rest_init+0x8a/0x90
 [<ffffffff81d4cf88>] start_kernel+0x43f/0x44c
 [<ffffffff81d4c120>] ? early_idt_handler_array+0x120/0x120
 [<ffffffff81d4c4d7>] x86_64_start_reservations+0x2a/0x2c
 [<ffffffff81d4c614>] x86_64_start_kernel+0x13b/0x14a
</pre>

Other hardware info: http://hw.rosalinux.ru/index.php?probe=f0a0117ca9

I encountered no such problems with kernel 4.1.6 and mt7601u from this project (rev. ae46ece5). Looks like the source code of the driver itself is the same as in 4.2.0-rc8, which makes it strange.

kuba-moo, 2015-09-01:
Linux 4.2 doesn't have the four commits from *Jun 23, 2015*. I will contemplate pushing those commits into a stable release (4.2.1), they are quite big thought.

closed 2015-09-01 by kuba-moo

euspectre, 2015-09-01:
I see. Thanks for your quick reply. 
I will try to use the code from this project on kernel 4.2 as well then.

euspectre, 2015-09-03:
Well, I patched kernel 4.2 with the changes from the following commits from this repository:
c10471f714826f623e65e915d73efb845066cec4
a65998456202ba099f264bfbb3500e4f075cc02e
f53f406150fe8d9c2ec1cf783eef3d4d5df0f6c4
d3faadb5da88f2d2874572757b087e4b95942796

The problem is gone, the drivers has worked OK so far.

Thanks for your help!

fonini, 2015-11-06:
I'm using Kernel 4.2.3-3 from OSMC and I'm getting the following warnings:

````
[  486.961249] ------------[ cut here ]------------
[  486.961379] WARNING: CPU: 0 PID: 95 at drivers/net/wireless/mediatek/mt7601u/dma.c:291 mt7601u_dma_enqueue_tx+0x29c/0x2b4 [mt7601u]()
[  486.961397] Modules linked in: evdev uinput sha256_generic sha256_arm hmac drbg ctr ccm arc4 mt7601u mac80211 sg cfg80211 uas rfkill bcm2835_thermal bcm2835_gpiomem lirc_rpi(C) lirc_dev uio_pdrv_genirq uio rc_core fuse ipv6
[  486.961503] CPU: 0 PID: 95 Comm: systemd-journal Tainted: G        WC      4.2.3-3-osmc #1
[  486.961516] Hardware name: BCM2708
[  486.961563] [<c001dc88>] (unwind_backtrace) from [<c001ad28>] (show_stack+0x20/0x24)
[  486.961605] [<c001ad28>] (show_stack) from [<c0729348>] (dump_stack+0x20/0x28)
[  486.961634] [<c0729348>] (dump_stack) from [<c002d8fc>] (warn_slowpath_null+0x78/0xfc)
[  486.961708] [<c002d8fc>] (warn_slowpath_null) from [<bf2a86b4>] (mt7601u_dma_enqueue_tx+0x29c/0x2b4 [mt7601u])
[  486.961822] [<bf2a86b4>] (mt7601u_dma_enqueue_tx [mt7601u]) from [<bf2af1ec>] (mt7601u_tx+0x2dc/0x408 [mt7601u])
[  486.962349] [<bf2af1ec>] (mt7601u_tx [mt7601u]) from [<bf1f97d8>] (__ieee80211_tx+0x340/0x69c [mac80211])
[  486.962918] [<bf1f97d8>] (__ieee80211_tx [mac80211]) from [<bf1fa92c>] (ieee80211_xmit+0x314/0x690 [mac80211])
[  486.963474] [<bf1fa92c>] (ieee80211_xmit [mac80211]) from [<bf1fc768>] (__ieee80211_subif_start_xmit+0x470/0x1650 [mac80211])
[  486.964032] [<bf1fc768>] (__ieee80211_subif_start_xmit [mac80211]) from [<bf1fd964>] (ieee80211_subif_start_xmit+0x1c/0x24 [mac80211])
[  486.964346] [<bf1fd964>] (ieee80211_subif_start_xmit [mac80211]) from [<c061e518>] (dev_hard_start_xmit+0x24c/0x458)
[  486.964392] [<c061e518>] (dev_hard_start_xmit) from [<c06455cc>] (__qdisc_run+0x1b8/0x38c)
[  486.964420] [<c06455cc>] (__qdisc_run) from [<c0614fe4>] (net_tx_action+0x11c/0x244)
[  486.964450] [<c0614fe4>] (net_tx_action) from [<c0031a8c>] (__do_softirq+0x170/0x49c)
[  486.964473] [<c0031a8c>] (__do_softirq) from [<c0032258>] (irq_exit+0xf8/0x140)
[  486.964494] [<c0032258>] (irq_exit) from [<c0083df8>] (__handle_domain_irq+0x88/0xe8)
[  486.964522] [<c0083df8>] (__handle_domain_irq) from [<c0017680>] (handle_IRQ+0x2c/0x30)
[  486.964548] [<c0017680>] (handle_IRQ) from [<c0009714>] (bcm2835_handle_irq+0x1c4/0x1ec)
[  486.964576] [<c0009714>] (bcm2835_handle_irq) from [<c0730834>] (__irq_usr+0x54/0x80)
[  486.964588] Exception stack(0xd4e47fb0 to 0xd4e47ff8)
[  486.964602] 7fa0:                                     7f6d8df8 00000002 00000002 00000002
[  486.964622] 7fc0: 7f6db150 7f6d8030 00000010 00000000 00000016 00000001 00000003 00000000
[  486.964637] 7fe0: 000000ec be951970 7f575fd0 7f564580 20000010 ffffffff
[  486.964654] ---[ end trace c7fd71fbe4ef81b6 ]---
````

My Raspberry is freezing sometimes and the connection hangs frequently.