kuba-moo/mt7601u #35: kernel panics and warnings

furlongm, 2015-09-03:
Hi,

Getting consistent kernel panics and warnings with the builtin driver on kernel 4.2.0-1.gefc468a-desktop. I haven't captured the panics yet, but the output from the warning is below.

```
Sep 03 15:57:53 linux-e3ncd5 systemd-udevd[585]: worker [604] /devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb8 is taking a long time
Sep 03 15:57:53 linux-e3ncd5 kernel: usb 8-2: device descriptor read/64, error -110
Sep 03 15:57:54 linux-e3ncd5 kernel: usb 8-2: reset high-speed USB device number 2 using xhci_hcd
Sep 03 15:57:54 linux-e3ncd5 kernel: mt7601u 8-2:1.0: ASIC revision: 76010001 MAC revision: 76010500
Sep 03 15:57:54 linux-e3ncd5 kernel: mt7601u 8-2:1.0: Warning: unsupported EEPROM version 0d
Sep 03 15:57:54 linux-e3ncd5 kernel: mt7601u 8-2:1.0: EEPROM ver:0d fae:00
Sep 03 15:57:54 linux-e3ncd5 kernel: ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Sep 03 15:57:54 linux-e3ncd5 mtp-probe[1841]: checking bus 8, device 2: "/sys/devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb8/8-2"
Sep 03 15:57:54 linux-e3ncd5 mtp-probe[1841]: bus: 8, device: 2 was not an MTP device
Sep 03 15:57:55 linux-e3ncd5 kernel: mt7601u 8-2:1.0 wlp5s0u2: renamed from wlan0
Sep 03 15:57:55 linux-e3ncd5 systemd[1]: Starting system-systemd\x2drfkill.slice.
Sep 03 15:57:55 linux-e3ncd5 systemd[1]: Created slice system-systemd\x2drfkill.slice.
Sep 03 15:57:55 linux-e3ncd5 systemd[1]: Starting Load/Save RF Kill Switch Status of rfkill0...
Sep 03 15:57:55 linux-e3ncd5 systemd[1]: Started Load/Save RF Kill Switch Status of rfkill0.
Sep 03 15:57:55 linux-e3ncd5 systemd-sysctl[1880]: Overwriting earlier assignment of kernel/sysrq in file '/etc/sysctl.d/99-sysctl.conf'.
Sep 03 15:57:55 linux-e3ncd5 kernel: IPv6: ADDRCONF(NETDEV_UP): wlp5s0u2: link is not ready
Sep 03 15:57:55 linux-e3ncd5 wpa_supplicant[1008]: ioctl[SIOCSIWENCODEEXT]: Invalid argument
Sep 03 15:57:55 linux-e3ncd5 wpa_supplicant[1008]: ioctl[SIOCSIWENCODEEXT]: Invalid argument
Sep 03 15:57:55 linux-e3ncd5 wickedd[1056]: unable to translate eap protocol AKA
Sep 03 15:57:55 linux-e3ncd5 wickedd[1056]: unable to translate eap protocol AKA'
Sep 03 15:57:55 linux-e3ncd5 wickedd[1056]: unable to translate eap protocol FAST
Sep 03 15:57:56 linux-e3ncd5 kernel: wlp5s0u2: authenticate with 00:08:54:99:bf:9e
Sep 03 15:57:56 linux-e3ncd5 wpa_supplicant[1008]: ioctl[SIOCSIWFREQ]: Device or resource busy
Sep 03 15:57:57 linux-e3ncd5 kernel: wlp5s0u2: send auth to 00:08:54:99:bf:9e (try 1/3)
Sep 03 15:57:57 linux-e3ncd5 kernel: wlp5s0u2: authenticated
Sep 03 15:57:57 linux-e3ncd5 kernel: mt7601u 8-2:1.0 wlp5s0u2: disabling HT as WMM/QoS is not supported by the AP
Sep 03 15:57:57 linux-e3ncd5 kernel: mt7601u 8-2:1.0 wlp5s0u2: disabling VHT as WMM/QoS is not supported by the AP
Sep 03 15:57:57 linux-e3ncd5 kernel: wlp5s0u2: associate with 00:08:54:99:bf:9e (try 1/3)
Sep 03 15:57:57 linux-e3ncd5 kernel: wlp5s0u2: RX AssocResp from 00:08:54:99:bf:9e (capab=0x411 status=0 aid=1)
Sep 03 15:57:57 linux-e3ncd5 kernel: wlp5s0u2: associated
Sep 03 15:57:57 linux-e3ncd5 kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlp5s0u2: link becomes ready
Sep 03 15:57:57 linux-e3ncd5 kernel: ------------[ cut here ]------------
Sep 03 15:57:57 linux-e3ncd5 kernel: WARNING: CPU: 1 PID: 570 at ../kernel/softirq.c:150 __local_bh_enable_ip+0x77/0xa0()
Sep 03 15:57:57 linux-e3ncd5 kernel: Modules linked in: arc4 xt_tcpudp iptable_filter ip_tables x_tables rfcomm bluetooth fuse iscsi_ibft iscsi_boot_sysfs w83627ehf af_packet hwmon_vid snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel snd_hda_codec snd
Sep 03 15:57:57 linux-e3ncd5 kernel: CPU: 1 PID: 570 Comm: systemd-journal Not tainted 4.2.0-1.gefc468a-desktop #1
Sep 03 15:57:57 linux-e3ncd5 kernel: Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./E350M1/USB3, BIOS P1.80 12/21/2012
Sep 03 15:57:57 linux-e3ncd5 kernel:  ffffffff81a48485 ffff88006e503990 ffffffff816a0b9c 0000000000000007
Sep 03 15:57:57 linux-e3ncd5 kernel:  0000000000000000 ffff88006e5039d0 ffffffff81068226 0000000000000092
Sep 03 15:57:57 linux-e3ncd5 kernel:  0000000000000201 ffff8800472d40b0 ffff88004719d830 ffff880036c706a0
Sep 03 15:57:57 linux-e3ncd5 kernel: Call Trace:
Sep 03 15:57:57 linux-e3ncd5 kernel:  <IRQ>  [<ffffffff816a0b9c>] dump_stack+0x4c/0x6e
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81068226>] warn_slowpath_common+0x86/0xc0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff8106831a>] warn_slowpath_null+0x1a/0x20
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff8106c017>] __local_bh_enable_ip+0x77/0xa0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81091242>] ? default_wake_function+0x12/0x20
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a671a>] _raw_write_unlock_bh+0x1a/0x20
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a672e>] _raw_spin_unlock_bh+0xe/0x10
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa00c1109>] ieee80211_tx_status+0x769/0xd10 [mac80211]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa01a9087>] mt7601u_tx_status+0x87/0xa0 [mt7601u]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa01a391c>] mt7601u_complete_tx+0x8c/0x1c0 [mt7601u]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff814f7055>] __usb_hcd_giveback_urb+0x85/0x130
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff814f7203>] usb_hcd_giveback_urb+0x43/0x140
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa0466460>] handle_tx_event+0x820/0x1420 [xhci_hcd]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81650401>] ? ip6_finish_output2+0x131/0x460
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa04672f5>] xhci_irq+0x295/0xb20 [xhci_hcd]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81652c4f>] ? ip6_finish_output+0x8f/0xf0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81652cf4>] ? ip6_output+0x44/0xe0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffffa0467b91>] xhci_msi_irq+0x11/0x20 [xhci_hcd]
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff810c02f9>] handle_irq_event_percpu+0x39/0x1b0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff810c04ba>] handle_irq_event+0x4a/0x70
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff810c3674>] handle_edge_irq+0x84/0x150
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff81006175>] handle_irq+0x25/0x40
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a96af>] do_IRQ+0x4f/0xe0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a77ab>] common_interrupt+0x6b/0x6b
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff810cc9b1>] ? rcu_process_callbacks+0x2d1/0x620
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff8106c52f>] __do_softirq+0xcf/0x290
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff8106c863>] irq_exit+0xa3/0xb0
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a9786>] smp_apic_timer_interrupt+0x46/0x60
Sep 03 15:57:57 linux-e3ncd5 kernel:  [<ffffffff816a7b2b>] apic_timer_interrupt+0x6b/0x70
Sep 03 15:57:57 linux-e3ncd5 kernel:  <EOI>
Sep 03 15:57:57 linux-e3ncd5 kernel: ---[ end trace b7460ecb686ada67 ]---
```

Trying to build the out-of-tree module also fails on 4.2 so I currently need to fallback to the out of tree module on 4.1.

```
# make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-4.2.0-1.gefc468a-obj/x86_64/desktop'
  CC [M]  /usr/local/src/mt7601u/init.o
/usr/local/src/mt7601u/init.c: In function ‘mt7601u_register_device’:
/usr/local/src/mt7601u/init.c:599:12: error: incompatible types when assigning to type ‘long unsigned int[1]’ from type ‘int’
  hw->flags = IEEE80211_HW_SIGNAL_DBM |
            ^
/usr/src/linux-4.2.0-1.gefc468a/scripts/Makefile.build:274: recipe for target '/usr/local/src/mt7601u/init.o' failed
make[4]: *** [/usr/local/src/mt7601u/init.o] Error 1
/usr/src/linux-4.2.0-1.gefc468a/Makefile:1407: recipe for target '_module_/usr/local/src/mt7601u' failed
make[3]: *** [_module_/usr/local/src/mt7601u] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make[1]: *** [__sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-4.2.0-1.gefc468a-obj/x86_64/desktop'
Makefile:8: recipe for target 'default' failed
make: *** [default] Error 2
```

mwprado, 2015-09-20:
I've  changed the code to:
hw->flags[0] = 

The code had compiled successful, but I cant connect on AP.

furlongm, 2016-02-19:
Fixed in mainline kernel.

closed 2016-02-19 by furlongm