kuba-moo/mt7601u #36: Linux 4.2: Error -2

SemperPeritus, 2015-09-08:
After installing 4.2 kernel wi-fi dongle didn't work.
Manjaro (Arch) Linux.
lsusb:
`Bus 002 Device 008: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter`
dmesg:
``` bash
[ 4788.551728] usb 2-2: new high-speed USB device number 8 using xhci_hcd
[ 4789.157195] usb 2-2: reset high-speed USB device number 8 using xhci_hcd
[ 4789.280946] mt7601u 2-2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[ 4789.286080] mt7601u 2-2:1.0: Direct firmware load for mt7601u.bin failed with error -2
[ 4789.286089] mt7601u 2-2:1.0: Falling back to user helper
```

kuba-moo, 2015-09-09:
@Nossiac any progress on upstreaming the firmware?

SemperPeritus, 2015-09-09:
Out of `ps aux`
```bash
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.0  0.0 182076  5040 ?        Ss   14:06   0:03 /sbin/init splash
root         2  0.0  0.0      0     0 ?        S    14:06   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        S    14:06   0:00 [ksoftirqd/0]
root         5  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/0:0H]
root         7  0.1  0.0      0     0 ?        S    14:06   0:23 [rcu_preempt]
root         8  0.0  0.0      0     0 ?        S    14:06   0:00 [rcu_sched]
root         9  0.0  0.0      0     0 ?        S    14:06   0:00 [rcu_bh]
root        10  0.0  0.0      0     0 ?        S    14:06   0:00 [migration/0]
root        11  0.0  0.0      0     0 ?        S    14:06   0:00 [watchdog/0]
root        12  0.0  0.0      0     0 ?        S    14:06   0:00 [watchdog/1]
root        13  0.0  0.0      0     0 ?        S    14:06   0:00 [migration/1]
root        14  0.0  0.0      0     0 ?        S    14:06   0:00 [ksoftirqd/1]
root        16  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/1:0H]
root        17  0.0  0.0      0     0 ?        S    14:06   0:00 [watchdog/2]
root        18  0.0  0.0      0     0 ?        S    14:06   0:00 [migration/2]
root        19  0.0  0.0      0     0 ?        S    14:06   0:00 [ksoftirqd/2]
root        21  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/2:0H]
root        22  0.0  0.0      0     0 ?        S    14:06   0:00 [watchdog/3]
root        23  0.0  0.0      0     0 ?        S    14:06   0:00 [migration/3]
root        24  0.0  0.0      0     0 ?        S    14:06   0:00 [ksoftirqd/3]
root        26  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/3:0H]
root        27  0.0  0.0      0     0 ?        S<   14:06   0:00 [khelper]
root        28  0.0  0.0      0     0 ?        S    14:06   0:00 [kdevtmpfs]
root        29  0.0  0.0      0     0 ?        S<   14:06   0:00 [netns]
root        30  0.0  0.0      0     0 ?        S<   14:06   0:00 [perf]
root        31  0.0  0.0      0     0 ?        S    14:06   0:00 [khungtaskd]
root        32  0.0  0.0      0     0 ?        S<   14:06   0:00 [writeback]
root        33  0.0  0.0      0     0 ?        SN   14:06   0:00 [ksmd]
root        34  0.0  0.0      0     0 ?        SN   14:06   0:01 [khugepaged]
root        35  0.0  0.0      0     0 ?        S<   14:06   0:00 [crypto]
root        36  0.0  0.0      0     0 ?        S<   14:06   0:00 [kintegrityd]
root        37  0.0  0.0      0     0 ?        S<   14:06   0:00 [bioset]
root        38  0.0  0.0      0     0 ?        S<   14:06   0:00 [kblockd]
root        39  0.0  0.0      0     0 ?        S<   14:06   0:00 [devfreq_wq]
root        41  0.0  0.0      0     0 ?        S    14:06   0:00 [kswapd0]
root        42  0.0  0.0      0     0 ?        S    14:06   0:00 [fsnotify_mark]
root        47  0.0  0.0      0     0 ?        S<   14:06   0:00 [kthrotld]
root        51  0.0  0.0      0     0 ?        S<   14:06   0:00 [ipv6_addrconf]
root        52  0.0  0.0      0     0 ?        S<   14:06   0:00 [deferwq]
root        84  0.0  0.0      0     0 ?        S    14:06   0:00 [irq/19-mmc0]
root        85  0.0  0.0      0     0 ?        S<   14:06   0:00 [ata_sff]
root        87  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_0]
root        88  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_0]
root        89  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_1]
root        90  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_1]
root        91  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_2]
root        92  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_2]
root        93  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_3]
root        94  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_3]
root        95  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_4]
root        96  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_4]
root        97  0.0  0.0      0     0 ?        S    14:06   0:00 [scsi_eh_5]
root        98  0.0  0.0      0     0 ?        S<   14:06   0:00 [scsi_tmf_5]
root       101  0.0  0.0      0     0 ?        S<   14:06   0:00 [rtsx_pci_sdmmc_]
root       118  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/0:1H]
root       160  0.0  0.0      0     0 ?        S    14:06   0:01 [jbd2/sda4-8]
root       161  0.0  0.0      0     0 ?        S<   14:06   0:00 [ext4-rsv-conver]
root       180  0.0  0.4  71720 35844 ?        Ss   14:06   0:01 /usr/lib/systemd/systemd-journald
root       184  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/2:1H]
root       201  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/3:1H]
root       207  0.0  0.0      0     0 ?        S<   14:06   0:00 [kworker/1:1H]
root       212  0.0  0.0      0     0 ?        S<   14:06   0:00 [iprt-VBoxWQueue]
root       213  0.0  0.0      0     0 ?        S    14:06   0:00 [iprt-VBoxTscThr]
root       230  0.0  0.0  36652  4856 ?        Ss   14:06   0:00 /usr/lib/systemd/systemd-udevd
root       233  0.0  0.0      0     0 ?        S<   14:06   0:00 [cfg80211]
root       254  0.0  0.0      0     0 ?        S<   14:06   0:00 [kmemstick]
root       256  0.0  0.0      0     0 ?        S<   14:06   0:00 [acpi_thermal_pm]
[... 99 lines cut ...]
[... cut ...]
platon    6928 11.0  2.7 1058592 224352 ?      Sl   19:27   0:04 /usr/lib/chromium/chromium --type=renderer --lang=ru --enable-offline-auto-reload --enable-offline-auto-reload-visible-only --enable-delegated-renderer --num-raster-threads=2 --gpu-rasterization-msaa-sample-count=8 --content-image-texture-target=3553 --video-image-texture-target=3553 --disable-accelerated-video-decode --channel=1172.108.600309441 --v8-natives-passed-by-fd --v8-snapshot-passed-by-fd
platon    6961  0.0  0.0  36440  3388 pts/0    R+   19:28   0:00 ps aux
```

SemperPeritus, 2015-09-10:
The issue has solved! Reason: wrong name of mt7601u.bin (mt7601.bin) by default.

closed 2015-09-10 by SemperPeritus