kuba-moo/mt7601u #37: Driver EEPROM-Failure

danishbelal, 2015-09-17:
Hi,
I installed this driver for my WiFi-Stick. It crashes with the following message:
```
[ 6955.573047] usb 1-1.5: new high-speed USB device number 4 using ehci-pci
[ 6955.762912] usb 1-1.5: reset high-speed USB device number 4 using ehci-pci
[ 6955.847673] mt7601u 1-1.5:1.0: ASIC revision: 76010001 MAC revision: 76010500
[ 6955.849573] mt7601u 1-1.5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[ 6956.194039] mt7601u 1-1.5:1.0: Error: your device needs default EEPROM file and this driver doesn't support it!
[ 6956.199984] mt7601u: probe of 1-1.5:1.0 failed with error -22
[ 6956.200027] usbcore: registered new interface driver mt7601u
```
This is my device:
```
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
```

kuba-moo, 2015-09-22:
Wow! You've got a weird device, my friend :S There is something wrong with EEPROM. Unfortunately I won't have time to investigate (I can't work on this driver ATM).

SandeshKa07, 2016-01-08:
I am using TL-WN727N wifi module which you supported. I am using it for beaglebone. When i clone it in beaglebone and run make , i got error like this

/bin/sh: ./scripts/recordmcount: cannot execute binary file
 how to solve it and make it working for beaglebone.

kuba-moo, 2016-01-09:
Please don't hijack issues. Your problem is not related to @danishbelal's problem so you should've opened a new issue.

The error you quote seems to indicate there is something wrong with how you built the kernel. Perhaps ./scripts/recordmcount was not cross compiled properly? Perhaps google has the answer I don't know off the top of my head, anyway, doesn't seem like problem with this driver.