kuba-moo/mt7601u #38: dmesg: Direct firmware load for mt7601u.bin failed with error -2

MarcoFalke, 2015-09-21:
[The readme](https://github.com/kuba-moo/mt7601u/blob/86547cfa033a362ef5c00c437016af4ac9dcdd8f/README.md) suggest this is already included in kernel 4.2. Yet, [this comment](https://bugzilla.redhat.com/show_bug.cgi?id=1264631#c1) on the redhat bug report suggest it is missing.

Is there anything I need to touch after a fresh install with kernel 4.2 to load the mt7601u driver?

kuba-moo, 2015-09-22:
You need the firmware. MediaTek didn't sent it upstream, yet :/

MarcoFalke, 2015-09-22:
I.e. I just try to install it the usual way?

kuba-moo, 2015-09-22:
Yep!

MarcoFalke, 2015-09-22:
Awesome, it works. Thank you!

```
git clone https://github.com/porjo/mt7601
cd mt7601/src/
make
sudo mv mcu/bin/MT7601.bin /lib/firmware/mt7601u.bin
```

closed 2015-09-22 by MarcoFalke

SizemoreSystem, 2026-08-28:
its gone,,