kuba-moo/mt7601u #40: Build fails fo kernel 4.2.

PikkuJose, 2015-10-09:


cosmicnag, 2015-10-09:
+1 same here for latest arch kernel

RohanPrabhuNomadly, 2015-11-26:
Does anyone have any update as to whether this builds or not for 4.3?

PikkuJose, 2015-11-26:
Didn't give it a try, as long as for any other reasons, I should stay in 
4.1 for now.

kuba-moo, 2015-11-27:
@RohanPrabhuNomadly 4.3 does not need this driver any more. The problem with 4.2 was that it was missing last few patches which fixed DMA issues, 4.3 already contains them, so it should work OK.

RohanPrabhuNomadly, 2015-11-28:
The problem is that in 4.3, whenever I use this driver in linux 4.2.5 or 4.3.1, my linux entirely freezes after a certain amount of data that is sent (I know this because if I don't open any applications which uses network data, then it doesn't freeze). But if I am sending/receiving data, then after a certain amount of time, it freezes and crashes invariably (rebooting is the only option, I cannot even switch to another vt, I cannot ssh/telnet into my machine). The only thing that worked for me was to downgrade to 4.1 (where this driver still compiles and back then I was using 4.2.1 I think) and then use it. Using a rolling release like arch has it's own series of problems when you downgrade selectively (especially if it is the kernel), and recently when I upgraded to 4.2.5, it freezes again. I have no idea where to even begin if I have to troubleshoot. All the logs seem fine, there is no kernel panic, just a plain ol' system freeze.

EchoUSB, 2015-12-19:
If you change that to the following, it will compile and work.
Not sure if this fixes kernel panics/freeze, but at least it compiles and works

hw->flags[0] =  IEEE80211_HW_PS_NULLFUNC_STACK |
		    IEEE80211_HW_SUPPORTS_HT_CCK_RATES |
#ifdef MAC80211_IS_PATCHED
		    IEEE80211_HW_TX_STATS_EVERY_MPDU |
#endif
		    0;

That's really quick and dirty because I have no idea what I'm doing, but it should be good

RohanPrabhuNomadly, 2016-01-09:
I didn't try the changes as mentioned in this thread, but updating to the 4.3 kernel fixed the system freeze issues for me. In one of the other issues (can't recall which) it is mentioned that it is fixed by a couple of commits esp relating to logging the rx/tx on the device which are there in 4.3 and not in 4.2. I was waiting for 4.3 to be rolled out on arch core, looks like it happened just a couple of days ago. After a system update, everything works fine.

kuba-moo, 2016-05-28:
Build on v4.2 should be now fixed!

closed 2016-05-28 by kuba-moo

iamashwin26, 2016-06-21:
Why am i still getting this error?