kuba-moo/mt7601u #42: MT7601u driver is corrupting system

trulio, 2015-11-08:
Hi everyone,
i'm using an MT7601u device since 2-3 month with the default  raspian driver.

After this time of intense daily usage (my system daily plays mopidy spotify several hours each day )

I noticed that the system is freezing, or partially frrezing,  in an undefined state on average once a day.

Curiously, when i tryed to observe the state of the frozen system

When i mentionned undefined, i mean different sates like:
-unable to reach sshd deamon while mopidy server keeps running OK
-or never seen warning message unseen before like access not permitted to some files/directories that does not exist
-...

Anyway, i think there is a bug in the driver that corrupt randomly the system after running afew hours.

Did anyone else had the same symptoms?

Trulio

kuba-moo, 2015-11-09:
What kernel version does raspbian use?

kuba-moo, 2016-02-15:
Closing this one since there is no response for 3 months.

closed 2016-02-15 by kuba-moo