kuba-moo/mt7601u #44: kernel 4.3.0 - ERROR Transfer event TRB DMA ...

baoyankun, 2015-12-25:
OS - debian testing
Device - 360 mini wifi gen 2 
uname -a:
Linux localhost 4.3.0-1-amd64 #1 SMP Debian 4.3.3-2 (2015-12-17) x86_64 GNU/Linux
lsusb:
(Bus 002 Device 003: ID 148f:760b Ralink Technology, Corp. MT7601U Wireless Adapter)

As soon as there is any network transfer, loads of warnings and errors flooded syslog:
Complete log, from plugging in to failure: http://paste.ubuntu.com/14204379/

```
Dec 25 13:37:46 localhost kernel: [  206.267363] xhci_hcd 0000:05:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
Dec 25 13:37:46 localhost kernel: [  206.267388] xhci_hcd 0000:05:00.0: Looking for event-dma 000000008401e4c0 trb-start 000000008401e4a0 trb-end 000000008401e4b0 seg-start 000000008401e000 seg-end 000000008401eff0
Dec 25 13:37:46 localhost kernel: [  206.267667] xhci_hcd 0000:05:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
Dec 25 13:37:46 localhost kernel: [  206.267680] xhci_hcd 0000:05:00.0: Looking for event-dma 000000008401e4d0 trb-start 000000008401e4a0 trb-end 000000008401e4b0 seg-start 000000008401e000 seg-end 000000008401eff0
Dec 25 13:37:46 localhost kernel: [  206.273352] xhci_hcd 0000:05:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
Dec 25 13:37:46 localhost kernel: [  206.273377] xhci_hcd 0000:05:00.0: Looking for event-dma 000000008401e4e0 trb-start 000000008401e4a0 trb-end 000000008401e4b0 seg-start 000000008401e000 seg-end 000000008401eff0
```
```
Dec 25 13:37:49 localhost kernel: [  209.434362] ieee80211 phy3: wlx24050f7bac46: No probe response from AP 88:25:93:5b:c7:47 after 500ms, disconnecting.
Dec 25 13:37:49 localhost kernel: [  209.794400] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:49 localhost kernel: [  210.094441] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.394485] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.694495] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.994570] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.994593] mt7601u 2-2:1.0: Error: mt7601u_mcu_wait_resp timed out
Dec 25 13:37:51 localhost kernel: [  211.494986] mt7601u 2-2:1.0: Error: send MCU cmd failed:-110
```
```
Dec 25 13:37:49 localhost kernel: [  209.434362] ieee80211 phy3: wlx24050f7bac46: No probe response from AP 88:25:93:5b:c7:47 after 500ms, disconnecting.
Dec 25 13:37:49 localhost kernel: [  209.794400] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:49 localhost kernel: [  210.094441] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.394485] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.694495] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.994570] mt7601u 2-2:1.0: Warning: mt7601u_mcu_wait_resp retrying
Dec 25 13:37:50 localhost kernel: [  210.994593] mt7601u 2-2:1.0: Error: mt7601u_mcu_wait_resp timed out
Dec 25 13:37:51 localhost kernel: [  211.494986] mt7601u 2-2:1.0: Error: send MCU cmd failed:-110
```

baoyankun, 2015-12-30:
Compiled my own 4.3.3 kernel and the problem is gone. Strange.
```
Linux debian 4.3.3 #1 SMP Tue Dec 29 07:18:47 UTC 2015 armv7l GNU/Linux
```
```
[  111.807285] usb 1-1: new high-speed USB device number 2 using ehci-platform
[  111.968831] usb 1-1: New USB device found, idVendor=148f, idProduct=760b
[  111.975889] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  111.983422] usb 1-1: Product: 802.11 n WLAN
[  111.987851] usb 1-1: Manufacturer: MediaTek
[  111.992247] usb 1-1: SerialNumber: 1.0
[  112.227260] usb 1-1: reset high-speed USB device number 2 using ehci-platform
[  112.380288] mt7601u 1-1:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  112.393023] mt7601u 1-1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  112.793671] mt7601u 1-1:1.0: EEPROM ver:0c fae:00
[  113.027347] usbcore: registered new interface driver mt7601u
```

fafactx, 2018-08-27:
it seems like me . just do as follow it works well for me on kali 2018.2 which kernel ver. >=4.15.*
I repaire this on my pull request 。 or you can just add follow code to you proj in mt7601u.h。
//just for kali 2018.2 kernel version >4.2
#define RX_FLAG_SHORTPRE (1<<8 )
#define RX_FLAG_HT_GF (1<<13)
#define RX_FLAG_HT (1<<9 )
#define RX_FLAG_SHORT_GI (1<<11)
#define RX_FLAG_40MHZ (1<<10)
#define RX_FLAG_STBC_SHIFT (26)
and commit this code in usb.c 128 line.
//mt7601u_vendor_request(dev, MT_VEND_DEV_MODE, USB_DIR_OUT,MT_VEND_DEV_MODE_RESET, 0, NULL, 0);

so
enjoy. but the drive have a bug , it is not so good,so if you want a short delay in ping echo. if i have time,i will repaire it ,later.

pedrombmachado, 2019-04-03:
The solution described here worked for me in Ubuntu 16.04 and 18.04:
https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-352250127

I will reference it here:

Originally written by ingate.
Thanks to aleksander and Nidroide.
Tested on Ubuntu 14.04 (kernel 4.4), Ubuntu 17.10 (kernel 4.13) and Ubuntu 18.04 (4.15.0-36-generic)

1. Download corresponding kernel source from kernel.org. For example: if you have 4.4.0-104-generic download version 4.4.
2. From archive unpack just folder drivers/net/wireless/mediatek/mt7601u
3. Edit phy.c. Find function mt7601u_init_cal and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0); like so:

// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
// if (ret)
// return ret;
// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->dpd_temp);
// if (ret)
// return ret;

Find function mt7601u_phy_recalibrate_after_assoc and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp); like so:

void mt7601u_phy_recalibrate_after_assoc(struct mt7601u_dev *dev)
{
// mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp);

    mt7601u_rxdc_cal(dev);
}

4. Build module: make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
5. Remove device
6. sudo rmmod mt7601u
7. sudo insmod ./mt7601u.ko
8. Insert device
9. Check there are no errors in dmesg and interface appeared in ip link, check connection stability.
10. To make change persistent till next kernel upgrade: backup original module and replace with compiled. To find out where is original module run modinfo mt7601u (view string filename: /lib/modules/_KERNEL_VERSION_/kernel/drivers/net/wireless/mediatek/mt7601u/mt7601u.ko).

closed 2021-02-15 by kuba-moo