kuba-moo/mt7601u #46: Kernel 4.2 hangs

BLaDZer, 2016-01-12:
Hello thanks for the driver!
I'm not really sure to write it here but you've wrote that your driver is merged in kernel 4.2 so I'll try :)

My system is Linux Mint 17.3 Rosa with
Linux localhost 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
here your driver works fine! really fine(with really rare associate bug).

But if I'm load 4.2.0-18 or 4.2.0-22 kernel (available in update manager) after some network traffic I'm getting system hang - nothing work(even Magic SysRq). And same everytime.
> Bus 001 Device 002: ID 148f:7601 Ralink Technology, Corp.

If I disconnect the usb dongle(TP-LINK TL-WN727N) system works fine.

Call trace and some other lines from kern.log in attachment

[mt7601_hang.zip](https://github.com/kuba-moo/mt7601u/files/86878/mt7601_hang.zip)

BLaDZer, 2016-01-12:
[mt7601u_syslog.zip](https://github.com/kuba-moo/mt7601u/files/86902/mt7601u_syslog.zip)

kuba-moo, 2016-01-12:
Ugh... damn.  v4.2 is a bit unlucky because I didn't make it on time with some fixes and the driver in v4.2 has issues, unfortunately :disappointed: Please try v4.3 if you can. I'll try to fix code in this repo to build on v4.2 over the weekend.

BLaDZer, 2016-01-13:
Thanks for the reply.
>>Please try v4.3 if you can
sry but it's not available in update manager and I'm not so good to compile own kernel on my work

>>I'll try to fix code in this repo to build on v4.2 over the weekend.
This will be great. I'll check your driver when it'll be ready

kuba-moo, 2016-05-28:
Build on v4.2 should be now fixed!

closed 2016-05-28 by kuba-moo