kuba-moo/mt7601u #49: Using built in raspbian driver, unable to connect to any networks (though they are visible)

oscarandjo, 2016-02-05:
Please see my thread here on reddi for diagnostic commands and configurations used: https://www.reddit.com/r/raspberry_pi/comments/4485g3/cannot_get_my_raspberry_pi_compatible_wifi/

I'm simply not able to connect to any networks, I can see them but they will not connect. I've tested it on my normal and guest network.

kuba-moo, 2016-02-15:
Strange. So you can use this adapter without problems on your PC but on RPi it only sees the networks but cannot connect?

Did you try building the driver from source on RPi or are you using the version which comes out of the box?

oscarandjo, 2016-02-15:
I’ve tried both and neither seem to work. I’ve got the latest Raspbain currently which is supposed to have a driver for this chipset installed yet it will not connect.
 
As you can see by that reddit thread, I’ve tried lots of things to fix it yet nothing seems to work. I bought it from the ebay listing [here](http://www.ebay.co.uk/itm/381313515950?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT)

kuba-moo, 2016-02-16:
> I’ve tried both and neither seem to work.

[1] Does this refer to the question about the driver version or RPi vs. PC?  

[2] AFAICT RPi uses kernel 4.1 now (please tell me output of `uname -a` command). 

[3] This driver has some bugs in 4.1 could you try to get 4.2, 4.3, 4.4 or 4.5 kernels on RRi?

[4] Did you try the vendor driver? You can download it from here:
https://github.com/porjo/mt7601
Please let me know that driver does work for you, we'll then try to figure out what is different and provide a fix to mainline Linux and RPi. Thanks.

oscarandjo, 2016-02-16:
1. "I tried both and neither seem to work" was in response to "Did you try building the driver from source on RPi or are you using the version which comes out of the box?"

2. Linux raspberrypi 4.1.17+ #834 Mon Feb 1 15:13:47 GMT 2016 armv6l GNU/Linux

3. I tried the vendor driver just now following the steps in an older version of it's readme. After running "make" It gave me the following error:

`make
make -C tools
make[1]: Entering directory '/home/pi/mt7601/src/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/home/pi/mt7601/src/tools'
/home/pi/mt7601/src/tools/bin2h
cp -f os/linux/Makefile.6 /home/pi/mt7601/src/os/linux/Makefile
make -C /lib/modules/4.1.17+/build SUBDIRS=/home/pi/mt7601/src/os/linux modules
make[1]: *** /lib/modules/4.1.17+/build: No such file or directory.  Stop.
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2`

kuba-moo, 2016-02-17:
[1], [2] Thanks.
[3] I think you need to install `linux-devel` or `linux-headers` or something like that with apt-get. Maybe there is more information on the old @porjo's README https://github.com/porjo/mt7601/tree/e2b39d7de75489b4682921012213ad1697b78ce2

closed 2016-04-20 by kuba-moo