kuba-moo/mt7601u #55: System hang @ arm 4.1.24

3s1d, 2016-07-14:
Hi

I am using this driver on an embedded arm platform (NXP/Freescale Colibri Vybrid VF50, 500Mhz 128MB Ram). Everything works fine for slow traffic. However, as the throughput increases (>10Mbit/s) and the data gets stored on the disk I keep getting the following logs. It does not happen when storing to /dev/null. I assume that I receive data faster than I can process/store and thus something weird happens? If so, the whole systems seems to stop for 1 or 2sec before it recovers...

Thanks
Juergen

[13690.072997] kworker/0:1: page allocation failure: order:3, mode:0x6120
[13690.079707] CPU: 0 PID: 5164 Comm: kworker/0:1 Tainted: G        W  O    4.1.24-00977-gc94f076 #9
[13690.088693] Hardware name: Freescale Vybrid VF5xx/VF6xx (Device Tree)
[13690.095214] Backtrace: 
[13690.097739] [<80012a54>] (dump_backtrace) from [<80012c74>] (show_stack+0x18/0x1c)
[13690.105383]  r7:00000000 r6:00000000 r5:00000003 r4:00006120
[13690.111141] [<80012c5c>] (show_stack) from [<80584a6c>] (dump_stack+0x24/0x28)
[13690.118450] [<80584a48>] (dump_stack) from [<8008c390>] (warn_alloc_failed+0xe4/0x120)
[13690.126466] [<8008c2b0>] (warn_alloc_failed) from [<8008ec48>] (__alloc_pages_nodemask+0x580/0x89c)
[13690.135571]  r3:00000000 r2:00000000
[13690.139191]  r6:00000030 r5:00000003 r4:00006120
[13690.144004] [<8008e6c8>] (__alloc_pages_nodemask) from [<7f0aef7c>] (mt7601u_rx_tasklet+0x114/0x46c [mt7601u])
[13690.154088]  r10:85cccb80 r9:80fe0000 r8:00000000 r7:854cef00 r6:0000061c r5:85cccf68
[13690.162055]  r4:80fe0000
[13690.164679] [<7f0aee68>] (mt7601u_rx_tasklet [mt7601u]) from [<80026610>] (tasklet_action+0x70/0xbc)
[13690.173875]  r10:40000004 r9:807dfc88 r8:00000006 r7:807dfc40 r6:807b4648 r5:00000000
[13690.181840]  r4:85cccee0
[13690.184417] [<800265a0>] (tasklet_action) from [<80026778>] (__do_softirq+0x11c/0x23c)
[13690.192391]  r7:00000100 r6:83298000 r5:807dfc98 r4:00000000
[13690.198143] [<8002665c>] (__do_softirq) from [<80026b2c>] (irq_exit+0xb8/0x100)
[13690.205512]  r10:807b5a88 r9:00000000 r8:87806000 r7:00000001 r6:00000000 r5:00000000
[13690.213480]  r4:807bf4d0
[13690.216054] [<80026a74>] (irq_exit) from [<8004cb04>] (__handle_domain_irq+0x60/0xb0)
[13690.223947]  r5:00000000 r4:807bf4d0
[13690.227611] [<8004caa4>] (__handle_domain_irq) from [<8000935c>] (gic_handle_irq+0x2c/0x5c)
[13690.236032]  r9:00000000 r8:85ff5300 r7:88002100 r6:83299e78 r5:807ae364 r4:8800210c
[13690.243944] [<80009330>] (gic_handle_irq) from [<80013700>] (__irq_svc+0x40/0x54)
[13690.251443] Exception stack(0x83299e78 to 0x83299ec0)
[13690.256557] 9e60:                                                       85cac980 85eb2600
[13690.264794] 9e80: 00000000 83298000 85cb6780 85cac980 00000001 85ff5300 85ff5300 00000000
[13690.273036] 9ea0: 807b5a88 83299edc 83299ee0 83299ec0 80585a4c 8003ef38 a0070013 ffffffff
[13690.281223]  r7:83299eac r6:ffffffff r5:a0070013 r4:8003ef38
[13690.287027] [<8003eee0>] (finish_task_switch) from [<80585a4c>] (__schedule+0x1c0/0x4bc)
[13690.295176]  r7:85ff5300 r6:807b6350 r5:85eb1c80 r4:85eb2600
[13690.300928] [<8058588c>] (__schedule) from [<80585d8c>] (schedule+0x44/0x9c)
[13690.308038]  r9:84619400 r8:00000008 r7:83298000 r6:807b5a98 r5:807b5a88 r4:83298000
[13690.315935] [<80585d48>] (schedule) from [<8003629c>] (worker_thread+0xc8/0x4b4)
[13690.323391]  r5:807b5a88 r4:807b5a88
[13690.327030] [<800361d4>] (worker_thread) from [<8003b208>] (kthread+0xdc/0xf4)
[13690.334323]  r10:00000000 r9:00000000 r8:00000000 r7:800361d4 r6:84619400 r5:85c58ec0
[13690.342294]  r4:00000000
[13690.344876] [<8003b12c>] (kthread) from [<8000f628>] (ret_from_fork+0x14/0x2c)
[13690.352161]  r7:00000000 r6:00000000 r5:8003b12c r4:85c58ec0
[13690.357894] Mem-Info:
[13690.360205] active_anon:3780 inactive_anon:4060 isolated_anon:0
[13690.360205]  active_file:7667 inactive_file:8799 isolated_file:0
[13690.360205]  unevictable:0 dirty:1427 writeback:0 unstable:0
[13690.360205]  slab_reclaimable:2176 slab_unreclaimable:1286
[13690.360205]  mapped:2468 shmem:3037 pagetables:121 bounce:0
[13690.360205]  free:475 free_pcp:39 free_cma:0
[13690.393169] Normal free:1900kB min:1380kB low:1724kB high:2068kB active_anon:15120kB inactive_anon:16240kB active_file:30668kB inactive_file:35196kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:129024kB managed:119824kB mlocked:0kB dirty:5708kB writeback:0kB mapped:9872kB shmem:12148kB slab_reclaimable:8704kB slab_unreclaimable:5144kB kernel_stack:504kB pagetables:484kB unstable:0kB bounce:0kB free_pcp:156kB local_pcp:156kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[13690.438226] lowmem_reserve[]: 0 0
[13690.441597] Normal: 143*4kB (UMR) 92*8kB (UEMR) 37*16kB (UMR) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 1900kB
[13690.454062] 19513 total pagecache pages
[13690.457916] 0 pages in swap cache
[13690.461239] Swap cache stats: add 0, delete 0, find 0/0
[13690.466519] Free swap  = 131068kB
[13690.469845] Total swap = 131068kB
[13690.473208] 32256 pages RAM
[13690.476015] 0 pages HighMem/MovableOnly
[13690.479851] 4294965500 pages reserved
[13690.483571] 4096 pages cma reserved

kuba-moo, 2016-07-15:
Your system runs out of memory, this driver is quite aggressive on memory usage.  Could you try setting 
`MT_USB_AGGR_SIZE_LIMIT` to `7`   on _mt7601u.h:37_
and
`MT_RX_ORDER` to `1`   on _mt7601u.h:39_
?

3s1d, 2016-07-15:
Works a treat! Thank you.

Right after my post I started playing around vm.min_free_kbyte. I had to reserve almost 32MB to get it working properly... -> not the right approach.

I would recommend to use this values a default, or at least make it optional for low memory devices.

NB:
Upon tt the first disconnect I always get this message:
[   23.811161] mt7601u 1-1:1.0: Error: submit RX URB failed:-19
[   23.817076] mt7601u 1-1:1.0: Error: submit RX URB failed:-19
[   23.835940] mt7601u 1-1:1.0: Warning: TX DMA did not stop!
[   27.858677] mt7601u 1-1:1.0: Warning: MAC TX did not stop!
[   27.978752] mt7601u 1-1:1.0: Warning: MAC RX did not stop!
[   27.984283] mt7601u 1-1:1.0: Warning: RX DMA did not stop!
But not on the second and any further disconnects. I'm note sure whether this is normal or if the same messages were already present before the mod (haven't checked). The driver seems to work perfectly fine.

Regards,
Juergen

closed 2016-07-15 by 3s1d