kuba-moo/mt7601u #57: TSSI upper saturation

vyasgiridhar, 2016-09-27:
Journalctl gives me this error while waiting for DHCP lease and I haven't checked this with all routers.

kuba-moo, 2016-10-13:
Hi!  Are you still seeing this?  I think the error is completely harmless but we should investigate a bit further.  Could you please post contents of the `eeprom_param` file from the DebugFS?  Could you check if `CONFIG_DYNAMIC_DEBUG` is set in your kernel and if so load the driver with `dyndbg=+pt` parameter to get more detailed log?

LooMaclin, 2017-08-05:
I'm get the error too:

```
[68150.440082] IPv6: ADDRCONF(NETDEV_UP): wlp0s20f0u10: link is not ready
[68150.482223] IPv6: ADDRCONF(NETDEV_UP): wlp0s20f0u10: link is not ready
[68152.730190] IPv6: ADDRCONF(NETDEV_UP): wlp0s20f0u10: link is not ready
[68153.872778] wlp0s20f0u10: authenticate with 84:16:f9:30:11:98
[68153.900887] wlp0s20f0u10: send auth to 84:16:f9:30:11:98 (try 1/3)
[68153.902962] wlp0s20f0u10: authenticated
[68153.905164] wlp0s20f0u10: associate with 84:16:f9:30:11:98 (try 1/3)
[68153.908876] wlp0s20f0u10: RX AssocResp from 84:16:f9:30:11:98 (capab=0x431 status=0 aid=1)
[68153.926720] wlp0s20f0u10: associated
[68153.926811] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f0u10: link becomes ready
[68157.917889] mt7601u 1-10:1.0: Error: TSSI upper saturation
[68161.970706] mt7601u 1-10:1.0: Error: TSSI upper saturation
[68170.077732] mt7601u 1-10:1.0: Error: TSSI upper saturation
[68174.131150] mt7601u 1-10:1.0: Error: TSSI upper saturation
[68178.184238] mt7601u 1-10:1.0: Error: TSSI upper saturation
[68182.240081] mt7601u 1-10:1.0: Error: TSSI upper saturation
```

liathit, 2017-08-06:
I get the error too. ):
```
[ 9577.891693] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9581.922783] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9585.954211] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9589.986255] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9594.017035] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9598.050906] mt7601u 3-8:1.0: Error: TSSI upper saturation
[ 9602.080775] mt7601u 3-8:1.0: Error: TSSI upper saturation
```

PhoenixStarEclipse, 2018-08-14:
I get this error too. ):
```
[  610.979518] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  615.012110] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  619.107686] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  623.204139] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  627.235593] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  631.269294] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  635.364863] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  639.396567] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  643.427892] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  647.461339] mt7601u 1-10:1.0: Error: TSSI upper saturation
[  651.493696] mt7601u 1-10:1.0: Error: TSSI upper saturation

```

teadom, 2019-08-19:
[ 8019.396775] [5326] mt7601u 3-6:1.0: tssi dc:0085 db:0551 hvga:0
[ 8019.396779] [5326] mt7601u 3-6:1.0: Power curr:000281de diff:00039cb1
[ 8019.396781] mt7601u 3-6:1.0: Error: TSSI upper saturation
[ 8019.396788] [5326] mt7601u 3-6:1.0: final diff: 00000000
[ 8023.428627] [5326] mt7601u 3-6:1.0: tx_rate:03 pwr:00021000
[ 8023.429137] [5326] mt7601u 3-6:1.0: tssi:7f t_power:00061e8f temp:fb pkt_type:e9
[ 8023.429142] [5326] mt7601u 3-6:1.0: tssi dc:0085 db:0551 hvga:0
[ 8023.429146] [5326] mt7601u 3-6:1.0: Power curr:000281de diff:00039cb1
[ 8023.429149] mt7601u 3-6:1.0: Error: TSSI upper saturation
[ 8023.429156] [5326] mt7601u 3-6:1.0: final diff: 00000000
[ 8027.460583] [5326] mt7601u 3-6:1.0: tx_rate:03 pwr:00021000
[ 8027.460967] [5326] mt7601u 3-6:1.0: tssi:7f t_power:00061e8f temp:fb pkt_type:e9
[ 8027.460973] [5326] mt7601u 3-6:1.0: tssi dc:0085 db:0551 hvga:0
[ 8027.460977] [5326] mt7601u 3-6:1.0: Power curr:000281de diff:00039cb1
[ 8027.460980] mt7601u 3-6:1.0: Error: TSSI upper saturation
[ 8027.460987] [5326] mt7601u 3-6:1.0: final diff: 00000000
[eeprom_param.txt](https://github.com/kuba-moo/mt7601u/files/3514785/eeprom_param.txt)