kuba-moo/mt7601u #60: iwconfig showing no P2P support

mosajjal, 2017-01-06:
Hi. 

using mt7601 driver provided inside 4.2 kernel and above, I don't see P2P functionality when running `iwconfig` command. is P2P supported by this chip? I've seen some forks on github saying P2P should be supported as the hardware is capable of it. 

any comments?

kuba-moo, 2017-01-06:
The driver doesn't support it. I think neither does the vendor Linux driver, or am I wrong?  The chip itself is probably capable of doing P2P if the driver is clever enough, but I have no plans of working on it, unfortunately (note: this driver doesn't even support AP mode).

mosajjal, 2017-01-06:
can you refer me to some (newbie) documentation on how to develop it? I might find some time to work on it.

kuba-moo, 2017-01-07:
I think you would have to read the kernel code and see how others do it.  Also I don't know if there is any documentation for the chip, at least I'm not aware of any.  Sorry, I don't have much to offer.

mosajjal, 2017-01-07:
ok thanks anyway. if I find something valid, I'll put it up here

closed 2021-02-15 by kuba-moo