kuba-moo/mt7601u #61: Error while install mt7601u driver

hpmuthu, 2017-01-22:
$ make && sudo make install && depmod
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-4.7.0-040700rc3-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.7.0-040700rc3-generic'
make -C /lib/modules/`uname -r`/build M=$PWD modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.7.0-040700rc3-generic'
  INSTALL /home/vittal/mt7601u/mt7601u.ko
At main.c:158:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.7.0-040700rc3-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.7.0-040700rc3-generic'

kuba-moo, 2017-01-25:
It looks like you have a signed kernel but lack tools to sign the modules.  Perhaps openssl package missing?

Please note, however, that the driver should already be included in Linux 4.7 and if it doesn't work then quite likely it's a different chip, not mt7601u.  Could you post output of ``lsusb``?

hpmuthu, 2017-01-25:
lsusb returns 

Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp.

I removed kernel 4.7.  I am using Ubuntu 16.04. Kernel version is 4.4.  I installed kuba-moo/mt7601u driver, still my wifi adapter is not working.

kuba-moo, 2017-01-26:
Could you post the output of ``dmesg -c`` after you plug in the card?  Perhaps you don't have the firmware file?

hpmuthu, 2017-01-26:
I wifi adapter working in Ubuntu 16.04 live cd without installing any drivers.  But in my pc(Ubuntu 16.04) wifi adapter is not working.  
[dmesg.txt](https://github.com/kuba-moo/mt7601u/files/731680/dmesg.txt)

kuba-moo, 2017-01-28:
You seem to have the vendor driver installed (mt7601 without the 'u'), please remove it.

hpmuthu, 2017-01-28:
Hi I reinstalled Ubuntu 16.04.  Now wifi adapter is working.

kuba-moo, 2017-01-29:
Good to hear that.  As I said in previous message you had the broken vendor driver installed and it took priority over mt7601u. I thought a bit about how to explain to you finding and removing the vendor driver, but it was hard without knowing how it got there in the first place.  I'm glad it's working after reinstallation! 👍

closed 2017-01-29 by kuba-moo