kuba-moo/mt7601u #64: driver mt7601 is unable to initialize

jrfabio, 2017-04-21:
Hi everyone, i'm trying to make it work but no sucess yet. Would you help me?
Folows the dmesg output...

[  187.755599] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  187.755604] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  187.759466] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  187.759841] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  187.759845] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  187.763725] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  187.764120] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  187.764124] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  187.767971] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  187.768338] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  187.768343] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  187.768346] mt7601u 8-3:1.0: Error: mt7601u_mcu_wait_resp timed out
[  187.988074] mt7601u 8-3:1.0: Vendor request req:07 off:0080 failed:-71
[  188.188075] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  188.388085] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  188.388139] mt7601u: probe of 8-3:1.0 failed with error -110
[  188.388534] usb 8-3: USB disconnect, device number 29
[  188.652025] usb 8-3: new high-speed USB device number 30 using ehci-pci
[  188.811416] usb 8-3: New USB device found, idVendor=148f, idProduct=7601
[  188.811420] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  188.811424] usb 8-3: Product: 802.11 n WLAN
[  188.811426] usb 8-3: Manufacturer: MediaTek
[  188.811429] usb 8-3: SerialNumber: 1.0
[  188.936096] usb 8-3: reset high-speed USB device number 30 using ehci-pci
[  189.087079] mt7601u 8-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  189.089398] mt7601u 8-3:1.0: firmware: direct-loading firmware mt7601u.bin
[  189.089406] mt7601u 8-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  189.494965] mt7601u 8-3:1.0: Warning: unsupported EEPROM version 0d
[  189.494970] mt7601u 8-3:1.0: EEPROM ver:0d fae:00
[  189.666916] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  189.667657] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  189.667665] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  189.671164] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  189.671667] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  189.671672] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  189.675409] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  189.675908] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  189.675913] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  189.679654] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  189.680228] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  189.680234] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  189.683910] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  189.684421] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  189.684426] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  189.684430] mt7601u 8-3:1.0: Error: mt7601u_mcu_wait_resp timed out
[  189.904074] mt7601u 8-3:1.0: Vendor request req:07 off:0080 failed:-71
[  190.104067] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  190.308074] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  190.308133] mt7601u: probe of 8-3:1.0 failed with error -110
[  190.308534] usb 8-3: USB disconnect, device number 30
[  190.572025] usb 8-3: new high-speed USB device number 31 using ehci-pci
[  190.731497] usb 8-3: New USB device found, idVendor=148f, idProduct=7601
[  190.731502] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  190.731505] usb 8-3: Product: 802.11 n WLAN
[  190.731508] usb 8-3: Manufacturer: MediaTek
[  190.731510] usb 8-3: SerialNumber: 1.0
[  190.856090] usb 8-3: reset high-speed USB device number 31 using ehci-pci
[  191.007039] mt7601u 8-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  191.008730] mt7601u 8-3:1.0: firmware: direct-loading firmware mt7601u.bin
[  191.008738] mt7601u 8-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  191.415052] mt7601u 8-3:1.0: Warning: unsupported EEPROM version 0d
[  191.415056] mt7601u 8-3:1.0: EEPROM ver:0d fae:00
[  191.591255] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  191.591450] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  191.591462] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  191.595502] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  191.595751] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[... 824 lines cut ...]
[  239.435598] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  239.435603] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  239.439341] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  239.439842] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  239.439845] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  239.443583] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  239.444125] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  239.444130] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  239.447852] mt7601u 8-3:1.0: Error: RX urb failed:-71
[  239.448350] mt7601u 8-3:1.0: Error: MCU resp urb failed:-71
[  239.448355] mt7601u 8-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  239.448358] mt7601u 8-3:1.0: Error: mt7601u_mcu_wait_resp timed out
[  239.668068] mt7601u 8-3:1.0: Vendor request req:07 off:0080 failed:-71
[  239.872078] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  240.076076] mt7601u 8-3:1.0: Vendor request req:02 off:0080 failed:-71
[  240.076131] mt7601u: probe of 8-3:1.0 failed with error -110
[  240.076532] usb 8-3: USB disconnect, device number 56
[  240.340083] usb 8-3: new high-speed USB device number 57 using ehci-pci
[  240.499483] usb 8-3: New USB device found, idVendor=148f, idProduct=7601
[  240.499486] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  240.499488] usb 8-3: Product: 802.11 n WLAN
[  240.499491] usb 8-3: Manufacturer: MediaTek
[  240.499492] usb 8-3: SerialNumber: 1.0
[  240.624030] usb 8-3: reset high-speed USB device number 57 using ehci-pci
^C

kuba-moo, 2017-04-22:
Hm.  Something seems wrong with the device at USB layer.  Does it work with any other driver/OS?

jrfabio, 2017-04-23:
Yes, it worked perfectly on windows 7.

kuba-moo, 2017-04-24:
Hm.  What is the platform (CPU)?  Where did you get the firmware from?  Is there chance you could try to compile the driver from Mediatek's website?

jrfabio, 2017-04-24:
I have a dell inspiron 1545, Pentium(R) Dual-core, 2,00 GB, Kali linux
light wich cames with the firmware in it but, as I didn´t know, i
downloaded it from github. Now about Mediatek´s website I don´t know
if I can LOL, but I can try.

jrfabio, 2017-04-24:
Oh I´m sorry because i forgot to say that at Mediatek´s website there
is no download option for my MT7601U, just Contact sales option.

2017-04-24 22:57 GMT, Fábio Santa Helena <mhagora@gmail.com>:

> I have a dell inspiron 1545, Pentium(R) Dual-core, 2,00 GB, Kali linux
> light wich cames with the firmware in it but, as I didn´t know, i
> downloaded it from github. Now about Mediatek´s website I don´t know
> if I can LOL, but I can try.
>
>
>
>> Hm.  What is the platform (CPU)?  Where did you get the firmware from?
>> Is
>> there chance you could try to compile the driver from Mediatek's website?
>>
>> https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-296839869
>
>
> *Lead, folow or get out of my way!*

kuba-moo, 2017-04-25:
Ah, I think they may have removed it after I upstreamed this driver...
You can find some copies on GH, like this one: https://github.com/porjo/mt7601u
As far as FW goes, can you re-download it to make sure it's not corrupted?  This is the MD5 sum it should have:
```
$ md5sum /lib/firmware/mt7601u.bin 
696cedb8e76ecc0cda9f9b0d3972c64d  /lib/firmware/mt7601u.bin
```

jrfabio, 2017-04-28:
Hi friend, i tried but got this message:

hammer@hammer:~/mt7601/src$ ls
ate                                  mcu
chips                                mgmt
common                               os
include                              phy
iwpriv_usage.txt                     rate_ctrl
Kismet-20140903-17-34-03-1.alert     README_STA_usb
Kismet-20140903-17-34-03-1.gpsxml    RT2870STACard.dat
Kismet-20140903-17-34-03-1.nettxt    RT2870STA.dat
Kismet-20140903-17-34-03-1.netxml    sta
Kismet-20140903-17-34-03-1.pcapdump  sta_ate_iwpriv_usage.txt
mac                                  tools
Makefile
hammer@hammer:~/mt7601/src$ sudo make
sudo: make: command not found
hammer@hammer:~/mt7601/src$

Any idea?

jrfabio, 2017-04-28:
More...

hammer@hammer:~/mt7601/src$ sudo make
make -C tools
make[1]: Entering directory '/home/hammer/mt7601/src/tools'
gcc -g bin2h.c -o bin2h
make[1]: Leaving directory '/home/hammer/mt7601/src/tools'
/home/hammer/mt7601/src/tools/bin2h
cp -f os/linux/Makefile.6 /home/hammer/mt7601/src/os/linux/Makefile
make -C /lib/modules/4.9.0-kali3-amd64/build
SUBDIRS=/home/hammer/mt7601/src/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-kali3-amd64'
  CC [M]  /home/hammer/mt7601/src/os/linux/../../os/linux/rt_profile.o
/home/hammer/mt7601/src/os/linux/../../os/linux/rt_profile.c: In
function ‘announce_802_3_packet’:
/home/hammer/mt7601/src/os/linux/../../os/linux/rt_profile.c:331:16:
warning: unused variable ‘pAd’ [-Wunused-variable]
  RTMP_ADAPTER *pAd = (RTMP_ADAPTER *)pAdSrc;
                ^~~
  CC [M]  /home/hammer/mt7601/src/os/linux/../../sta/assoc.o
  CC [M]  /home/hammer/mt7601/src/os/linux/../../sta/auth.o
  CC [M]  /home/hammer/mt7601/src/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/hammer/mt7601/src/os/linux/../../sta/sync.o
/home/hammer/mt7601/src/os/linux/../../sta/sync.c: In function
‘MlmeForceJoinReqAction’:
/home/hammer/mt7601/src/os/linux/../../sta/sync.c:181:24: warning:
unused variable ‘pInfo’ [-Wunused-variable]
  MLME_JOIN_REQ_STRUCT *pInfo = (MLME_JOIN_REQ_STRUCT *)(Elem->Msg);
                        ^~~~~
/home/hammer/mt7601/src/os/linux/../../sta/sync.c: In function ‘PeerBeacon’:
/home/hammer/mt7601/src/os/linux/../../sta/sync.c:2180:12: error:
passing argument 8 of ‘StaAddMacTableEntry’ from incompatible pointer
type [-Werror=incompatible-pointer-types]
            ie_list,
            ^~~~~~~
In file included from /home/hammer/mt7601/src/include/rt_config.h:59:0,
                 from /home/hammer/mt7601/src/os/linux/../../sta/sync.c:28:
/home/hammer/mt7601/src/include/rtmp.h:7892:9: note: expected
‘IE_LISTS * {aka struct _IE_lists *}’ but argument is of type
‘BCN_IE_LIST * {aka struct _bcn_ie_list *}’
 BOOLEAN StaAddMacTableEntry(
         ^~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.9.0-kali3-common/scripts/Makefile.build:298:
recipe for target '/home/hammer/mt7601/src/os/linux/../../sta/sync.o'
failed
make[4]: *** [/home/hammer/mt7601/src/os/linux/../../sta/sync.o] Error 1
/usr/src/linux-headers-4.9.0-kali3-common/Makefile:1507: recipe for
target '_module_/home/hammer/mt7601/src/os/linux' failed
make[3]: *** [_module_/home/hammer/mt7601/src/os/linux] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-kali3-amd64'
Makefile:394: recipe for target 'LINUX' failed
make: *** [LINUX] Error 2
hammer@hammer:~/mt7601/src$

2017-04-28 2:23 GMT, Fábio Santa Helena <mhagora@gmail.com>:

> Hi friend, i tried but got this message:
>
> hammer@hammer:~/mt7601/src$ ls
> ate                                  mcu
> chips                                mgmt
> common                               os
> include                              phy
> iwpriv_usage.txt                     rate_ctrl
> Kismet-20140903-17-34-03-1.alert     README_STA_usb
> Kismet-20140903-17-34-03-1.gpsxml    RT2870STACard.dat
> Kismet-20140903-17-34-03-1.nettxt    RT2870STA.dat
> Kismet-20140903-17-34-03-1.netxml    sta
> Kismet-20140903-17-34-03-1.pcapdump  sta_ate_iwpriv_usage.txt
> mac                                  tools
> Makefile
> hammer@hammer:~/mt7601/src$ sudo make
> sudo: make: command not found
> hammer@hammer:~/mt7601/src$
>
>
> Any idea?
>
>
>> Ah, I think they may have removed it after I upstreamed this driver...
>> You can find some copies on GH, like this one:
>> https://github.com/porjo/mt7601u
>> As far as FW goes, can you re-download it to make sure it's not
>> corrupted?
>> This is the MD5 sum it should have:
>> ```
>> $ md5sum /lib/firmware/mt7601u.bin
>> 696cedb8e76ecc0cda9f9b0d3972c64d  /lib/firmware/mt7601u.bin
>> ```
>>
>> https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-296892366
>
>
> *Lead, folow or get out of my way!*

kuba-moo, 2017-04-28:
Hm.  Looks like the vendor driver doesn't build for 4.9 kernel.
What about the firmware?  Could you show what `md5sum` the firmware you got has?

jrfabio, 2017-04-28:
hammer@hammer:~/mt7601$ md5sum
DPO_MT7601U_LinuxSTA_3.0.0.4_20130913-Linux-3.17.0-v2.patch
e79778e5f75407865d42c16b265ef250
DPO_MT7601U_LinuxSTA_3.0.0.4_20130913-Linux-3.17.0-v2.patch

kuba-moo, 2017-04-29:
I was asking about the `md5sum` of the firmware, so `/lib/firmware/mt7601u.bin`:
```
$ md5sum /lib/firmware/mt7601u.bin 
696cedb8e76ecc0cda9f9b0d3972c64d  /lib/firmware/mt7601u.bin
```

jrfabio, 2017-04-29:
Hi friend that's it

hammer@hammer:/lib/firmware$ md5sum mt7601u.bin
696cedb8e76ecc0cda9f9b0d3972c64d  mt7601u.bin
hammer@hammer:/lib/firmware$

kuba-moo, 2017-05-01:
Sorry, I run out of ideas :(  Either the card is broken, or the USB controller, or the two just don't like each other :(

jrfabio, 2017-05-01:
LOL.
Oh friend... what a pity, I'm so glad for your help. I think I'll have
to buy another one.
Thanks.

aleksander, 2017-06-07:
Hi!
I have exactly the same problem:
```
[  100.874586] usb 3-3: new high-speed USB device number 2 using xhci_hcd
[  101.468104] usb 3-3: reset high-speed USB device number 2 using xhci_hcd
[  101.600320] mt7601u 3-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  101.608217] mt7601u 3-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  102.012412] mt7601u 3-3:1.0: Warning: unsupported EEPROM version 0d
[  102.012414] mt7601u 3-3:1.0: EEPROM ver:0d fae:00
[  102.165074] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  102.165097] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  102.165100] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  102.165109] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  102.165154] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  102.165189] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  102.165205] usb 3-3: USB disconnect, device number 2
[  102.165210] mt7601u 3-3:1.0: Error: submit URB dir:128 ep:1 failed:-19
[  102.166009] usbcore: registered new interface driver mt7601u
[  102.444765] usb 3-3: new high-speed USB device number 3 using xhci_hcd
[  102.698367] usb 3-3: reset high-speed USB device number 3 using xhci_hcd
[  102.830737] mt7601u 3-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  102.831287] mt7601u 3-3:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  103.242023] mt7601u 3-3:1.0: Warning: unsupported EEPROM version 0d
[  103.242026] mt7601u 3-3:1.0: EEPROM ver:0d fae:00
[  103.395708] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395743] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395763] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  103.395767] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  103.395788] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395802] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  103.395805] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  103.395833] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395850] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  103.395853] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  103.395859] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395871] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  103.395874] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  103.395937] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395962] mt7601u 3-3:1.0: Error: RX urb failed:-71
[  103.395974] mt7601u 3-3:1.0: Error: MCU resp urb failed:-71
[  103.395977] mt7601u 3-3:1.0: Error: MCU resp evt:0 seq:4-3!
[  103.395980] mt7601u 3-3:1.0: Error: mt7601u_mcu_wait_resp timed out
[  103.531572] mt7601u 3-3:1.0: Vendor request req:07 off:0080 failed:-71
[  103.664926] mt7601u 3-3:1.0: Vendor request req:02 off:0080 failed:-71
[  103.798251] mt7601u 3-3:1.0: Vendor request req:02 off:0080 failed:-71
[  103.798294] mt7601u: probe of 3-3:1.0 failed with error -110
[  103.798472] usb 3-3: USB disconnect, device number 3
```
This error reproduced on two different desktops and one laptop all running latest Arch Linux. But It's working fine on windows 10.
```
~ » uname -a
Linux 4.11.3-1-ARCH #1 SMP PREEMPT Sun May 28 10:40:17 CEST 2017 x86_64 GNU/Linux
~ » md5sum /lib/firmware/mt7601u.bin 
696cedb8e76ecc0cda9f9b0d3972c64d  /lib/firmware/mt7601u.bin
```

How can I help debug this? Could you please give any directions where I can start the investigation?

aleksander, 2017-06-07:
The problem is in `mt7601u_phy_init` > `mt7601u_init_cal` > `mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0)`. With this lines commented initialization ends successfully and interface appears. Don't know is it working or not.

jrfabio, 2017-06-07:
Thanks friend but I gave up on mt7601.

aleksander, 2017-06-07:
@jrfabio, I want to solve this problem because I also have mt7601u based device with the same symptoms, so I believe @kuba-moo suggest some way :)

@kuba-moo, should I open another issue?

jrfabio, 2017-06-07:
I tried to solve this with him but we've got no sucess and he said he
has no idea what was happening but you can try maybe you get luck.<div
id="DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2"><br /> <table
style="border-top: 1px solid #D3D4DE;">
	<tr>
      <td style="width: 55px; padding-top: 18px;"><a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank"><img
src="https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif"
alt="" width="46" height="29" style="width: 46px; height: 29px;"
/></a></td>
		<td style="width: 470px; padding-top: 17px; color: #41424e;
font-size: 13px; font-family: Arial, Helvetica, sans-serif;
line-height: 18px;">Livre de vírus. <a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank" style="color: #4453ea;">www.avast.com</a>. 		</td>
	</tr>
</table>
<a href="#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2" width="1" height="1"></a></div>

kuba-moo, 2017-06-07:
Hi @aleksander!  
Thanks for investigating, did you check if the device works with calibration removed?
Now we have two bad devices, perhaps we can try to find commonalities.  What's your dongle Aleksander?  @jrfabio how about yours?

aleksander, 2017-06-08:
> did you check if the device works with calibration removed?

Yes, device works without calibration. I successfully connected to my phone and loaded some web sites. Not sure about speed and stability though. Need more testing.

> What's your dongle Aleksander?

Mine is Elisona USB Wi-Fi adapter ordered from aliexpress (https://aliexpress.com/item/150Mbps-Mini-Wireless-USB-Wlan-Adapter-802-11g-n-Wifi-Dongle-Lan-Card-w-Antenna-big/32527991067.html)

if this helps I can dissassemble the device to see whats in there.

I still can't find a cd-rom to check CD contents, maybe I'll find modified driver there.

jrfabio, 2017-06-08:
Hi everyone.
As I said, I gave up on my device and will buy another one which
matches with kali linux.
Thanks.<div id="DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2"><br /> <table
style="border-top: 1px solid #D3D4DE;">
	<tr>
      <td style="width: 55px; padding-top: 18px;"><a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank"><img
src="https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif"
alt="" width="46" height="29" style="width: 46px; height: 29px;"
/></a></td>
		<td style="width: 470px; padding-top: 17px; color: #41424e;
font-size: 13px; font-family: Arial, Helvetica, sans-serif;
line-height: 18px;">Livre de vírus. <a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank" style="color: #4453ea;">www.avast.com</a>. 		</td>
	</tr>
</table>
<a href="#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2" width="1" height="1"></a></div>

kuba-moo, 2017-06-09:
I think the chip reports to be the same version of silicon as all the others I've seen (well there are two variants but this is the more common one).  I will try to get one from AliExpress, although it will take a month to be delivered ;)
Definitely check out if there is a Linux driver on the CD.  The version of Linux driver I had from MediaTek was quite old, perhaps they changed something.

aleksander, 2017-06-15:
I found a cd-rom and cd contents is
```
├── MTK WlanU
│   ├── DPO_MT7601U_LinuxSTA_3.0.0.4_20130913.tar.bz2
└── RTL WlanU
    ├── RTL8188C_8192C_USB_linux_v4.0.2_9000.20130911.zip
```
Archives contain sources with timestamps of 9.12.2013.
Look like it's the same old mediatek driver.

jrfabio, 2017-06-15:
It´s like my one.
Did you got any success on mt-7601u?<div
id="DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2"><br /> <table
style="border-top: 1px solid #D3D4DE;">
	<tr>
      <td style="width: 55px; padding-top: 18px;"><a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank"><img
src="https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif"
alt="" width="46" height="29" style="width: 46px; height: 29px;"
/></a></td>
		<td style="width: 470px; padding-top: 17px; color: #41424e;
font-size: 13px; font-family: Arial, Helvetica, sans-serif;
line-height: 18px;">Livre de vírus. <a
href="https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail"
target="_blank" style="color: #4453ea;">www.avast.com</a>. 		</td>
	</tr>
</table>
<a href="#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2" width="1" height="1"></a></div>

kuba-moo, 2017-06-15:
3.0.0.4 is the version I was basing this driver on :(  One thing you could try to do is installing a very old kernel (~3.8 times probably) and see if the vendor driver works there.  For now I think you should be safe to hack the calibration out, I think it only affects sensitivity.

kuba-moo, 2017-06-23:
I just got my device from Alibaba and it works :/  Can you try your with `Fedora 25`, kernel `4.11.4-200.fc25.x86_64`?  That works fine for me..

kuba-moo, 2017-06-23:
FWIW my device has a white round sticker on the back which says `Tolvia`.

romeumellobr, 2017-06-26:
No, my device don't have this sticker, ia white round but says "AF"

romeumellobr, 2017-07-06:
Hello, Any news? I hope to get my device to work.

kuba-moo, 2017-07-06:
Damn. If Fedora 25 doesn't work for you I must have gotten a device from a different batch. I will try to inspect the calibration code over the weekend to see if there are any bugs I can spot. Stay tuned.

jguidini, 2017-07-14:
Hi @kuba-moo, I have the same problem. I buyed 3 devices, 2 work well, 1 not. Can I send to you this one diferent, if you hope, for you analyse and, hope, enhance the driver?

drbeat, 2017-08-25:
just bought one of those black & red devices off ebay and got the same issue on my raspberry pi

also the repository of porjo does not exists anymore so I do not have any clue how to fix this

aleksander, 2017-08-25:
@romeumellobr @jguidini @drbeat 
This is the temporary solution I use for my mt7601u stick from aliexpress:
1. From `kernel.org` dowload linux kernel source corresponding to your working kernel.
2. Unpack the kernel source. `cd` to `drivers/net/wireless/mediatek/mt7601u`
3. Open `phy.c`. Find function `mt7601u_init_cal` and comment out call `mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);` like so:
```
        //ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
        //if (ret)
        //      return ret;
```
4. Build module `make -C /lib/modules/$(uname -r)/build M=$(pwd) modules`
5. Remove the device
6. `rmmod mt7601u`
7. `insmod mt7601u.ko`
8. Insert the device
9. Check there are no errors in `dmesg` and interface appeared in `ip link`

stevewalsh123, 2017-08-25:
@drbeat
I also encountered a VERY similar error as you this morning with my MT6701 and rpi (1B)... 
I stumbled across found this thread whilst trying to figure out the cause... 

And for me the SOLUTION was to use a USB powered hub between pi and MT7601.
Turned out, it just wasn't getting enough power from the pi's own USB ports...
(I'm now also providing power to the pi from the same hub so it means I don't use any extra power sockets!)
https://s27.postimg.org/ul4rm32ab/pi1_USB-power_ralink-_MT7601.jpg

I hope this helps.

Safeuq, 2017-10-01:
@aleksander Can you explain it more clearly? As I'm running Ubuntu 17.04, I don't know what and which to download. I just googled the path and found a result on GitHub. I downloaded the mt7601u folder alone. On running the 'make' command, it throws an error. Please help.

mcquiggd, 2017-10-06:
@aleksander 

> This is the temporary solution I use for my mt7601u stick from aliexpress:
> 
> From kernel.org dowload linux kernel source corresponding to your working kernel.
> Unpack the kernel source. cd to drivers/net/wireless/mediatek/mt7601u
> Open phy.c. Find function mt7601u_init_cal and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0); like so:
>         //ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
>         //if (ret)
>         //      return ret;
> Build module make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
> Remove the device
> rmmod mt7601u
> insmod mt7601u.ko
> Insert the device
> Check there are no errors in dmesg and interface appeared in ip link

Thanks very much - this worked for me - I receive the same warning message in `dmesg`:

> "Warning unsupported EEPROM version 0d"

But the device works perfectly, and will happily max out the 25 Mbit ADSL I tested it with. It seems to be stable after shifting a few gigs through it, no drops, no problem with invalid packets, signal strength is as expected, and negotiation of network and security protocols works fine.

I do have one issue - the driver seems to be restored back to the kernel default on each reboot, and I have to run

`rmmod mt7601u`
`insmod ./mt7601u.ko`

from the directory where I compiled the updated driver, each time.

Perhaps you have encountered that? I presume it's because the driver is not 'signed', and it detects a 'tainted kernel' (I am pretty new to Linux).

(As an aside, during my attempts to get two mt7601u based devices to work, I upgraded my Ubuntu 17.04 kernel to 4.13.5, and suddenly the Atheros based internal wifi on my Dell Venue Pro 11 5130 sprang into life.)

@jrfabio I believe the workaround by @aleksander will solve your original problem, if you are still looking for a fix. 

I am living in Brasil, and mt7601u cards are _everywhere_... getting this adjustment to the driver into the kernel will make a lot of people happy.

Nidroide, 2017-11-02:
The same problem here, to get the dongle running I had to comment the following lines too:
```
//ret = mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->dpd_temp);
        //if (ret)
        //      return ret;
```
and:
`//mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp);`
in function mt7601u_phy_recalibrate_after_assoc

hoping this may help.

khfeng, 2017-11-08:
Does working mt7601u also have same ASIC and MAC revision, "ASIC revision: 76010001 MAC revision: 76010500"?

We can probably workaround this by checking the revision.

mcquiggd, 2017-11-08:
@khfeng 

Perhaps your comment was not directly addressed to me, but in my case, my card that now works, with the workaround of commenting out `mt7601u_mcu_calibrate` is indeed "ASIC revision: 76010001 MAC revision: 76010500". 

Below is the output as reported by `dmesg`, currently running on Ubuntu 17.10, and I have successfully made and used the driver with kernels 4.13 through 14.3.10, inclusive.

It`s been reliable and good performance with IPv4 and IPv6,

[   56.482127] mt7601u: probe of 1-2:1.0 failed with error -110
[   56.482418] usb 1-2: USB disconnect, device number 29
[   56.749085] usb 1-2: new high-speed USB device number 30 using xhci_hcd
[   56.899398] usb 1-2: New USB device found, idVendor=148f, idProduct=7601
[   56.899403] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   56.899406] usb 1-2: Product: 802.11 n WLAN
[   56.899408] usb 1-2: Manufacturer: MediaTek
[   56.899410] usb 1-2: SerialNumber: 1.0
[   57.016406] usb 1-2: reset high-speed USB device number 30 using xhci_hcd
[   57.158266] mt7601u 1-2:1.0: **ASIC revision: 76010001 MAC revision: 76010500**
[   57.158911] mt7601u 1-2:1.0: **Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____**
[   57.601782] mt7601u 1-2:1.0: Warning: unsupported EEPROM version 0d
[   57.601787] mt7601u 1-2:1.0: EEPROM ver:0d fae:00
[   57.797394] ieee80211 phy28: Selected rate control algorithm 'minstrel_ht'
[   57.931955] mt7601u 1-2:1.0 wlx20e11701dfa3: renamed from wlan1
[   57.986596] IPv6: ADDRCONF(NETDEV_UP): wlx20e11701dfa3: link is not ready
[   58.006540] IPv6: ADDRCONF(NETDEV_UP): wlx20e11701dfa3: link is not ready
[   58.077819] IPv6: ADDRCONF(NETDEV_UP): wlx20e11701dfa3: link is not ready
[   60.158190] wlx20e11701dfa3: authenticate with d4:63:fe:3e:44:46
[   60.181602] wlx20e11701dfa3: send auth to d4:63:fe:3e:44:46 (try 1/3)
[   60.187345] wlx20e11701dfa3: authenticated
[   60.195654] wlx20e11701dfa3: associate with d4:63:fe:3e:44:46 (try 1/3)
[   60.199338] wlx20e11701dfa3: RX AssocResp from d4:63:fe:3e:44:46 (capab=0x1411 status=0 aid=4)
[   60.229473] wlx20e11701dfa3: associated
[   60.229584] IPv6: ADDRCONF(NETDEV_CHANGE): wlx20e11701dfa3: link becomes ready
[   60.313844] wlx20e11701dfa3: Limiting TX power to 21 (21 - 0) dBm as advertised by d4:63:fe:3e:44:46

khfeng, 2017-11-08:
@mcquiggd 
I'd like to know the revision from devices that can work without the workaround. Apparently some owners here have such device.

mcquiggd, 2017-11-09:
@khfeng 

Understood.

... As an aside, I did have another card that would **not** work with the workaround - it did have the same revision we are discussing, but did not report any Device strings or Manufacturer string at all. It was the 'unbranded with red logo and antenna' that seems so common - the same as mentioned by aleksander [above](https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-306724936); note that his also does not report the device strings / manufacturer.

I tried two of them, shipped from different cities, same behaviour, neither worked with 'the fix'. Bought a different card without an antenna, that reports the device strings and manufacturer, and the workaround is successful. All the ASIC revisions, Mac Revisions, Firmware versions, build times are identical._

khfeng, 2017-11-13:
Sorry for using mail instead of GitHub issue.

Somehow your message doesn't show up in GitHub.

> @khfeng <https://github.com/khfeng>
>
> Understood.
>
> ... As an aside, I did have another card that would *not* work with the
> workaround - it did have the same revision we are discussing, but did not [...]
Does anyone contact the mt7601u vendor? I believe it's MediaTek.

> I tried two of them, shipped from different cities, same behaviour,
> neither worked with 'the fix'. Bought a different card without an antenna,
> that reports the device strings and manufacturer, and [...]
If we can't differentiate at all, we can't upstream the workaround.

kuba-moo, 2017-11-13:
@khfeng FWIW I think even if someone did already try to contact MTK, it's definitely worth contacting them again.  The more people write to them, the more likely they are to care.

mrgromov, 2017-12-17:
**My nonane black-red 148f:7601 mt7601u mini stick is working now!
Tested on Ubuntu 14.04 (kernel 4.4) and Ubuntu 17.10 (kernel 4.13).
Thanks to aleksander and Nidroide!
Just want to clarify instruction:**
1. Download corresponding kernel source from kernel.org.  For example: if you have `4.4.0-104-generic` download version 4.4.
2. From archive unpack just folder `drivers/net/wireless/mediatek/mt7601u`
3. Edit phy.c. Find function `mt7601u_init_cal` and comment out call `mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);` like so:
```
//  ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
//  if (ret)
//      return ret;
//  ret = mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->dpd_temp);
//  if (ret)
//      return ret;
```
Find function ` mt7601u_phy_recalibrate_after_assoc ` and comment out call ` mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp); ` like so:

```
void mt7601u_phy_recalibrate_after_assoc(struct mt7601u_dev *dev)
{
//  mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp);

    mt7601u_rxdc_cal(dev);
}
```

4. Build module: ` make -C /lib/modules/$(uname -r)/build M=$(pwd) modules `
5. Remove device
6. `sudo rmmod mt7601u`
7. `sudo insmod ./mt7601u.ko`
8. Insert device
9. Check there are no errors in dmesg and interface appeared in ip link, check connection stability.
10. To make change _persistent_ **till next kernel upgrade**: backup original module and replace with compiled. To find out where is original module run `modinfo mt7601u` (view string `filename:       /lib/modules/_KERNEL_VERSION_/kernel/drivers/net/wireless/mediatek/mt7601u/mt7601u.ko`).

Plodstack, 2017-12-23:
@mrgromov You are a stud and a king amongst men. I've been smashing my head against the wall for hours on this. Your fix plus a powered USB hub on my Pi 1 fixed this.

ranjan-purbey, 2018-01-28:
@mrgromov Is AP mode working? On my system, `iw list` only shows `managed` and `monitor` modes.

mrgromov, 2018-01-28:
On my system its same, only managed and monitor modes.

stralex7, 2018-01-31:
I have TP-Link TL-WN727N device (148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter) that works on ubuntu 4.4.0-109-generic kernel, but doesn't work on 4.10.0-30-generic even with @mrgromov patch applied.

I keep getting `mt7601u 1-1:1.0: Vendor request req:07 off:1000 failed:-110` messages. Any idea?

kuba-moo, 2018-01-31:
Any chance you could try 4.8 and 4.9 kernels?  There is pretty much a single commit added in each of these releases if you could try both we will find out what broke things for you.

stralex7, 2018-02-01:
@kuba-moo actually my assumption that this is related to kernel version is kind of wrong as after i have upgraded linux-firmware package it stopped working for me immediately. I will try to rollback and re-test to provide a working kernel/firmware package combination at least for ubuntu

stralex7, 2018-02-01:
ok, it works on ubuntu 16.04.3 kernel 4.4.0-109-generic, linux-firmware-1.157.14

`[ 5116.353758] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[ 5116.706060] usb 3-2: New USB device found, idVendor=148f, idProduct=7601
[ 5116.706066] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5116.706069] usb 3-2: Product: 802.11 n WLAN
[ 5116.706072] usb 3-2: Manufacturer: MediaTek
[ 5116.706074] usb 3-2: SerialNumber: 1.0
[ 5117.085473] usb 3-2: reset high-speed USB device number 6 using xhci_hcd
[ 5122.441736] usb 3-2: device descriptor read/64, error -110
[ 5137.898116] usb 3-2: device descriptor read/64, error -110
[ 5138.346331] usb 3-2: reset high-speed USB device number 6 using xhci_hcd
[ 5143.690606] usb 3-2: device descriptor read/64, error -110
[ 5159.139115] usb 3-2: device descriptor read/64, error -110
[ 5159.587049] usb 3-2: reset high-speed USB device number 6 using xhci_hcd
[ 5159.655267] mt7601u 3-2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[ 5159.666963] mt7601u 3-2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[ 5160.263378] mt7601u 3-2:1.0: Warning: unsupported EEPROM version 0d
[ 5160.263388] mt7601u 3-2:1.0: EEPROM ver:0d fae:00
[ 5160.271159] mt7601u 3-2:1.0: EEPROM country region 01 (channels 1-13)
[ 5160.887849] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[ 5162.043300] mt7601u 3-2:1.0 wlxc025e917475f: renamed from wlan0
[ 5162.089295] IPv6: ADDRCONF(NETDEV_UP): wlxc025e917475f: link is not ready
[ 5162.227533] IPv6: ADDRCONF(NETDEV_UP): wlxc025e917475f: link is not ready
[ 5162.284165] IPv6: ADDRCONF(NETDEV_UP): wlxc025e917475f: link is not ready
[ 5165.240766] wlxc025e917475f: authenticate with 64:31:50:f0:60:35
[ 5165.356570] wlxc025e917475f: send auth to 64:31:50:f0:60:35 (try 1/3)
[ 5165.363218] wlxc025e917475f: authenticated
[ 5165.371722] wlxc025e917475f: associate with 64:31:50:f0:60:35 (try 1/3)
[ 5165.412753] wlxc025e917475f: RX AssocResp from 64:31:50:f0:60:35 (capab=0xc31 status=0 aid=3)
[ 5165.532339] wlxc025e917475f: associated
[ 5165.532355] IPv6: ADDRCONF(NETDEV_CHANGE): wlxc025e917475f: link becomes ready`

This is the original module that comes with this kernel.

stralex7, 2018-02-01:
ok, i tried a different ubuntu 16.04.3 with kernel/firmware as I described above and it doesn't work, so please disregard the statement above.

gyunaev, 2018-02-14:
I just received a batch of 10 of those, and the symptoms are the same. Same error message, same fix.

Same info:

ASIC revision: 76010001 MAC revision: 76010500
Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
EEPROM ver:0c fae:00

Same fix. I can spare one of those dongles with a developer if needed.

DS28, 2018-02-18:
It works! Thank you!
Noname black-red 148f:7601 mt7601u on Manjaro 4.14.

My adds to mrgromov's instruction:
0. update kernel if it possible and install linux-headers.

meudayhegde, 2018-04-12:
I had the same issue on my rpi
Im new to linux, i couldn't compile 
Im getting this error while make

make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/lib/modules/4.14.30-v7+/build'
make: *** No rule to make target 'modules'.  Stop.
make: Leaving directory '/lib/modules/4.14.30-v7+/build'

meudayhegde, 2018-04-12:
Even i have the same issue as @jeffreyantony 
I managed to install linux-headers for my kernel version-4.14.30-v7+

I got these errors while make
make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/lib/modules/4.14.30-v7+/build'
make: *** No rule to make target 'modules'.  Stop.
make: Leaving directory '/lib/modules/4.14.30-v7+/build'

How could i fix this..?

ice-simx, 2018-04-16:
I was facing similar issue with mt7601u on VirtualBox with LUbuntu 16.04 kernel 4.4.0-45-generic. Doing following solves the issue:
1) sudo sed -i 's/wifi.powersave = 3/wifi.powersave = 2/' /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf
2) using USB 2.0 instead of USB 1.0 controller on virtualbox. For this I need to install virtualbox extension pack.

rserroyen, 2018-08-01:
Hello

I have a Raspberry Pi rev b `Natacha` Linux 4.14.50+ # 1122 Tue Jun 19 12:21:21 BST 2018 GNU / Linux `armv6l` and a wifi dongle `Bus` 001 Device 046: 148f ID: 7601 Ralink Technology, Corp. MT7601U Wireless `Adapter`

I tried the solution of mrgromov but I am blocked the makefile which module introduce ?? in which directory to execute the Makefile ??

help me

thank you very much

fafactx, 2018-08-27:
which version of kernel you use? if >=4.2 just do as follow ,it works for me. just try it.
I repaire this on my pull request 。 or you can just add follow code to you proj in mt7601u.h。
//just for kali 2018.2 kernel version >4.2
#define RX_FLAG_SHORTPRE (1<<8 )
#define RX_FLAG_HT_GF (1<<13)
#define RX_FLAG_HT (1<<9 )
#define RX_FLAG_SHORT_GI (1<<11)
#define RX_FLAG_40MHZ (1<<10)
#define RX_FLAG_STBC_SHIFT (26)
and commit this code in usb.c 128 line.
//mt7601u_vendor_request(dev, MT_VEND_DEV_MODE, USB_DIR_OUT,MT_VEND_DEV_MODE_RESET, 0, NULL, 0);

so
enjoy. but the drive have a bug , it is not so good,so if you want a short delay in ping echo. if i have time,i will repaire it ,later.

[40 more replies]

closed 2025-11-03 by kuba-moo