kuba-moo/mt7601u #66: Driver/Firmware Broken in newers Kernels

romeumellobr, 2017-06-19:
Hi Friends,
I tried to compile this driver in all well known distros (Mint, Debian, Ubuntu and Manjaro), in all of them I've got "Makefile:288 recipe for target 'LINUX' failed.

I can't get MT7601u work in this newer version, so I have a lof of these adapters, I can't give up on them.

Thank you!

kuba-moo, 2017-06-20:
As state in the `README` file, you don't need to compile the driver separately if your kernel version (as reported by `uname -r`) is `4.2` or newer.  This exact driver is included in the kernel.

romeumellobr, 2017-06-20:
Thank you but does not work in 4.4+ kernels, I tried all of them. Include Fedora now. So is broken.
I'm trying to get some help from their foruns, but I will try to install in an older vesion.

kuba-moo, 2017-06-20:
Do you see anything of note in the kernel logs when you plug the device in?

jeremyb31, 2017-06-20:
There are some dmesg errors posted at https://forums.linuxmint.com/viewtopic.php?f=49&t=247811&p=1331530#p1331394

romeumellobr, 2017-06-20:
Here the Kernel logs:
https://forums.linuxmint.com/viewtopic.php?f=49&t=247811&p=1331547#p1331547

kuba-moo, 2017-06-20:
What kind of dongle is that?  Could you post a link to where you bought them or a photo?

romeumellobr, 2017-06-20:
It's like this:
https://va3paw.files.wordpress.com/2014/03/7601.jpg

kuba-moo, 2017-06-20:
OK, so it's one of those $2-3 ones :)  I think this is a duplicated of bug #64.  It appears that whoever is making those cheap dongles have released a new revision which doesn't work well with this driver.  You can try the workaround mentioned in bug #64 or wait until I get my hand on one of those dongles (delivery from China takes a bit...). It's not related to the kernel version, then.

closed 2017-06-20 by kuba-moo