kuba-moo/mt7601u #70: Error While Compiling on Arch Linux Kernel 4.13.2-1

Biprodeep, 2017-09-21:
❯ make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/4.13.2-1-MANJARO/build'
  AR      /home/gh0st/git/mt7601u/built-in.o
  CC [M]  /home/gh0st/git/mt7601u/usb.o
  CC [M]  /home/gh0st/git/mt7601u/init.o
  CC [M]  /home/gh0st/git/mt7601u/main.o
  CC [M]  /home/gh0st/git/mt7601u/mcu.o
  CC [M]  /home/gh0st/git/mt7601u/trace.o
  CC [M]  /home/gh0st/git/mt7601u/dma.o
  CC [M]  /home/gh0st/git/mt7601u/core.o
  CC [M]  /home/gh0st/git/mt7601u/eeprom.o
  CC [M]  /home/gh0st/git/mt7601u/phy.o
  CC [M]  /home/gh0st/git/mt7601u/mac.o
/home/gh0st/git/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/gh0st/git/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
    status->flag |= RX_FLAG_SHORTPRE;
                    ^~~~~~~~~~~~~~~~
                    RX_ENC_FLAG_SHORTPRE
/home/gh0st/git/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/gh0st/git/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT_GF;
                   ^~~~~~~~~~~~~
                   RX_ENC_FLAG_HT_GF
/home/gh0st/git/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function); did you mean ‘RX_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT;
                   ^~~~~~~~~~
                   RX_FLAG_HT_GF
/home/gh0st/git/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function); did you mean ‘RX_FLAG_SHORTPRE’?
   status->flag |= RX_FLAG_SHORT_GI;
                   ^~~~~~~~~~~~~~~~
                   RX_FLAG_SHORTPRE
/home/gh0st/git/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_STBC_SHIFT’?
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^~~~~~~~~~~~~~~~~~
                        RX_ENC_FLAG_STBC_SHIFT
/home/gh0st/git/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function); did you mean ‘RX_FLAG_HT’?
   status->flag |= RX_FLAG_40MHZ;
                   ^~~~~~~~~~~~~
                   RX_FLAG_HT
make[2]: *** [scripts/Makefile.build:303: /home/gh0st/git/mt7601u/mac.o] Error 1
make[1]: *** [Makefile:1512: _module_/home/gh0st/git/mt7601u] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.13.2-1-MANJARO/build'
make: *** [Makefile:8: default] Error 2

OliverLew, 2017-09-28:
@Biprodeep I don't think you have to do this on latest archlinux, linux kernel after 4.2 already have and enabled the driver for mt7601u. I tested today, just plugged in the Xiaomi wifi dongle, and it is working, hadn't done anything else.

Goudacraft, 2017-09-28:
I'm having the same issue when compiling on Kali Linux !
If anyone has an idea on why it does that and what can be done to fix it, feel free to let us know!

kuba-moo, 2017-09-28:
Why not just use the in-tree driver?

mcquiggd, 2017-10-05:
@kuba-moo 

I have the identical problem attempting to compile on Ubuntu 17.04.

The in-tree driver errors:

sudo modprobe mt7601u && dmesg | grep mt76

```
[    4.635853] mt7601u 1-2.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[    4.652271] mt7601u 1-2.2:1.0: Error: your device needs default EEPROM file and this driver doesn't support it!
[    4.654294] mt7601u: probe of 1-2.2:1.0 failed with error -22
[    4.654343] usbcore: registered new interface driver mt7601u
[  404.540770] mt7601u 1-2.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  404.557904] mt7601u 1-2.2:1.0: Error: your device needs default EEPROM file and this driver doesn't support it!
[  404.560556] mt7601u: probe of 1-2.2:1.0 failed with error -22
[ 3742.145310] mt7601u 1-2.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
[ 3742.173255] mt7601u 1-2.2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[ 3742.575939] mt7601u 1-2.2:1.0: Error: your device needs default EEPROM file and this driver doesn't support it!
[ 3742.577205] mt7601u: probe of 1-2.2:1.0 failed with error -22

```

There is an open issue about this in this repo... https://github.com/kuba-moo/mt7601u/issues/37

xdujiang, 2018-06-15:
The same problem here on RaspberryPi2...
I also want to use the in-tree driver, but I have a mt7601 with ID: "2955:1002", which cannot be recognized, so I have to compile the driver by my self...

fafactx, 2018-08-27:
I repaire this on my pull request 。 or you can just add follow code to you proj in mt7601u.h。
//just for kali 2018.2 kernel version >4.2
#define RX_FLAG_SHORTPRE   (1<<8 )
#define RX_FLAG_HT_GF      (1<<13)
#define RX_FLAG_HT         (1<<9 )
#define RX_FLAG_SHORT_GI   (1<<11)
#define RX_FLAG_40MHZ      (1<<10)
#define RX_FLAG_STBC_SHIFT (26)
and  commit this code in usb.c 128 line.   
//mt7601u_vendor_request(dev, MT_VEND_DEV_MODE, USB_DIR_OUT,MT_VEND_DEV_MODE_RESET, 0, NULL, 0);

so
enjoy. but the drive have a bug , it is not so good,so if you want a short delay in ping echo. if i have time,i will repaire it ,later.

Phoeniwx, 2018-10-03:
> I repaire this on my pull request 。 or you can just add follow code to you proj in mt7601u.h。
> //just for kali 2018.2 kernel version >4.2
> #define RX_FLAG_SHORTPRE (1<<8 )
> #define RX_FLAG_HT_GF [...]

@fafactx  Your solution is the closest one I can find, but it seems like not working on Kali 4.18 and tplink wn726n. Would you please help me solve this out?

afjer, 2018-10-15:
@Phoeniwx

> @fafactx Your solution is the closest one I can find, but it seems like not working on Kali 4.18 and tplink wn726n. Would you please help me solve this out?

Are you solve this problem? what can be done to fix it? i have the same problem

pedrombmachado, 2019-04-03:
The solution described here worked for me in Ubuntu 16.04 and 18.04:
https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-352250127

I will reference it here:

Originally written by ingate.
Thanks to aleksander and Nidroide.
Tested on Ubuntu 14.04 (kernel 4.4), Ubuntu 17.10 (kernel 4.13) and Ubuntu 18.04 (4.15.0-36-generic)

1. Download corresponding kernel source from kernel.org. For example: if you have 4.4.0-104-generic download version 4.4.
2. From archive unpack just folder drivers/net/wireless/mediatek/mt7601u
3. Edit phy.c. Find function mt7601u_init_cal and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0); like so:

// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
// if (ret)
// return ret;
// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->dpd_temp);
// if (ret)
// return ret;

Find function mt7601u_phy_recalibrate_after_assoc and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp); like so:

void mt7601u_phy_recalibrate_after_assoc(struct mt7601u_dev *dev)
{
// mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp);

    mt7601u_rxdc_cal(dev);
}

4. Build module: make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
5. Remove device
6. sudo rmmod mt7601u
7. sudo insmod ./mt7601u.ko
8. Insert device
9. Check there are no errors in dmesg and interface appeared in ip link, check connection stability.
10. To make change persistent till next kernel upgrade: backup original module and replace with compiled. To find out where is original module run modinfo mt7601u (view string filename: /lib/modules/_KERNEL_VERSION_/kernel/drivers/net/wireless/mediatek/mt7601u/mt7601u.ko).

YSZL, 2019-07-17:
This answer can resolve this issure.
https://github.com/kuba-moo/mt7601u/issues/75#issuecomment-399734885

AlexMavpa, 2020-12-18:
Hello, i have the same problem (building on raspbian). I have done modifications in my _/home/pi/mt7601u/phy.c_ file, cleaned the make cache and tried to run make command again - but still getting :

```
 /home/pi/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/pi/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
```

What is wrong?

AlexMavpa, 2020-12-18:
Oh, finally i build the driver. I have added this:

```
#define RX_FLAG_SHORTPRE (1<<8 )
#define RX_FLAG_HT_GF (1<<13)
#define RX_FLAG_HT (1<<9 )
#define RX_FLAG_SHORT_GI (1<<11)
#define RX_FLAG_40MHZ (1<<10)
#define RX_FLAG_STBC_SHIFT (26)
```

to mt7601u.h file (after "_#define MT_RX_URB_SIZE   (PAGE_SIZE << MT_RX_ORDER)"_). Now everything is ok!

kuba-moo, 2020-12-18:
Why go through all this effort, tho? The driver is upstream, you should already have it built in your kernel. Maybe you're missing some "extra modules" package?

closed 2021-02-15 by kuba-moo