kuba-moo/mt7601u #72: Driver does not work with the newer version of the adapter

ars0, 2017-12-20:
I guess I have a newer version of the adapter. It is detected as a virtual CD first, and after that (depending on which Linux distro/version you are running) you can switch it to the wifi adapter mode, or it would be switched automatically by the OS. 

```
[11600.120154] usb 2-7: new high-speed USB device number 5 using ehci-pci
[11600.279671] usb 2-7: New USB device found, idVendor=148f, idProduct=7601
[11600.279681] usb 2-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[11600.279687] usb 2-7: Product: 802.11 n WLAN
[11600.279693] usb 2-7: Manufacturer: MediaTek
[11600.279698] usb 2-7: SerialNumber: 1.0
[11600.720159] usb 2-7: reset high-speed USB device number 5 using ehci-pci
[11600.871383] mt7601u 2-7:1.0: ASIC revision: 76010001 MAC revision: 76010500
[11600.929216] mt7601u 2-7:1.0: firmware: direct-loading firmware mt7601u.bin
[11600.929232] mt7601u 2-7:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[11604.532162] mt7601u 2-7:1.0: Vendor request req:07 off:09a8 failed:-110
[11607.792168] mt7601u 2-7:1.0: Vendor request req:02 off:09a8 failed:-110
[11611.024155] mt7601u 2-7:1.0: Vendor request req:07 off:0734 failed:-110
[11614.224127] mt7601u 2-7:1.0: Vendor request req:42 off:0230 failed:-110
[11617.488159] mt7601u 2-7:1.0: Vendor request req:07 off:0080 failed:-110
[11620.720186] mt7601u 2-7:1.0: Vendor request req:02 off:0080 failed:-110
[11623.952117] mt7601u 2-7:1.0: Vendor request req:02 off:0080 failed:-110
[11623.952210] mt7601u: probe of 2-7:1.0 failed with error -110
```
The driver would not initialize the device to a working state. 

The embedded virtual CD contains a Windows driver. I found the newer firmware in the Windows driver: FW_7662.bin (79216 bytes). Tried loading that firmware, but the result is just the same: 

```
[18219.503175] usb 1-3: new high-speed USB device number 26 using xhci_hcd
[18219.643839] usb 1-3: New USB device found, idVendor=148f, idProduct=2878
[18219.643843] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[18219.643846] usb 1-3: Product: Љ
[18219.643848] usb 1-3: Manufacturer: Љ
[18219.644809] usb-storage 1-3:1.0: USB Mass Storage device detected
[18219.645095] scsi host3: usb-storage 1-3:1.0
[18220.675434] scsi 3:0:0:0: CD-ROM            MediaTek Flash autorun    0.01 PQ: 0 ANSI: 0 CCS
[18220.676608] sr 3:0:0:0: [sr1] scsi3-mmc drive: 52x/52x cd/rw xa/form2 cdda tray
[18220.676736] sr 3:0:0:0: Attached scsi CD-ROM sr1
[18220.676798] sr 3:0:0:0: Attached scsi generic sg3 type 5
[18237.492826] usb 1-3: USB disconnect, device number 26
[18237.926996] usb 1-3: new high-speed USB device number 27 using xhci_hcd
[18238.077968] usb 1-3: New USB device found, idVendor=148f, idProduct=7601
[18238.077972] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18238.077975] usb 1-3: Product: 802.11 n WLAN
[18238.077978] usb 1-3: Manufacturer: MediaTek
[18238.077980] usb 1-3: SerialNumber: 1.0
[18238.203360] usb 1-3: reset high-speed USB device number 27 using xhci_hcd
[18238.346269] mt7601u 1-3:1.0: ASIC revision: 76010001 MAC revision: 76010500
[18238.347185] mt7601u 1-3:1.0: Firmware Version: 0.0.00 Build: 1 Build time: 201411280941____
[18241.650922] mt7601u 1-3:1.0: Vendor request req:02 off:0a44 failed:-110
[18244.882926] mt7601u 1-3:1.0: Vendor request req:02 off:0230 failed:-110
[18248.178877] mt7601u 1-3:1.0: Vendor request req:02 off:0400 failed:-110
[18251.506868] mt7601u 1-3:1.0: Vendor request req:02 off:0800 failed:-110
[18254.834788] mt7601u 1-3:1.0: Vendor request req:07 off:0404 failed:-110
[18258.098833] mt7601u 1-3:1.0: Vendor request req:02 off:0404 failed:-110
[18261.394772] mt7601u 1-3:1.0: Vendor request req:02 off:0800 failed:-110
[18264.690679] mt7601u 1-3:1.0: Vendor request req:02 off:0238 failed:-110
[18267.922673] mt7601u 1-3:1.0: Vendor request req:07 off:0238 failed:-110
[18271.218679] mt7601u 1-3:1.0: Vendor request req:02 off:0238 failed:-110
[18274.546658] mt7601u 1-3:1.0: Vendor request req:02 off:0238 failed:-110
[18277.846612] mt7601u 1-3:1.0: Vendor request req:02 off:09a0 failed:-110
[18281.110494] mt7601u 1-3:1.0: Vendor request req:02 off:09a4 failed:-110
[18284.438492] mt7601u 1-3:1.0: Vendor request req:02 off:09c4 failed:-110
[18287.666413] mt7601u 1-3:1.0: Vendor request req:02 off:0a6c failed:-110
[18290.898392] mt7601u 1-3:1.0: Vendor request req:42 off:0230 failed:-110
[18294.230357] mt7601u 1-3:1.0: Vendor request req:07 off:0080 failed:-110
[18297.494309] mt7601u 1-3:1.0: Vendor request req:02 off:0080 failed:-110
[18300.786366] mt7601u 1-3:1.0: Vendor request req:02 off:0080 failed:-110
[18300.786440] mt7601u: probe of 1-3:1.0 failed with error -110
```

kuba-moo, 2017-12-21:
AFAIK MediaTek no longer provides open source drivers for this device, I don't have any info on how to fix this...

stralex7, 2018-01-31:
@ars0 if you are refering to the revision 4 of this device it still works on 4.4 kernel

ars0, 2018-01-31:
@stralex7 I am probably referring to another revision because the device does not work on 4.4.0. How do I find out revision of the device?

stralex7, 2018-02-01:
@ars0 Sorry, for misleading you, it really just works on my particular installation with 4.4.0, i tried another PC and it failed with the same error.

P.S. The version is written on the sticker new the S/N, mine says V4.0

kuba-moo, 2018-02-01:
FWIW that's a TP-Link board version sticker, only indirectly related to the chip itself: https://wikidevi.com/wiki/TP-LINK_TL-WN727N_v4
Please do complain to MediaTek, the more people complain the more likely they are to help.

BaconCatBug, 2018-02-17:
I too am having this same issue. "Warning: unsupported EEPROM version 0d" is one error I get.

fafactx, 2018-08-26:
I have solve the same problem, just see the latest pull requests i pushed. or just commit the error you occured in usb.c .and the head file should the same to my mt7691.h.

fafactx, 2018-08-27:
I repaire this on my pull request 。 or you can just add follow code to you proj in mt7601u.h。
//just for kali 2018.2 kernel version >4.2
#define RX_FLAG_SHORTPRE (1<<8 )
#define RX_FLAG_HT_GF (1<<13)
#define RX_FLAG_HT (1<<9 )
#define RX_FLAG_SHORT_GI (1<<11)
#define RX_FLAG_40MHZ (1<<10)
#define RX_FLAG_STBC_SHIFT (26)
and commit this code in usb.c 128 line.
//mt7601u_vendor_request(dev, MT_VEND_DEV_MODE, USB_DIR_OUT,MT_VEND_DEV_MODE_RESET, 0, NULL, 0);

so
enjoy. but the drive have a bug , it is not so good,so if you want a short delay in ping echo. if i have time,i will repaire it ,later.

jeremyb31, 2019-11-17:
To fix go into mcu.c and comment line 450 // mt7601u_vendor_reset(dev);
This same issue is in the kernel module in kernels 4.15 and newer

closed 2021-02-15 by kuba-moo