kuba-moo/mt7601u #75: error when compiling in Linux deepin 。

Seven4X, 2018-06-19:
uname -a
Linux seven-PC 4.15.0-21deepin-generic #21 SMP Mon May 21 05:20:09 UTC 2018 x86_64 GNU/Linux

seven@seven-PC:~/mt7601u$ make 
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-21deepin-generic'
Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /home/seven/mt7601u/usb.o
  CC [M]  /home/seven/mt7601u/init.o
  CC [M]  /home/seven/mt7601u/main.o
  CC [M]  /home/seven/mt7601u/mcu.o
  CC [M]  /home/seven/mt7601u/trace.o
  CC [M]  /home/seven/mt7601u/dma.o
symbolmap: 00000001: invalid section
  CC [M]  /home/seven/mt7601u/core.o
  CC [M]  /home/seven/mt7601u/eeprom.o
  CC [M]  /home/seven/mt7601u/phy.o
  CC [M]  /home/seven/mt7601u/mac.o
/home/seven/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/seven/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
    status->flag |= RX_FLAG_SHORTPRE;
                    ^~~~~~~~~~~~~~~~
                    RX_ENC_FLAG_SHORTPRE
/home/seven/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/seven/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT_GF;
                   ^~~~~~~~~~~~~
                   RX_ENC_FLAG_HT_GF
/home/seven/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function); did you mean ‘RX_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT;
                   ^~~~~~~~~~
                   RX_FLAG_HT_GF
/home/seven/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function); did you mean ‘RX_FLAG_SHORTPRE’?
   status->flag |= RX_FLAG_SHORT_GI;
                   ^~~~~~~~~~~~~~~~
                   RX_FLAG_SHORTPRE
/home/seven/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_STBC_SHIFT’?
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^~~~~~~~~~~~~~~~~~
                        RX_ENC_FLAG_STBC_SHIFT
/home/seven/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function); did you mean ‘RX_FLAG_HT’?
   status->flag |= RX_FLAG_40MHZ;
                   ^~~~~~~~~~~~~
                   RX_FLAG_HT
scripts/Makefile.build:332: recipe for target '/home/seven/mt7601u/mac.o' failed
make[2]: *** [/home/seven/mt7601u/mac.o] Error 1
Makefile:1552: recipe for target '_module_/home/seven/mt7601u' failed
make[1]: *** [_module_/home/seven/mt7601u] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-21deepin-generic'
Makefile:8: recipe for target 'default' failed
make: *** [default] Error 2

ziqiangxu, 2018-06-19:
Change the 4th line in makefile to:
KDIR ?= /lib/modules/4.9.0-deepin11-amd64/build

Seven4X, 2018-06-20:
@ziqiangxu  but in Makefile : 
```
ifneq ($(KERNELRELEASE),)
KDIR ?= /lib/modules/$(KERNELRELEASE)/build
endif

KDIR ?= /lib/modules/`uname -r`/build

default:
	$(MAKE) -C $(KDIR) M=$$PWD
clean:
	$(MAKE) -C $(KDIR) M=$$PWD clean
install:
	$(MAKE) -C $(KDIR) M=$$PWD modules_install
```

I had a right KDIR directory : 
```
echo /lib/modules/`uname -r`/build
/lib/modules/4.15.0-21deepin-generic/build
seven@seven-PC:~/kuba_moo/mt7601u$ l /lib/modules/4.15.0-21deepin-generic/build
lrwxrwxrwx 1 root root 46 6月  18 09:52 /lib/modules/4.15.0-21deepin-generic/build -> /usr/src/linux-headers-4.15.0-21deepin-generic
```

kuba-moo, 2018-06-20:
May I ask why compile this driver out-of-tree at all? The upstream version in v4.15 is almost identical.

Seven4X, 2018-06-20:
@kuba-moo  Beacuse that's not work  when I pluging the usb-wifi .and also I disable security boot .
and  try  reinstall the mod using  inmod rmmod command,also not work.
 
 seven@seven-PC:~$ dmesg  | grep mt
[    0.000000] Found optimal setting for mtrr clean up
[    0.032000] Performance Events: PEBS fmt2+, Haswell events, 16-deep LBR, full-width counters, Intel PMU driver.
[    3.102485] mt7601u 3-4:1.0: ASIC revision: 76010001 MAC revision: 76010500
[    3.104035] mt7601u 3-4:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[    6.735919] mt7601u 3-4:1.0: Vendor request req:07 off:09a8 failed:-110
[    9.995916] mt7601u 3-4:1.0: Vendor request req:02 off:09a8 failed:-110
[   13.295904] mt7601u 3-4:1.0: Vendor request req:07 off:0734 failed:-110
[   16.619908] mt7601u 3-4:1.0: Vendor request req:42 off:0230 failed:-110
[   19.915895] mt7601u 3-4:1.0: Vendor request req:07 off:0080 failed:-110
[   23.147922] mt7601u 3-4:1.0: Vendor request req:02 off:0080 failed:-110
[   23.335951] mt7601u: probe of 3-4:1.0 failed with error -110
[   23.335978] usbcore: registered new interface driver mt7601u
[   26.098401] mt7601u 3-1:1.0: ASIC revision: 76010001 MAC revision: 76010500
[   26.098920] mt7601u 3-1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[   29.675912] mt7601u 3-1:1.0: Vendor request req:07 off:09a8 failed:-110
[   32.875931] mt7601u 3-1:1.0: Vendor request req:02 off:09a8 failed:-110
[   36.075924] mt7601u 3-1:1.0: Vendor request req:07 off:0734 failed:-110
[   39.275896] mt7601u 3-1:1.0: Vendor request req:42 off:0230 failed:-110
[   42.539896] mt7601u 3-1:1.0: Vendor request req:07 off:0080 failed:-110
[   45.803894] mt7601u 3-1:1.0: Vendor request req:02 off:0080 failed:-110
[   49.131924] mt7601u 3-1:1.0: Vendor request req:02 off:0080 failed:-110
[   49.131957] mt7601u: probe of 3-1:1.0 failed with error -110

ziqiangxu, 2018-06-20:
```4.9.0-deepin11-amd64``` is the previous kernel version, it is diffrent from ```4.15.0-21deepin-generic```

Seven4X, 2018-06-20:
@ziqiangxu  dmesg is :" mt7601u: probe of 3-3:1.0 failed with error -110" ,after I reinstall the os

LIGHTYUAN, 2018-06-21:
I have the same problem.
I refer to [this](https://elixir.bootlin.com/linux/v3.14/source/include/net/mac80211.h#L846), from line 846 to line 878.
I define the six macros in file ./mt7601u/mt7601u.h:

1. #define RX_FLAG_SHORTPRE    1<<8
2. #define RX_FLAG_HT          1<<9
3. #define RX_FLAG_40MHZ       1<<10
4. #define RX_FLAG_SHORT_GI    1<<11
5. #define RX_FLAG_HT_GF       1<<13
6. #define RX_FLAG_STBC_SHIFT  26

It works in my OS.

Seven4X, 2018-06-24:
I gave up at last .
I repurchase a wireless network card  that have ` Realtek Semiconductor Corp.` chip

closed 2018-06-24 by Seven4X

ziqiangxu, 2018-06-24:
@Seven4X 
define the six macros in file ./mt7601u/mt7601u.h:

#define RX_FLAG_SHORTPRE 1<<8
#define RX_FLAG_HT 1<<9
#define RX_FLAG_40MHZ 1<<10
#define RX_FLAG_SHORT_GI 1<<11
#define RX_FLAG_HT_GF 1<<13
#define RX_FLAG_STBC_SHIFT 26
Some macro has been deleted or renamed in the new kernel header file，so we failed to compline。