kuba-moo/mt7601u #77: Error while 'make'

VikasTiwari199915, 2018-10-30:
**Here is the output :-**

vikas@vikas-desktop:~/Desktop/mt7601u$ make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-30-generic'
  CC [M]  /home/vikas/Desktop/mt7601u/usb.o
  CC [M]  /home/vikas/Desktop/mt7601u/init.o
  CC [M]  /home/vikas/Desktop/mt7601u/mcu.o
  CC [M]  /home/vikas/Desktop/mt7601u/trace.o
  CC [M]  /home/vikas/Desktop/mt7601u/dma.o
  CC [M]  /home/vikas/Desktop/mt7601u/phy.o
  CC [M]  /home/vikas/Desktop/mt7601u/mac.o
/home/vikas/Desktop/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/vikas/Desktop/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function)
    status->flag |= RX_FLAG_SHORTPRE;
                    ^
/home/vikas/Desktop/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/vikas/Desktop/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function)
   status->flag |= RX_FLAG_HT_GF;
                   ^
/home/vikas/Desktop/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function)
   status->flag |= RX_FLAG_HT;
                   ^
/home/vikas/Desktop/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function)
   status->flag |= RX_FLAG_SHORT_GI;
                   ^
/home/vikas/Desktop/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function)
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^
/home/vikas/Desktop/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function)
   status->flag |= RX_FLAG_40MHZ;
                   ^
scripts/Makefile.build:332: recipe for target '/home/vikas/Desktop/mt7601u/mac.o' failed
make[2]: *** [/home/vikas/Desktop/mt7601u/mac.o] Error 1
Makefile:1552: recipe for target '_module_/home/vikas/Desktop/mt7601u' failed
make[1]: *** [_module_/home/vikas/Desktop/mt7601u] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-30-generic'
Makefile:8: recipe for target 'default' failed
make: *** [default] Error 2
vikas@vikas-desktop:~/Desktop/mt7601u$

kuba-moo, 2018-10-30:
The driver is now upstream, the upstream version is kept more up to date. The 4.15 version should work just fine, but if you're having issues could you update the whole kernel?

Augus-Huang, 2018-11-14:
Is there any branch named 4.15?

Augus-Huang, 2018-11-14:
oo,I see. Just support linux 4.15?

rudy2steiner, 2018-11-20:
i met the same problem,so how to solve it

BSOrider, 2018-12-25:
:point_up: this worked for me

pedrombmachado, 2019-04-03:
The solution described here worked for me in Ubuntu 16.04 and 18.04:
https://github.com/kuba-moo/mt7601u/issues/64#issuecomment-352250127

I will reference it here:

Originally written by ingate.
Thanks to aleksander and Nidroide.
Tested on Ubuntu 14.04 (kernel 4.4), Ubuntu 17.10 (kernel 4.13) and Ubuntu 18.04 (4.15.0-36-generic)

1. Download corresponding kernel source from kernel.org. For example: if you have 4.4.0-104-generic download version 4.4.
2. From archive unpack just folder drivers/net/wireless/mediatek/mt7601u
3. Edit phy.c. Find function mt7601u_init_cal and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0); like so:

// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_RXIQ, 0);
// if (ret)
// return ret;
// ret = mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->dpd_temp);
// if (ret)
// return ret;

Find function mt7601u_phy_recalibrate_after_assoc and comment out call mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp); like so:

void mt7601u_phy_recalibrate_after_assoc(struct mt7601u_dev *dev)
{
// mt7601u_mcu_calibrate(dev, MCU_CAL_DPD, dev->curr_temp);

    mt7601u_rxdc_cal(dev);
}

4. Build module: make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
5. Remove device
6. sudo rmmod mt7601u
7. sudo insmod ./mt7601u.ko
8. Insert device
9. Check there are no errors in dmesg and interface appeared in ip link, check connection stability.
10. To make change persistent till next kernel upgrade: backup original module and replace with compiled. To find out where is original module run modinfo mt7601u (view string filename: /lib/modules/_KERNEL_VERSION_/kernel/drivers/net/wireless/mediatek/mt7601u/mt7601u.ko).

closed 2021-02-15 by kuba-moo