kuba-moo/mt7601u #80: error while 'make' in Ubuntu1904,Kernel 5.0.0

quanbisen, 2019-07-04:
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-21-generic'
  CC [M]  /home/lollipop/Documents/mt7601u/eeprom.o
/home/lollipop/Documents/mt7601u/eeprom.c: In function ‘mt7601u_efuse_physical_size_check’:
/home/lollipop/Documents/mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
  u8 data[map_reads * 16];
  ^~
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:286: /home/lollipop/Documents/mt7601u/eeprom.o] Error 1
make[1]: *** [Makefile:1606: _module_/home/lollipop/Documents/mt7601u] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-21-generic'
make: *** [Makefile:8: default] Error 2

jeremyb31, 2019-11-17:
Why are you using this, the kernel now has the module

quanbisen, 2019-11-20:
> Why are you using this, the kernel now has the module

I know.The author had written this message on Readme file.But my device not work,so I just would like to build this driver for a try.And you know,it failed.Now I am using another WIFI device and it work perfectly without install any driver.Aha.

Maranzalla, 2020-03-19:
/home/sithis/mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
   77 |  u8 data[map_reads * 16];
      |  ^~
cc1: all warnings being treated as errors                                                                   
make[3]: *** [/usr/src/linux-headers-5.4.0-kali4-common/scripts/Makefile.build:271: /home/sithis/mt7601u/eeprom.o] Error 1                                                                                              
make[2]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:1665: /home/sithis/mt7601u] Error 2        
make[1]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:179: sub-make] Error 2                     
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali4-amd64'                                       
make: *** [Makefile:8: default] Error 2

kuba-moo, 2021-02-15:
Please use the upstream driver, building this driver from GitHub is only needed on very old kernels which don't have the in-tree support. The in-tree / upstream driver is exactly the same code.
If something doesn't work please open an issue describing the problem you're having and we'll try to help with that particular issue.

closed 2021-02-15 by kuba-moo