kuba-moo/mt7601u #81: debian build, kernel 4.19, fails at mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’

supaplextor, 2019-08-04:
```
supaplex@claw:~/usr/src/mt7601u$ make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-5-amd64'
  CC [M]  /home/supaplex/usr/src/mt7601u/usb.o
  CC [M]  /home/supaplex/usr/src/mt7601u/init.o
  CC [M]  /home/supaplex/usr/src/mt7601u/main.o
  CC [M]  /home/supaplex/usr/src/mt7601u/mcu.o
  CC [M]  /home/supaplex/usr/src/mt7601u/trace.o
  CC [M]  /home/supaplex/usr/src/mt7601u/dma.o
  CC [M]  /home/supaplex/usr/src/mt7601u/core.o
  CC [M]  /home/supaplex/usr/src/mt7601u/eeprom.o
  CC [M]  /home/supaplex/usr/src/mt7601u/phy.o
  CC [M]  /home/supaplex/usr/src/mt7601u/mac.o
/home/supaplex/usr/src/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/supaplex/usr/src/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
    status->flag |= RX_FLAG_SHORTPRE;
                    ^~~~~~~~~~~~~~~~
                    RX_ENC_FLAG_SHORTPRE
/home/supaplex/usr/src/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/supaplex/usr/src/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT_GF;
                   ^~~~~~~~~~~~~
                   RX_ENC_FLAG_HT_GF
/home/supaplex/usr/src/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function); did you mean ‘RX_ENC_HT’?
   status->flag |= RX_FLAG_HT;
                   ^~~~~~~~~~
                   RX_ENC_HT
/home/supaplex/usr/src/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORT_GI’?
   status->flag |= RX_FLAG_SHORT_GI;
                   ^~~~~~~~~~~~~~~~
                   RX_ENC_FLAG_SHORT_GI
/home/supaplex/usr/src/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_STBC_SHIFT’?
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^~~~~~~~~~~~~~~~~~
                        RX_ENC_FLAG_STBC_SHIFT
/home/supaplex/usr/src/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function)
   status->flag |= RX_FLAG_40MHZ;
                   ^~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.19.0-5-common/scripts/Makefile.build:309: /home/supaplex/usr/src/mt7601u/mac.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-5-common/Makefile:1539: _module_/home/supaplex/usr/src/mt7601u] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-5-amd64'
make: *** [Makefile:8: default] Error 2
supaplex@claw:~/usr/src/mt7601u$ echo $?
2
supaplex@claw:~/usr/src/mt7601u$ uname -a
Linux claw 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) x86_64 GNU/Linux
supaplex@claw:~/usr/src/mt7601u$ cat /etc/issue
Debian GNU/Linux 10 \n \l

```

sgruszka, 2019-08-05:
Why not use mt7601u driver that is included in 4.19 kernel ?

supaplextor, 2019-08-05:
I somehow read the docs as "4.20" vs my 4.19. I also overlooked apt-file searching for this before. My apt-cache search for the module provided no results. dpkg -S looks okay (never got to make install, so this is a .deb already).

```
supaplex@claw:~/usr/src$ dpkg -S mt7601u.ko
linux-image-4.19.0-5-amd64: /lib/modules/4.19.0-5-amd64/kernel/drivers/net/wireless/mediatek/mt7601u/mt7601u.ko
supaplex@claw:~/usr/src$ dpkg -S /boot/vmlinuz-* 
linux-image-4.19.0-5-amd64: /boot/vmlinuz-4.19.0-5-amd64
supaplex@claw:~/usr/src$ sudo dmesg -c > dmesg-c.txt # wipe it out for below commands.
supaplex@claw:~/usr/src$ sudo modprobe mt7601u
modprobe: ERROR: could not insert 'mt7601u': Invalid argument
supaplex@claw:~/usr/src$ sudo dmesg
[144872.517847] mt7601u: disagrees about version of symbol ieee80211_start_tx_ba_cb_irqsafe
[144872.517850] mt7601u: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -22)
[144872.517932] mt7601u: disagrees about version of symbol ieee80211_free_hw
[144872.517933] mt7601u: Unknown symbol ieee80211_free_hw (err -22)
[144872.517965] mt7601u: disagrees about version of symbol ieee80211_tx_status_ext
[144872.517967] mt7601u: Unknown symbol ieee80211_tx_status_ext (err -22)
[144872.517980] mt7601u: disagrees about version of symbol ieee80211_register_hw
[144872.517981] mt7601u: Unknown symbol ieee80211_register_hw (err -22)
[144872.517990] mt7601u: disagrees about version of symbol ieee80211_get_hdrlen_from_skb
[144872.517992] mt7601u: Unknown symbol ieee80211_get_hdrlen_from_skb (err -22)
[144872.517999] mt7601u: disagrees about version of symbol ieee80211_wake_queue
[144872.518000] mt7601u: Unknown symbol ieee80211_wake_queue (err -22)
[144872.518041] mt7601u: disagrees about version of symbol ieee80211_get_tx_rates
[144872.518045] mt7601u: Unknown symbol ieee80211_get_tx_rates (err -22)
[144872.518053] mt7601u: disagrees about version of symbol ieee80211_send_bar
[144872.518054] mt7601u: Unknown symbol ieee80211_send_bar (err -22)
[144872.518139] mt7601u: disagrees about version of symbol ieee80211_stop_tx_ba_cb_irqsafe
[144872.518139] mt7601u: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -22)
[144872.518144] mt7601u: disagrees about version of symbol ieee80211_queue_delayed_work
[144872.518145] mt7601u: Unknown symbol ieee80211_queue_delayed_work (err -22)
[144872.518195] mt7601u: disagrees about version of symbol ieee80211_wake_queues
[144872.518196] mt7601u: Unknown symbol ieee80211_wake_queues (err -22)
[144872.518199] mt7601u: disagrees about version of symbol ieee80211_free_txskb
[144872.518200] mt7601u: Unknown symbol ieee80211_free_txskb (err -22)
[144872.518203] mt7601u: disagrees about version of symbol ieee80211_alloc_hw_nm
[144872.518204] mt7601u: Unknown symbol ieee80211_alloc_hw_nm (err -22)
[144872.518222] mt7601u: disagrees about version of symbol ieee80211_tx_status
[144872.518223] mt7601u: Unknown symbol ieee80211_tx_status (err -22)
[144872.518225] mt7601u: disagrees about version of symbol ieee80211_stop_queue
[144872.518226] mt7601u: Unknown symbol ieee80211_stop_queue (err -22)
[144872.518229] mt7601u: disagrees about version of symbol ieee80211_stop_queues
[144872.518229] mt7601u: Unknown symbol ieee80211_stop_queues (err -22)
[144872.518241] mt7601u: disagrees about version of symbol ieee80211_unregister_hw
[144872.518242] mt7601u: Unknown symbol ieee80211_unregister_hw (err -22)
[144872.518272] mt7601u: disagrees about version of symbol ieee80211_rx_napi
[144872.518273] mt7601u: Unknown symbol ieee80211_rx_napi (err -22)
supaplex@claw:~/usr/src$ ls -l;
total 132
drwxr-xr-x 13 supaplex supaplex   4096 Aug  2 11:29 backport-iwlwifi
-rw-r--r--  1 supaplex supaplex 100289 Aug  5 07:46 dmesg-c.txt
drwxr-xr-x  3 supaplex supaplex   4096 Aug  4 14:16 hostapd
drwxr-xr-x 88 supaplex supaplex  20480 Aug  2 11:32 linux-firmware
drwxr-xr-x  4 supaplex supaplex   4096 Aug  4 14:17 mt7601u
```

I think I have a conflict with the Intel AX200 (WIFI 6) driver.  I reinstalled just the kernel, with no difference in dmesg logs.  Backing the Intel driver out in the meantime anyway (no 5GHZ softAP channels! GRR)

```
supaplex@claw:~/usr/src/backport-iwlwifi$ sudo make uninstall
  uninstall /lib/modules/4.19.0-5-amd64/updates/drivers/net/wireless/intel/iwlwifi/xvt/iwlxvt.ko
  uninstall /lib/modules/4.19.0-5-amd64/updates/drivers/net/wireless/intel/iwlwifi/mvm/iwlmvm.ko
  uninstall /lib/modules/4.19.0-5-amd64/updates/drivers/net/wireless/intel/iwlwifi/iwlwifi.ko
  uninstall /lib/modules/4.19.0-5-amd64/updates/net/wireless/cfg80211.ko
  uninstall /lib/modules/4.19.0-5-amd64/updates/net/mac80211/mac80211.ko
  uninstall /lib/modules/4.19.0-5-amd64/updates/compat/compat.ko

```

closed 2019-08-05 by supaplextor