kuba-moo/mt7601u #82: Error while make

MoudhafferBlg, 2020-01-19:
`make clean` runs fine.

`sudo make` or `make` runs into this error :

root@kali:~/mt7601u# make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-kali2-amd64'
  CC [M]  /root/mt7601u/eeprom.o
/root/mt7601u/eeprom.c: In function ‘mt7601u_efuse_physical_size_check’:
/root/mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
   77 |  u8 data[map_reads * 16];
      |  ^~
cc1: all warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.4.0-kali2-common/scripts/Makefile.build:271: /root/mt7601u/eeprom.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.4.0-kali2-common/Makefile:1665: /root/mt7601u] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-kali2-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali2-amd64'
make: *** [Makefile:8: default] Error 2

I'd really appreciate the help I've been working on this all day can't find a fix

Maranzalla, 2020-03-19:
the same problem
/home/sithis/mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
   77 |  u8 data[map_reads * 16];
      |  ^~
cc1: all warnings being treated as errors                                                                   
make[3]: *** [/usr/src/linux-headers-5.4.0-kali4-common/scripts/Makefile.build:271: /home/sithis/mt7601u/eeprom.o] Error 1                                                                                              
make[2]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:1665: /home/sithis/mt7601u] Error 2        
make[1]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:179: sub-make] Error 2                     
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali4-amd64'                                       
make: *** [Makefile:8: default] Error 2

kuba-moo, 2021-02-15:
Please use the upstream driver, building this driver from GitHub is only needed on very old kernels which don't have the in-tree support. The in-tree / upstream driver is exactly the same code.

closed 2021-02-15 by kuba-moo