kuba-moo/mt7601u #83: Failed to make

no-1imit, 2020-03-12:
Hi, I'm having Debian buster
when I run `make` I get these errors:
```
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-4.19.0-8-amd64'
  CC [M]  /home/behrooz/opt/mt7601u/usb.o
  CC [M]  /home/behrooz/opt/mt7601u/init.o
  CC [M]  /home/behrooz/opt/mt7601u/main.o
  CC [M]  /home/behrooz/opt/mt7601u/mcu.o
  CC [M]  /home/behrooz/opt/mt7601u/trace.o
  CC [M]  /home/behrooz/opt/mt7601u/dma.o
  CC [M]  /home/behrooz/opt/mt7601u/core.o
  CC [M]  /home/behrooz/opt/mt7601u/eeprom.o
  CC [M]  /home/behrooz/opt/mt7601u/phy.o
  CC [M]  /home/behrooz/opt/mt7601u/mac.o
/home/behrooz/opt/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/behrooz/opt/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
    status->flag |= RX_FLAG_SHORTPRE;
                    ^~~~~~~~~~~~~~~~
                    RX_ENC_FLAG_SHORTPRE
/home/behrooz/opt/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/behrooz/opt/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT_GF;
                   ^~~~~~~~~~~~~
                   RX_ENC_FLAG_HT_GF
/home/behrooz/opt/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function); did you mean ‘RX_ENC_HT’?
   status->flag |= RX_FLAG_HT;
                   ^~~~~~~~~~
                   RX_ENC_HT
/home/behrooz/opt/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORT_GI’?
   status->flag |= RX_FLAG_SHORT_GI;
                   ^~~~~~~~~~~~~~~~
                   RX_ENC_FLAG_SHORT_GI
/home/behrooz/opt/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_STBC_SHIFT’?
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^~~~~~~~~~~~~~~~~~
                        RX_ENC_FLAG_STBC_SHIFT
/home/behrooz/opt/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function)
   status->flag |= RX_FLAG_40MHZ;
                   ^~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.19.0-8-common/scripts/Makefile.build:309: /home/behrooz/opt/mt7601u/mac.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.19.0-8-common/Makefile:1537: _module_/home/behrooz/opt/mt7601u] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.0-8-amd64'
make: *** [Makefile:8: default] Error 2
```
what's wrong? thanks

Maranzalla, 2020-03-19:
while making
/home/sithis/mt7601u/eeprom.c: In function ‘mt7601u_efuse_physical_size_check’:
/home/sithis/mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
   77 |  u8 data[map_reads * 16];
      |  ^~
cc1: all warnings being treated as errors                                                                   
make[3]: *** [/usr/src/linux-headers-5.4.0-kali4-common/scripts/Makefile.build:271: /home/sithis/mt7601u/eeprom.o] Error 1                                                                                              
make[2]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:1665: /home/sithis/mt7601u] Error 2        
make[1]: *** [/usr/src/linux-headers-5.4.0-kali4-common/Makefile:179: sub-make] Error 2                     
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-kali4-amd64'                                       
make: *** [Makefile:8: default] Error 2

AlvifSandana, 2020-05-08:
hi, i get some trouble while compiling

/home/alvif/tplink/mt7601u/mac.c: In function ‘mt76_mac_process_rate’:
/home/alvif/tplink/mt7601u/mac.c:408:20: error: ‘RX_FLAG_SHORTPRE’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORTPRE’?
    status->flag |= RX_FLAG_SHORTPRE;
                    ^~~~~~~~~~~~~~~~
                    RX_ENC_FLAG_SHORTPRE
/home/alvif/tplink/mt7601u/mac.c:408:20: note: each undeclared identifier is reported only once for each function it appears in
/home/alvif/tplink/mt7601u/mac.c:417:19: error: ‘RX_FLAG_HT_GF’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_HT_GF’?
   status->flag |= RX_FLAG_HT_GF;
                   ^~~~~~~~~~~~~
                   RX_ENC_FLAG_HT_GF
/home/alvif/tplink/mt7601u/mac.c:420:19: error: ‘RX_FLAG_HT’ undeclared (first use in this function); did you mean ‘RX_ENC_HT’?
   status->flag |= RX_FLAG_HT;
                   ^~~~~~~~~~
                   RX_ENC_HT
/home/alvif/tplink/mt7601u/mac.c:429:19: error: ‘RX_FLAG_SHORT_GI’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_SHORT_GI’?
   status->flag |= RX_FLAG_SHORT_GI;
                   ^~~~~~~~~~~~~~~~
                   RX_ENC_FLAG_SHORT_GI
/home/alvif/tplink/mt7601u/mac.c:432:24: error: ‘RX_FLAG_STBC_SHIFT’ undeclared (first use in this function); did you mean ‘RX_ENC_FLAG_STBC_SHIFT’?
   status->flag |= 1 << RX_FLAG_STBC_SHIFT;
                        ^~~~~~~~~~~~~~~~~~
                        RX_ENC_FLAG_STBC_SHIFT
/home/alvif/tplink/mt7601u/mac.c:435:19: error: ‘RX_FLAG_40MHZ’ undeclared (first use in this function)
   status->flag |= RX_FLAG_40MHZ;
                   ^~~~~~~~~~~~~

banos, 2020-05-10:
Yup. I'm seeing same as above trying to build against raspbian on the Pi 2 (4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux) commit f16e91dad2af9d57aef477cc1f522040353849f5

This fork seems to have resolved these issues https://github.com/pedrombmachado/mt7601u
But this just leads to my next issue :-/

AlvifSandana, 2020-05-11:
I just fix for debian 10 kernel https://github.com/AlvifSandana/mt7601u

chalan51, 2020-08-27:
I got the same error in the make Maranzalla: 
mt7601u/eeprom.c:77:2: error: ISO C90 forbids array ‘data’ whose size can’t be evaluated [-Werror=vla]
have you got a fix for it?

kuba-moo, 2021-02-15:
Please use the upstream driver, building this driver from GitHub is only needed on very old kernels which don't have the in-tree support. The in-tree / upstream driver is exactly the same code.
If something doesn't work please open an issue describing the problem you're having and we'll try to help with that particular issue.

closed 2021-02-15 by kuba-moo