kuba-moo/mt7601u #93: TP-LINK tl-wn726n mt7601u mt7601u: probe of 1-5:1.0 failed with error -110

JuneZhao, 2023-07-04:
I have been trying to use TP-LINK USB wireless card on my x86 computer for openwrt OS, but it failed with below message with the driver, I've been tried to use different port on my computer but all failed with the same error message
[root@OpenWrt:03:02 PM ~] # dmesg | grep -i mt76
[ 19.525102] usbcore: registered new interface driver mt7601u
[ 19.546666] usbcore: registered new interface driver mt76x2u
[70202.627881] usbcore: registered new interface driver mt7663u
[70424.256744] usbcore: registered new interface driver mt76x0u
[72174.494949] mt7601u 1-5:1.0: ASIC revision: 76010001 MAC revision: 76010500
[72174.514874] mt7601u 1-5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[72177.952209] mt7601u 1-5:1.0: Vendor request req:07 off:09a8 failed:-110
[72181.362209] mt7601u 1-5:1.0: Vendor request req:02 off:09a8 failed:-110
[72184.772209] mt7601u 1-5:1.0: Vendor request req:07 off:0734 failed:-110
[72188.182209] mt7601u 1-5:1.0: Vendor request req:42 off:0230 failed:-110
[72191.592237] mt7601u 1-5:1.0: Vendor request req:07 off:0080 failed:-110
[72195.002209] mt7601u 1-5:1.0: Vendor request req:02 off:0080 failed:-110
[72198.412210] mt7601u 1-5:1.0: Vendor request req:02 off:0080 failed:-110
[72198.432054] mt7601u: probe of 1-5:1.0 failed with error -110
[74441.294950] mt7601u 1-5:1.0: ASIC revision: 76010001 MAC revision: 76010500
[74441.296349] mt7601u 1-5:1.0: Direct firmware load for mediatek/mt7601u.bin failed with error -2
[74441.296352] mt7601u 1-5:1.0: Falling back to sysfs fallback for: mediatek/mt7601u.bin
[74441.299246] mt7601u: probe of 1-5:1.0 failed with error -12
[74573.134950] mt7601u 1-5:1.0: ASIC revision: 76010001 MAC revision: 76010500
[74573.136382] mt7601u 1-5:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[74576.572247] mt7601u 1-5:1.0: Vendor request req:07 off:09a8 failed:-110
[74579.982247] mt7601u 1-5:1.0: Vendor request req:02 off:09a8 failed:-110
[74583.392247] mt7601u 1-5:1.0: Vendor request req:07 off:0734 failed:-110
[74586.802247] mt7601u 1-5:1.0: Vendor request req:42 off:0230 failed:-110
[74590.212248] mt7601u 1-5:1.0: Vendor request req:07 off:0080 failed:-110
[74593.622247] mt7601u 1-5:1.0: Vendor request req:02 off:0080 failed:-110
[74597.032247] mt7601u 1-5:1.0: Vendor request req:02 off:0080 failed:-110
[74597.052097] mt7601u: probe of 1-5:1.0 failed with error -110

My USB device:
[root@OpenWrt:03:02 PM ~] # lsusb
Bus 001 Device 011: ID 148f:7601 MediaTek 802.11 n WLAN

Any ideas?

m666m, 2023-07-12:
Is this a new buy one ? 

I have the same problem on Fedora 38, with a 5 years one

JuneZhao, 2023-07-12:
> n is unplug and re-plug, unplug and re-plug, unplug and re-plug...

It doesn't work for me, my wireless usb card never get recognized in openwrt system, but it can be recognized in my old Windows 7 or Windows 11 system without issue.

StevenKim523, 2023-11-20:
> > n is unplug and re-plug, unplug and re-plug, unplug and re-plug...
> 
> It doesn't work for me, my wireless usb card never get recognized in openwrt system, but it can be recognized in my old Windows 7 or Windows 11 system without issue.

does your adapter have a embedded flash chip which stores driver of Windows? and when you plug the adapter into Windows, it will be first recognized as a Virtual CD-ROM device, then you can open the CD-ROM and double click setup.exe to install the driver. after that, the adapter will be drived and begin to work normally

JuneZhao, 2023-11-20:
> does your adapter have a embedded flash chip which stores driver of Windows? and when you plug the adapter into Windows, it will be first recognized as a Virtual CD-ROM device, then you can open the [...]

It is an USB Wifi adaptor (no need of driver), no need to install driver in Windows